pyvisa/pyvisa #292: Issues with connecting to devices with PyVisa

MaWulfIST, 2017-11-20:
Hi everyone,

I have some issues with the python package PyVisa that you maintain. I have the hope that someone can help me to solve them. 
We use your software package to control various devices in our lab. We realized that while opening the resource manager (command “rm = visa.ResourceManager()” in PyVisa) there are two issues:
1.	It seems I am already accessing all the devices that the resource manager can find in our local area network with this command. Which means that if someone else is using the device for measuring at that moment I will disturb his measurements. 
2.	The resource manager does not find all the devices in our network in our network but only a selection.

My questions to these two issues are the following:
-	Can you tell me how I can open specific devices while not open / communicating with another device that I do not want to (issue 1)?
-	What determines which devices are found by the resource manager (issue 2)?
-	How can I remove/add devices from the list of devices found by the resource manager (issue 1 and 2)?

Thanks,
Matthias Wulf

MatthieuDartiailh, 2017-11-21:
Hi @MaWulfIST,

Looking at the code run when simply opening the resource manager and not attempting to list any device, VISA should not attempt to communicate with any instrument so I am a bit surprised by the behavior you describe in 1. Do you do anything in addition to creating the resource manager, like listing resources ? If you can confirm that simply opening the ResourceManager causes issues such as interrupting a query being run by another computer on the same network, I will try to investigate more but basically we simply call the `viOpenDefaultRM` function of VISA and if this the root of the issue, I do not know what we will be able to do.

Concerning your point 2, by default VISA does not look for raw socket instruments, which may be why some instruments do not show up. The `list_resources` method accept a string specifying for which resources to look for. At the moment the syntax is not described in pyvisa docs but the issue https://github.com/pyvisa/pyvisa/issues/280 link the relevant parts of the VISA documentation. Try playing with this and see if you can get all your resources to show up.

This should answer to your two last questions. For the first one, I believe that if you know the address of the instrument beforehand and simply do:
```python
rm = visa.ResourceManager()
res = rm.open_resource('GPIB::1::INSTR')
```
you should not see any conflict, but I may be wrong. If you can confirm the issue please post a minimal script reproducing the issue.

MaWulfIST, 2017-11-24:
Hi MatthieuDartiailh 
Thanks for your fast reply.
I looked into more detail into the problem and can tell you now that I was wrong before.
PyVisa is actually not the problem why I was contacting other devices but another library.
Thus, issue 1 is basically solved.
For issue 2: it cannot be explained by the fact that VISA does not look for raw socket instrument. I called the list_resources method with the option “?*” and it lists indeed also the raw documents. However, I know there are more devices in our lab network not listed.
Anyway, this is a second order problem, since I can connect to the devices, which are not listed, as long as I know their IP address.

MatthieuDartiailh, 2017-11-26:
For your second issue, are those instruments detected by NI-MAX ? Is there a relation between all the undetected instruments (different subnet (first 6 digits of the IP)) ?

MaWulfIST, 2017-11-27:
Indeed, NI-MAX does not find all instruments either. However, they are in the same subnet.

MatthieuDartiailh, 2017-11-27:
If NI-MAX does not find them, there is probably nothing we can do on our end. Closing as all issues that can be addressed are fixed it seems. Feel free to re-open if necessary.

closed 2017-11-27 by MatthieuDartiailh