pyvisa/pyvisa #293: resource write and query methods shall accept list

skrchnavy, 2017-11-22:
### Introduction:

It is possible to send multiple commands to resource using `;` as a delimiter.
* for write: `resource.write('CMDA;CMDB;CMDC')`
* for query:`resource.query('CMDA;CMDB;CMDC')` which could return `'ANSWA;ANSWB;ANSWC'`

For some cases when formatting is used, this could be not clean
```resource.write('CMD{0};CMD{1};CMD{2}'.format(parA, parB, parC))```

### Proposal

It shall be possible to send above commands as list of strings. And also when list was send in query, also list of strings can be returned (`;` as delimiter removed)

* for write: 
```
    resource.write(['CMDA',
                    'CMDB',
                    'CMDC']
```
* for query:
 ```
    resource.query(['CMDA',
                    'CMDB',
                    'CMDC'])
```

Then query could return array of strings: `['ANSWA', 'ANSWB', 'ANSWC']`

for formating it could be done per command and also commands can be then written per line

 ```
    resource.query(['CMD{0}'.format(parA),
                    'CMD{0}'.format(parB),
                    'CMD{0}'.format(parC)])
```

### Implementation
It is possible to convert list of strings to string delimited `;` using:
```
def write(input_data):
    if isinstance(input_data, list):
        command = ';'.join(input_data)
    else:
        command = input_data
```
 
for query, when input_data is list:
```
    if isinstance(input_data, list):
        output_data = output_data.split(';')
```

skrchnavy, 2017-11-22:
This is proposal for discussion, if it is worth to implement it. And I am not sure, which write methods shall be updated, there are few types of resources and I am sure about `MessageBasedResource` that shall be updated. Any other resource class affected to implement such change consistently?

MatthieuDartiailh, 2017-11-23:
I am not sure I like the idea for several reasons, but I am willing to discuss and could be convinced:
- not all instruments support concatenated commands (or use this separator but this is a detail)
- the write case can be emulated by `i.write(';'.join(cmd_list))` which is easy to use and the query case is similar
- I have not found concatenated commands to be needed so often that such a case requires a special treatment

I am curious of your use cases, could you describe some of them.

MatthieuDartiailh, 2018-01-25:
Feel free to re-open if you consider there is more to discuss here.

closed 2018-01-25 by MatthieuDartiailh