pyvisa/pyvisa #295: Drop support for Python 2.6 and 3.2 and 3.3

MatthieuDartiailh, 2017-11-26:
Those versions are really old. @hgrecco do you have an opinion

hugovk, 2017-12-05:
They're all EOL and no longer receive security updates.

Here's the pip installs for pyvisa from PyPI for the last month (via `pypinfo --percent --pip pyvisa pyversion`): showing virtually nothing for 2.6 and 3.3 and nothing for 3.2.

| python_version | percent | download_count |
| -------------- | ------: | -------------: |
| 2.7            |   45.3% |          1,364 |
| 3.6            |   30.5% |            918 |
| 3.5            |   17.0% |            513 |
| 3.4            |    6.7% |            201 |
| 3.7            |    0.2% |              6 |
| None           |    0.2% |              5 |
| 2.6            |    0.1% |              4 |
| 3.3            |    0.0% |              1 |

hugovk, 2017-12-05:
See PR https://github.com/pyvisa/pyvisa/pull/300.

commit 1b3a1a15 (2018-01-28): Merge #300

300: Drop support for EOL Python 2.6, 3.2 and 3.3 r=MatthieuDartiailh a=hugovk

Fixes #295.
PR #300 (2018-01-28): Drop support for EOL Python 2.6, 3.2 and 3.3
Fixes #295.
closed 2018-01-28 by bors[bot]