pyvisa/pyvisa #296: Make the code PEP8 compliant

MatthieuDartiailh, 2017-11-26:
When reading the code of Pyvisa the one thing I find the most uncomfortable is the random length of the line. As the project is now organization managed I propose to enforce PEP8 to improve the uniformity of the codebase.

DavidLutton, 2017-12-14:
E501 is pycodestyle code for line length
`pycodestyle --select=E501  | cut -d':' -f 1  | uniq -c  | sort -d `
Count and filename, where count > 20

>      24 ./pyvisa/resources/usb.py
>      32 ./pyvisa/resources/messagebased.py
>      35 ./pyvisa/constants.py
>      38 ./pyvisa/attributes.py
>      48 ./pyvisa/resources/resource.py
>      85 ./pyvisa/thirdparty/prettytable.py
>     201 ./pyvisa/highlevel.py
>     204 ./pyvisa/ctwrapper/functions.py

It may be a good idea to start using [pycodestyle](http://pycodestyle.readthedocs.io/en/latest/) but perhaps start with improvement, then at some later point enforcement. pycodestyle [formally known as PEP8](https://github.com/PyCQA/pycodestyle/issues/466)

`pycodestyle --statistics -qq`
> 21      E111 indentation is not a multiple of four
> 1       E114 indentation is not a multiple of four (comment)
> 2       E115 expected an indented block (comment)
> 1       E116 unexpected indentation (comment)
> 6       E122 continuation line missing indentation or outdented
> 2       E128 continuation line under-indented for visual indent
> 3       E201 whitespace after '{'
> 3       E202 whitespace before '}'
> 105     E203 whitespace before ':'
> 452     E221 multiple spaces before operator
> 1       E222 multiple spaces after operator
> 4       E225 missing whitespace around operator
> 1       E228 missing whitespace around modulo operator
> 16      E231 missing whitespace after ','
> 2       E251 unexpected spaces around keyword / parameter equals
> 16      E261 at least two spaces before inline comment
> 7       E262 inline comment should start with '# '
> 57      E265 block comment should start with '# '
> 1       E271 multiple spaces after keyword
> 24      E301 expected 1 blank line, found 0
> 21      E302 expected 2 blank lines, found 1
> 1       E303 too many blank lines (2)
> 4       E305 expected 2 blank lines after class or function definition, found 1
> 4       E402 module level import not at top of file
> 847     E501 line too long (82 > 79 characters)
> 2       E711 comparison to None should be 'if cond is not None:'
> 1       E712 comparison to True should be 'if cond is True:' or 'if cond:'
> 1       E713 test for membership should be 'not in'
> 4       E722 do not use bare except'
> 3       E741 ambiguous variable name 'l'
> 12      W291 trailing whitespace
> 27      W293 blank line contains whitespace
> 6       W391 blank line at end of file

MatthieuDartiailh, 2020-09-06:
Addressed as part of #511

closed 2020-09-06 by MatthieuDartiailh