pyvisa/pyvisa #297: Docstring format

MatthieuDartiailh, 2017-11-26:
I know that @hgrecco favors sphinx based docstring, but I have always found them quite unreadable. I would like to know the opinion of people about moving towards Numpy style docstrings.

DavidLutton, 2017-12-14:
[NumPyDoc vs. Sphinx and Google Format](https://bwanamarko.alwaysdata.net/napoleon/format_exception.html)  someone comparing 

[numpydoc docstring guide](https://numpydoc.readthedocs.io/en/latest/format.html) and [example](https://numpydoc.readthedocs.io/en/latest/example.html#example) is actually coherent guide of what to have in the different the sections of docstrings

Numpy style seems easier to read within the code as plain text

MatthieuDartiailh, 2020-09-06:
Addressed as part of #511

closed 2020-09-06 by MatthieuDartiailh