pyvisa/pyvisa #302: can't list ressources : VisaIOError: VI_ERROR_RSRC_NFOUND

elkevn, 2017-12-13:
Hi there,

I'm having trouble with pyvisa right now. I'm having the same error on both computers i'm using (one runs under ubuntu 16.04LTS and the other on Ubuntu 17.04) when i try to list ressources:

`rm.list_resources()
Traceback (most recent call last):

  File "<ipython-input-3-a14d33fc6d00>", line 1, in <module>
    rm.list_resources()

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)

  File "/opt/anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 238, in list_resources
    raise errors.VisaIOError(errors.StatusCode.error_resource_not_found.value)

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

I'm not entirely certain right now that the exact same error appears on the 17.04-running computer, but i do know the outcome is the same, and the error name is the same. 

I don't even really have any instrument connected yet, although i believe i should not have an error at this stage (as i did get a list on another computer in which this all worked just fine until i got another error, which i might post for another thread)

Thanks very much for your help in advance.

MatthieuDartiailh, 2017-12-13:
Indeed there is an inconsistency here between the visa wrapper implementation found in pyvisa.ctwrappers and the behavior you are seeing in pyvisa-py. I believe the original behavior found in pyvisa.ctwrappers is more correct. Would you be able to make a PR fixing the behavior in pyvisa-py ? It should be a simple fix.

elkevn, 2017-12-13:
Hey Matthieu,

thank you  very much for your answer. I should have said in my message that I a unfortunately very new/beginner to this whole topic of VISA implementations and to Python (if not to programming) in general.
Therefore,  i  must say i am not entirely sure of what you mean by inconsistency between the pyvisa.ctwrappers and the pyvisa-py.
Futhermore, english is not my mothertongue, so i don't really know what's PR: a quick search gave me perhaps the "pr" command in unix systems? as in comparing the two files, and fixing one with respect to the other?
Thank you very much for your help.

MatthieuDartiailh, 2017-12-13:
Sorry for the lack of clarity. In the context of Github, PR generally means pull request (the mechanism by which one can ask to merge some modification he made into the original repository).

By default PyVISA uses bindings towards the visa library implemented by National Instruments for example. Those bindings are based on the ctypes library. The implementation of list_resources is found here https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/highlevel.py#L196.
You are using pyvisa-py which emulates the behavior of the visa library for simple connections (serial, usb, tcpip, etc). The implementation of list resources is there: https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/highlevel.py#L223

Comparing the two, one can notice that in the first case we return an empty tuple if no resources are found whereas in the second case we raise an error. I believe we could simply skip the check at https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/highlevel.py#L237 and always return `resources`.

If you are not familiar with git, I will do the change.

MatthieuDartiailh, 2017-12-14:
Closed by https://github.com/pyvisa/pyvisa-py/pull/121

closed 2017-12-14 by MatthieuDartiailh