pyvisa/pyvisa #303: config electronics load

vazhuthanam1, 2017-12-14:
I am try to make interface(config) to my electronic load. I did open up usb  port, I expected my Eload config file should pop up like NI. I did not see any example how to do that.  Please explain how to do that. If you have any example please send it me.
Thanks
James samuel

DavidLutton, 2017-12-14:
PyVISA is a package to control test equipment 
> A Python package with bindings to the "Virtual Instrument Software Architecture" VISA library, in order to control measurement devices and test equipment via GPIB, RS232, or USB. 

Python has many ways to make a user interface, but PyVISA on its own does not provide one.

A different package could provide your user interface e.g. [Project Jupyter: From interactive Python to open science](https://www.youtube.com/watch?v=xuNj5paMuow) and [Jupyter Interactive Widget Ecosystem | SciPy 2017 Tutorial](https://www.youtube.com/watch?v=eWzY2nGfkXk) is an example of a package that could be used to provide controls

You will have to find the manual for your equipment that contains the programming commands that can be sent by PyVISA to your equipment. If it is a separate document its normally named the 'programming manual'

What is your equipments make and model?
Can you see your electronic load in rm.list_resources() like shown in the [Tutorial](http://pyvisa.readthedocs.io/en/stable/tutorial.html?highlight=list_resources())?

vazhuthanam1, 2017-12-15:
Thank you David,
Yes, I did see mu electronics load in rm.list_resource(),  the same name I copy and past at query did not like the name called syntax error. My instrument is B&K Precision  model 8600.I will try your advice and let you know.
Thanks
James Samuel

vazhuthanam1, 2017-12-15:
Hi David,
Here is some command from My Electronics load, how can I implement this command in python. I tried this command I got syntax error.
[SOURce:]FUNCtion
This command selects the input regulation mode of the electronic load.
CURRent constant current mode
RESistance constant resistance mode
VOLTage constant voltage mode
POWer constant power mode
45
Command Syntax [SOURce:] FUNCtion <function>
Parameters CURRent | RESistance | VOLTage | POWer |
*RST Value CURRent
Examples FUNC RES
Query Syntax [SOURce:]FUNCtion?
Returned Parameters <CRD>
[SOURce:]FUNCtion:MODE
This command determines whether the input regulation mode is controlled by values in a list or by the
FUNCtion command setting.
FIXed The regulation mode is determined by the FUNCtion or MODE command.
LIST The regulation mode is determined by the active list.
Command Syntax [SOURce:]FUNCtion:MODE <mode>
Parameters FIXed | LIST
*RST Value FIXed
Examples FUNC:MODE FIX
Query Syntax [SOURce:]FUNCtion:MODE?
Returned Parameters <CRD>
Related Commands FUN
>>> print (my_instrument.query('*IDN?'))
B&K Precision, 8600, 602197010727420008, 1.36-1.41

>>> print (my_instrument.query('[SOURce:]FUNCtion:MODE?'))
Traceback (most recent call last):
  File "<pyshell#13>", line 1, in <module>
    print (my_instrument.query('[SOURce:]FUNCtion:MODE?'))
  File "C:\Users\JSamuel\Documents\PyVISA-1.8\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()
  File "C:\Users\JSamuel\Documents\PyVISA-1.8\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Users\JSamuel\Documents\PyVISA-1.8\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\JSamuel\Documents\PyVISA-1.8\pyvisa\ctwrapper\functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\JSamuel\Documents\PyVISA-1.8\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> inst.timeout = 25000
Traceback (most recent call last):
  File "<pyshell#14>", line 1, in <module>
    inst.timeout = 25000
NameError: name 'inst' is not defined
Python result.
Any help will be appreciated.
Thanks
James Samuel

DavidLutton, 2017-12-16:
With **[SOURce:]FUNCtion:MODE?** the word **SOURce:** is optional under some circumstances, for now just remove the **[ ]** brackets, so **SOURce:FUNCtion:MODE?**

For querying the mode of your electronic load try
> print (my_instrument.query('SOURce:FUNCtion:MODE?'))

For your oscilloscope #304 timeout. Given that you used **my_instrument.** in your response change **inst.timeout = 25000** to **my_instrument.timeout = 25000**

vazhuthanam1, 2017-12-18:
Good morning Dave,
Thank you so much your help. I will follow your instructions.
Thanks
James Samuel

MatthieuDartiailh, 2018-02-25:
Closing feel free to re-open if there is more to discuss

closed 2018-02-25 by MatthieuDartiailh