pyvisa/pyvisa #304: ASCII value did not run

vazhuthanam1, 2017-12-14:
I tried to get some value from my instrument.this is my result. Can you help me what is I am missing here.
Thanks
>>> value = inst.query_ascii_values('CURV?')
	   
Traceback (most recent call last):
  File "<pyshell#42>", line 1, in <module>
    value = inst.query_ascii_values('CURV?')
  File "C:\Users\JSamuel\Documents\PyVISA-1.6.1\pyvisa\resources\messagebased.py", line 432, in query_ascii_values
    block = self.read()
  File "C:\Users\JSamuel\Documents\PyVISA-1.6.1\pyvisa\resources\messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "C:\Users\JSamuel\Documents\PyVISA-1.6.1\pyvisa\resources\messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\JSamuel\Documents\PyVISA-1.6.1\pyvisa\ctwrapper\functions.py", line 1569, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\JSamuel\Documents\PyVISA-1.6.1\pyvisa\ctwrapper\highlevel.py", line 181, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

DavidLutton, 2017-12-14:
CURV? is a command usually used with an oscilloscope, this can take some time to be sent from the scope.
So try setting the [timeout](http://pyvisa.readthedocs.io/en/stable/resources.html?highlight=timeout) of the instrument to allow for enough time
`inst.timeout = 25000  # 25 seconds`
`values = inst.query_ascii_values('CURV?')`

Possibly not relevant but you are not using the current release (1.8) of PyVISA

MatthieuDartiailh, 2018-02-25:
Closing fell free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh