pyvisa/pyvisa #305: pyvisa does not work

omerccohen, 2017-12-15:
hi, i tried to use pyvisa but when i run the python scrip i go the next error:
'''UnicodeDecodeError: 'ascii' codec can't decode byte 0xf9 in position 3: ordinal not in range(128)'''
here is the code:
'''import visa

def main():
    visa.log_to_screen()
    rm = visa.ResourceManager()

if __name__ == '__main__':
    main()'''

any ideas?

MatthieuDartiailh, 2017-12-15:
What version of Python are you running ? on which OS ? do you have non-ascii characters in your username or on the path ? Could you please post the full stack (I would like in particular to know if the error happens in the log_to_screen function or when creating the resource manager) ?

omerccohen, 2017-12-15:
python 2.7.6
windows 7 64bit 
2017-12-15 17:04:48,584 - pyvisa - DEBUG - No user defined library files
2017-12-15 17:04:48,585 - pyvisa - DEBUG - Could not open VISA wrapper <class 'pyvisa.ctwrapper.highlevel.NIVisaLibrary'>: 
'ascii' codec can't decode byte 0xf9 in position 3: ordinal not in range(128)
Traceback (most recent call last):
  File "C:/Users/Omer/PycharmProjects/syber/FinalProject/test.py", line 11, in <module>
    main()
  File "C:/Users/Omer/PycharmProjects/syber/FinalProject/test.py", line 7, in main
    rm = visa.ResourceManager()
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\site-packages\pyvisa\highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\site-packages\pyvisa\highlevel.py", line 1460, in open_visa_library
    return cls(argument)
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\site-packages\pyvisa\highlevel.py", line 87, in __new__
    for path in cls.get_library_paths():
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 75, in get_library_paths
    for library_path in ('visa', 'visa32', 'visa32.dll')]
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\ctypes\util.py", line 54, in find_library
    fname = os.path.join(directory, name)
  File "C:\Heights\PortableApps\PortablePython2.7.6.1\App\lib\ntpath.py", line 108, in join
    path += "\\" + b
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf9 in position 3: ordinal not in range(128)

MatthieuDartiailh, 2017-12-15:
Thanks for the log. It appears that you have not installed the VISA library for which PyVISA is only a wrapper. And for some reasons we get a path that is not properly handled by Python. You are using a quite old version of Python 2.7, updating it may help (if you can I would suggest updating to Python 3). You can also try to use the pure python backend by installing pyvisa-py and creating a resource manager as follows:
`rm = ResourceManager('@py')`

omerccohen, 2017-12-15:
Ok, if you have some steps which i should do please write it here

MatthieuDartiailh, 2017-12-17:
It depend what you can do and what you are trying to achieve. A simple solution could be to install Ni-VISA from this link: http://search.ni.com/nisearch/app/main/p/bot/no/ap/tech/lang/fr/pg/1/sn/catnav:du,n8:3.1637,ssnav:sup/ (the last version for your OS). After this, pyvisa should detect the library (and hopefully there will no path error)^and you should be able to create a resource manager adn experiment from there.

MatthieuDartiailh, 2018-02-25:
Closing please re-open if you have any more questions.

closed 2018-02-25 by MatthieuDartiailh