pyvisa/pyvisa #306: ValueError(UnicodeDecodeError) when reading from instrument

wiklod, 2018-01-12:
Recently I am trying to remotely control my R&S ZVL Spectrum Analyzer. Everything was going great so far, I successfully connected with the instrument and get some responses from it (eg. zvl.query('*IDN?') gave me 'Rohde&Schwarz,ZVL-3,1303.6509K03/102178,3.31'). 
But when I try to transfer some measurements data from it with query_ascii_values function (or any other) I receive:

```
>>> zvl.query_ascii_values('MMEM:DATA \'c:\Temp\TEST.ASC\'?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 457, in query_ascii_values
    return util.from_ascii_block(block, converter, separator, container)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/util.py", line 180, in from_ascii_block
    return container([converter(raw_value) for raw_value in data])
ValueError: could not convert string to float: 
```

I searched in the issues and found [this one](https://github.com/pyvisa/pyvisa/issues/120). According to it I changed log output  to screen, repeated and got:

```
>>> visa.log_to_screen()
>>> zvl.query_ascii_values('MMEM:DATA \'c:\Temp\TEST.ASC\'?')
2018-01-12 11:14:30,028 - pyvisa - DEBUG - Make call 11, (2, 10000, 10000, 136, "MMEM:DATA 'c:\\Temp\\TEST.ASC'?\r\n"), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb6a48f30>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb63515a8>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 861, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 734, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0xaf in position 15: ordinal not in range(128)
Logged from file rpc.py, line 296
2018-01-12 11:14:30,097 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5994e68>: '\x00\x00\x00\x06\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1f'
2018-01-12 11:14:30,103 - pyvisa - DEBUG - TCPIP::192.168.1.2::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-01-12 11:14:30,108 - pyvisa - DEBUG - Make call 12, (2, 2048, 10000, 10000, 0, 0), <bound method Vxi11Packer.pack_device_read_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb6a48f30>>, <bound method Vxi11Unpacker.unpack_device_read_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb63515a8>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 861, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 734, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0xaf in position 15: ordinal not in range(128)
Logged from file rpc.py, line 296
2018-01-12 11:14:40,251 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5994e68>: '\x00\x00\x00\x07\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x01\x00\x00\x00\x00'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 457, in query_ascii_values
    return util.from_ascii_block(block, converter, separator, container)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/util.py", line 180, in from_ascii_block
    return container([converter(raw_value) for raw_value in data])
ValueError: could not convert string to float: 

```

I tried with setting and changing zvl.read_termination, but id didn't seem to help. Next I changed sys.setdefaultencoding to 'utf-8' but it didn't helped also.

Then I found [this](https://stackoverflow.com/questions/27451942/importing-visa-waveform-from-an-oscilloscope-into-python) answer on stackoverflow, but neither changin instrument encoding to 'latin-1' or using query_binary_values helped in my case.

Interesting thing is that the problem occures also with read_raw()  function:

```
>>> zvl.read_raw()
2018-01-12 11:37:42,794 - pyvisa - DEBUG - TCPIP::192.168.1.2::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-01-12 11:37:42,797 - pyvisa - DEBUG - Make call 12, (2, 2048, 10000, 10000, 0, 0), <bound method Vxi11Packer.pack_device_read_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb6a48f30>>, <bound method Vxi11Unpacker.unpack_device_read_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb63515a8>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 861, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 734, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0xaf in position 15: ordinal not in range(128)
Logged from file rpc.py, line 296
2018-01-12 11:37:52,810 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5994e68>: '\x00\x00\x00\x08\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x01\x00\x00\x00\x00'
bytearray(b'')
```

Do you have any suggestions what else should I try?

MatthieuDartiailh, 2018-01-12:
Looking at the documentation (https://cdn.rohde-schwarz.com/pws/dl_downloads/dl_common_library/dl_manuals/gb_1/z/zvl_1/ZVL-K1_Operating_06.pdf), it is not clear in what format the instrument return the value. However as it accept a binary blob of data, I suspect it returns a binary block of data, so using query_ascii_values will not work. Do you know in what format the instrument is supposed to return you the data ?

I notice you are using pyvisa-py as a backend. What version are you using ? Could you try again with the master branch ? Also trying with Python 3 may help debug this as I suspect we would get less encoding issues. If you do not log to screen, what is the output of read_raw ?

wiklod, 2018-01-12:
I noticed when looking at [this](https://github.com/pyvisa/pyvisa-py/pull/97) pull request that I might be using outdated version so I uninstalled what I had and installed it directly from github. 
And my outputs changed. No more UnicodeDecodeErrors. They are replaced now with timeout errors:

```
>>> zvl.query_binary_values('MMEM:DATA \'c:\Temp\TEST.ASC\'?')
2018-01-12 12:48:38,688 - pyvisa - DEBUG - Make call 11, (5, 2000, 10000, 136, "MMEM:DATA 'c:\\Temp\\TEST.ASC'?\r\n"), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb61acc60>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb61acc38>>
2018-01-12 12:48:38,696 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb618ef80>: "\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x88\x00\x00\x00\x1fMMEM:DATA 'c:\\Temp\\TEST.ASC'?\r\n\x00"
2018-01-12 12:48:38,704 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb618ef80>: '\x00\x00\x00\x06\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1f'
2018-01-12 12:48:38,712 - pyvisa - DEBUG - TCPIP::192.168.1.2::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-01-12 12:48:38,717 - pyvisa - DEBUG - Make call 12, (5, 2048, 2000, 10000, 0, 0), <bound method Vxi11Packer.pack_device_read_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb61acc60>>, <bound method Vxi11Unpacker.unpack_device_read_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb61acc38>>
2018-01-12 12:48:38,725 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb618ef80>: "\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x00\x08\x00\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x00\x00\x00\x00\x00"
2018-01-12 12:48:40,888 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb618ef80>: '\x00\x00\x00\x07\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x01\x00\x00\x00\x00'
2018-01-12 12:48:40,894 - pyvisa - DEBUG - TCPIP::192.168.1.2::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 481, in query_binary_values
    block = self.read_raw()
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

So I found [this issue](https://github.com/pyvisa/pyvisa/issues/304), but changing timeout setting to 25s as mentioned there didn't helped.

Size of the file I would like to get is 29kB (don't know if important).

According to first part of your response.
It's my first contact with this instrument so, I don't have any valuable experience with it. Also I wasn't able to find specific information about format od data which instrument returns, but I used [this tutorial](https://www.rohde-schwarz.com/pl/driver-pages/remote-control/getting-started_231558.html) which I found on R&S ZVL product page. If you take a look at it then under "Step 3: Install Python, PyCharm and PyVISA" there are some Python 'Hello world' examples. I also found on R&S web page some more advanced [examples](https://cdn.rohde-schwarz.com/pws/service_support/driver_pagedq/files_1/directscpi/DirectSCPI_PyCharm_Python_Examples.zip)(direct link). 

This is one of the examples:
```python
# RTO/RTE Oscilloscope example for Python using PyVISA

import visa
import VISAresourceExtentions

# -----------------------------------------------------------
# Initialization
# -----------------------------------------------------------
rm = visa.ResourceManager()
RTB = rm.open_resource('TCPIP::10.212.1.90::INSTR')

# The file VISAresourceExtentions.py must be in the same folder as this file
# It contains PyVISA Visa.Resource extension functions for and 2 new exception types

# try block to catch any InstrumentErrorException()
try:
    RTB.write_termination = ''
    # Some instruments require LF at the end of each command. In that case, use:
    # scope.write_termination = '\n'
    RTB.ext_clear_status()  # Clear instrument io buffers and status
    print RTB.query('*IDN?')  # Query the Identification string
    RTB.write('*RST;*CLS')  # Reset the instrument, clear the Error queue
    RTB.ext_error_checking()  # Error Checking after Initialization block
    # -----------------------------------------------------------
    # Basic Settings:
    # -----------------------------------------------------------
    RTB.write('TIM:ACQT 0.01')  # 10ms Acquisition time
    RTB.write('CHAN1:RANG 5.0')  # Horizontal range 5V (0.5V/div)
[... cut ...]
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

What might be interesting, before updating pyvisa-py it had output:

```
b''
```
with no errors displayed.

wiklod, 2018-01-12:
So I guessed that it may be the problem with termination mark. For now I did not set it, because everything I tried gave me error:

```
>>> zvl.read_termination = '\n'
>>> zvl.query_binary_values('MMEM:DATA \'c:\Temp\TEST.ASC\'?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 481, in query_binary_values
    block = self.read_raw()
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 253, in read
    ret = self.sessions[session].read(count)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 107, in read
    self.lock_timeout, flags, term_char)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/vxi11.py", line 215, in device_read
    self.unpacker.unpack_device_read_resp)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 357, in make_call
    return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 247, in make_call
    pack_func(args)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/vxi11.py", line 113, in pack_device_read_parms
    self.pack_int(term_char)
  File "/usr/lib/python2.7/xdrlib.py", line 46, in result
    raise ConversionError(e.args[0])
xdrlib.ConversionError: cannot convert argument to integer
```

I tried '\n' at first because I noticed that at the end of instrument id there is one:
```
>>> zvl.query('*IDN?')
u'Rohde&Schwarz,ZVL-3,1303.6509K03/102178,3.31\n'
```

But I also tried with '\r' and ' ' with no luck.

Next, according to [this issue](https://github.com/pyvisa/pyvisa/issues/259) I changed baudrate, it didn't help.

MatthieuDartiailh, 2018-01-12:
Thanks for your detailed answer. 

Looking at the example you posted, I noticed they use a custom extension. When reading the memory the way you are trying they use:
```python
# This function queries a binary data from the instrument and writes them to a file on your PC
def ext_query_bin_data_to_file(device, query, pc_file_path):
    file_data = device.query_binary_values(query, datatype='s')[0]
    new_file = open(pc_file_path, "wb")
    new_file.write(file_data)
    new_file.close()
```
You could try something similar. 

Also I noticed that the command you use may be wrong. According to the manual of the instrument (cf my previous post) the command should be `"MMEM:DATA? 'c:\Temp\TEST.ASC.DAT'"` (Note that by using `"` to define the string you do not need to escape `'`). Could you try this ?

wiklod, 2018-01-12:
That was it!
Using their custom function everything worked flawlessly. I don't know why I decided not to test it.

Thank you a lot for all your help and effort.

To sum up: when using R&S instruments you have to use their functions :)

PS.
Should I change the Issue title?

MatthieuDartiailh, 2018-01-12:
It is not so much that you have to use their function but that their documentation is quite unclear as to the format in which the data are returned. Out of curiosity did you also change the command ? and can you use read_raw (you should be able to) ?
You can change the issue title to mention your instrument so that others can find it easily.

closed 2018-01-12 by MatthieuDartiailh

wiklod, 2018-01-12:
Yes, I did also change the command, it does not seem to work with the previous version. 
Changing the command and not using the function gave me exactly same result as before.

When it comes to read_raw it gives me error:
```
>>> data = zvl.read_raw("MMEM:DATA? 'c:\Temp\TEST.ASC'")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 253, in read
    ret = self.sessions[session].read(count)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/tcpip.py", line 84, in read
    if count < self.max_recv_size:
TypeError: unorderable types: str() < int()
```
or without an argument(I am not sure if it can be used in the way like above):
```
>>> data = zvl.read_raw()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

MatthieuDartiailh, 2018-01-12:
Sorry, I meant what do you get if you do:
```python 
zvl.write("MMEM:DATA? 'c:\Temp\TEST.ASC'")
zvl.read_raw()
```
The read command simply reads the instrument buffer so first you must use write to send it the order to transfer the data.

wiklod, 2018-01-12:
It also seems to work:

```
>>> zvl.write("MMEM:DATA? 'c:\Temp\TEST.ASC'")
(31, <StatusCode.success: 0>)
>>> zvl.read_raw()
b'#530315Type;ZVL-3;\r\nVersion;3.31;\r\nDate;12.Jan 18;\r\nMode;ANALYZER;\r\nCenter Freq;1010558076.152305;Hz\r\nFreq Offset;0.000000;Hz\r\nSpan;188160000.000000;Hz\r\nx-Axis;LIN;\r\nStart;916478076.152305;Hz\r\nStop;1104638076.152305;Hz\r\nRef Level;-20.000000;dBm\r\nLevel Offset;0.000000;dB\r\nRef Position;100.000000;%\r\ny-Axis;LOG;\r\nLevel Range;100.000000;dB\r\nRf Att;0.000000;dB\r\nRBW;3000000.000000;Hz\r\nVBW;10000000.000000;Hz\r\nSWT;0.002500;s\r\nTrace Mode;CLR/WRITE;\r\nDetector;AUTOPEAK;\r\nSweep Count;0;\r\nTrace 1:;;\r\nx-Unit;Hz;\r\ny-Unit;dBm;\r\nPreamplifier;OFF;\r\nTransducer;OFF;\r\nValues;501;\r\n916478076.15230465;-75.608840942382812;-104.71131134033203\r\n916854396.15230465;-71.835205078125;-96.659492492675781\r\n917230716.15230465;-74.159210205078125;-112.68564605712891\r\n917607036.15230465;-72.016853332519531;-96.100364685058594\r\n917983356.15230465;-74.931159973144531;-97.817840576171875\r\n918359676.15230465;-71.402320861816406;-131.38034057617187\r\n918735996.15230465;-73.80670166015625;-100.70699310302734\r\n919112316.15230465;-73.414260864257813;-104.90840148925781\r\n919488636.15230465;-71.969764709472656;-101.96784210205078\r\n919864956.15230465;-74.091064453125;-101.51921844482422\r\n920241276.15230465;-72.274360656738281;-106.75025177001953\r\n920617596.15230465;-72.905029296875;-106.49884796142578\r\n920993916.15230465;-73.673843383789063;-102.09170532226562\r\n921370236.15230465;-70.424270629882813;-98.9547119140625\r\n921746556.15230465;-74.887855529785156;-100.43540954589844\r\n922122876.15230465;-75.09906005859375;-100.14820861816406\r\n922499196.15230465;-68.377616882324219;-97.837112426757813\r\n922875516.15230465;-72.31744384765625;-100.08841705322266\r\n923251836.15230465;-73.40142822265625;-104.69190216064453\r\n923628156.15230465;-72.610252380371094;-91.795516967773438\r\n924004476.15230465;-73.382453918457031;-102.03539276123047\r\n924380796.15230465;-74.367851257324219;-102.07818603515625
...
\r\n1103885436.1523046;-73.359016418457031;-101.21112823486328\r\n1104261756.1523046;-71.383232116699219;-94.069786071777344\r\n1104638076.1523046;-73.900321960449219;-97.721290588378906\r\n\n'
```

MatthieuDartiailh, 2018-01-12:
Ok thanks. Actually from this you can deduce the type of data the instrument return, as first you have the `#` marking the beginning of the binary header, next 5 indicates that the length of the data is coded on five bytes and here we expect 30315 bytes. Next you see what look definitively like text hence the 's' datatype used in the tutorial. Hope this will help you should encounter similar issues in the future.

wiklod, 2018-01-12:
For sure it will. My main problem was not knowing how to properly use read_raw() function. In issues, topics, and answers I read many people mentioned that it is the best way to recognize the problem.  I should have started with it.

narayan-paudel, 2019-12-03:
I keep getting timeout error  VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed whenever I try to read the waveform in my PC from Rohde and Schwarz oscilloscope using scope.read_raw. I have connected my PC with oscilloscope using WLAN. My problem appears very similar to the original post. The solution to the original post could be helpful for me but I cannot figure out what the solution was.

MatthieuDartiailh, 2019-12-03:
The initial post had several issues. Could you post the output of `pyvisa-info` and what you have tried so far ?

fatemeh1179, 2023-11-02:
I had the same issue with timeout, as mentioned above, it was about the type of data.
In case it helps anyone, this code worked for me to communicate with a spectrum analyzer and get the resolution bandwidth.
import pyvisa
instrument_resource = 'GPIB0::20::INSTR'
rm = pyvisa.ResourceManager()
instrument = rm.open_resource(instrument_resource)
instrument.write('BAND:RES?')
resolution_bw = instrument.query_ascii_values('BAND:RES?')[0]
print(f"Current RBW: {resolution_bw} ")

Other commands like instrument.query(command) did not work for me and I got time out.