pyvisa/pyvisa #309: UnicodeDecodeError: 'ascii' codec can't decode byte 0xfe

svanan77, 2018-01-25:
Hi All,

I am trying to, 
a) connect to a signal generator (Agilent N5182A MXG) that supports ASCII range(256) using TCPIP. 
b) set a signal waveform to the signal generator

Unfortunately I am unable to set the waveform. Apparently the error shows that it supports only ASCII range(128) but not the extended ASCII range(129-256). The '0xfe' comes in the extended ASCII set.

Have anyone faced this problem? 
Is there a syntax to specify the encoding method pyvisa should use?

The error,

```
SK@PC:~/ubuntu/pyvisa$ python sg_pyvisa_simple.py
Traceback (most recent call last):
  File "sg_pyvisa_simple.py", line 89, in <module>
    inst_sg.write(set_waveform)
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 202, in write
    if message.endswith(term):
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfe in position 94: ordinal not in range(128)
```

Attaching
a) The script that produced the above error

```
import visa
import numpy
import itertools
import pylab as pl
import math

#######################################
# Variables
#######################################

IP_SG = '10.10.100.131'
siggen_filename ='myCombSig'
num_tone=16
num_smpl_tone=32 

#######################################
# Initialization
#######################################

rm = visa.ResourceManager('@py')

#######################################
# Helper Functions
#######################################

def df_open_connection(rm, ip):
	inst = rm.open_resource('TCPIP::'+ ip + '::INSTR')
	return inst

def df_close_connection(inst):
	inst.close()

def df_my_range(start, end, step):
	while start <= end:
		yield start
		start += step

def df_Comb_sig_gen(num_tone, num_smpl_tone):
	comb_real = []
	comb_imag = []
	comb_cmpx = []
	M = num_tone * num_smpl_tone
	N = 0
	C = num_tone
	k = pl.frange(float(-(C - 1)) / 2, float((C - 1)) / 2, 1)
	for n in df_my_range(-N + 1, M + N - 2, 1):
		cc = numpy.exp(2j * math.pi * k * numpy.array(n) / (C))
		comb_cmpx.append(numpy.sum(cc))  

	return (comb_cmpx)

def df_set_waveform(comb_signal, filename):

	scale  = numpy.amax([numpy.amax(comb_signal.real),numpy.amax(comb_signal.imag)])

	comb_signal_real_scale = numpy.array(comb_signal.real)/scale
	comb_signal_imag_scale = numpy.array(comb_signal.imag)/scale

	i = comb_signal_real_scale
	q = comb_signal_imag_scale

	i_dac = map(lambda x: int((x+32767) % 32767), i)
	q_dac = map(lambda x: int((x+32767) % 32767), q)

	iq_str = ""

	for i_val, q_val in itertools.izip(i_dac, q_dac):
		iq_str += chr((i_val >> 8) & 0x00FF) #32766 (0x7FFE) -> 7F
		iq_str += chr(i_val & 0x00FF)		 #32766 (0x7FFE) -> FE, error when writing FE
		iq_str += chr((q_val >> 8) & 0x00FF)
		iq_str += chr(q_val & 0x00FF)

	scpi_command = ':MEM:DATA "WFM1:{}",#{}{}{}\n'.format(filename.upper(), len(str(len(iq_str))), str(len(iq_str)), iq_str)

	return scpi_command

#################################################
# Action: To Set a Waveform on Signal Generator
#################################################

inst_sg = df_open_connection(rm, IP_SG)
inst_sg.write(':FREQ 1.8825 GHz; POWer -30DBM')

comb_signal = df_Comb_sig_gen(float(num_tone),float(num_smpl_tone))	
comb_signal = numpy.array(comb_signal)

set_waveform = df_set_waveform(comb_signal, siggen_filename)

inst_sg.write(set_waveform)

df_close_connection(inst_sg)`
```

b) The output of "python -m visa info"

```
Machine Details:
   Platform ID:    Linux-4.4.0-92-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /opt/anaconda2/bin/python
   Version:        2.7.12
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Jul  2 2016 17:42:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: N/A
      USB INSTR: Available via PyUSB (1.0.2). Backend: N/A
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
   sim:
      Version: 0.3
      Spec version: 1.1
```

If any other logs need to be produced to study the above issue, kindly share the syntax and I shall revert with the appropriate logs. Thanks for your help.

svanan77, 2018-01-25:
Thanks Matthieu for your advice.

Spent sometime to study the struct package.
Refer: https://docs.python.org/2/library/struct.html#format-characters

The instrumentation manual which I am using for Agilent N5182A MXG,
Refer: http://literature.cdn.keysight.com/litweb/pdf/N5180-90074.pdf

iq_str is a "string" datatype variable which collects "short int" i and q values along the way.

I have tried declaring the datatype as short int,

```
inst_sg.write_binary_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str), datatype='h')
```
pyvisa complains as below, 

```
Traceback (most recent call last):
  File "sg_pyvisa_simple.py", line 92, in <module>
    df_set_waveform(comb_signal, siggen_filename)
  File "sg_pyvisa_simple.py", line 75, in df_set_waveform
    inst_sg.write_binary_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str), datatype='h')
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 268, in write_binary_values
    block = util.to_ieee_block(values, datatype, is_big_endian)
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/util.py", line 383, in to_ieee_block
    return str(header) + struct.pack(fullfmt, *iterable)
struct.error: cannot convert argument to integer
```

I have tried declaring the datatype as "string" to match iq-str datatype,

```
inst_sg.write_binary_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str), datatype='s')
```

pyvisa complains as below this time, 

```
Traceback (most recent call last):
  File "sg_pyvisa_simple.py", line 92, in <module>
    df_set_waveform(comb_signal, siggen_filename)
  File "sg_pyvisa_simple.py", line 76, in df_set_waveform
    inst_sg.write_binary_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str), datatype='s')
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 268, in write_binary_values
    block = util.to_ieee_block(values, datatype, is_big_endian)
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/util.py", line 383, in to_ieee_block
    return str(header) + struct.pack(fullfmt, *iterable)
struct.error: pack expected 1 items for packing (got 2046)
```

I tried exploring sending the data stream by ASCII values as the iq_str variable consists of ASCII range(256) based characters,

```
inst_sg.write_ascii_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str))
```	
pyvisa have not spared me, complains the below, 

```
Traceback (most recent call last):
  File "sg_pyvisa_simple.py", line 92, in <module>
    df_set_waveform(comb_signal, siggen_filename)
  File "sg_pyvisa_simple.py", line 77, in df_set_waveform
    inst_sg.write_ascii_values(':MEM:DATA "WFM1:{}"'.format(filename.upper()), '#{}{}{}'.format(len(str(len(iq_str))), str(len(iq_str)), iq_str))
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 237, in write_ascii_values
    block = util.to_ascii_block(values, converter, separator)
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/util.py", line 202, in to_ascii_block
    return separator(converter % val for val in iterable)
  File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/util.py", line 202, in <genexpr>
    return separator(converter % val for val in iterable)
TypeError: a float is required
```

I would try to study this further while waiting for your opinion.

Thanks once again for your help.

MatthieuDartiailh, 2018-01-25:
First you do not need to format the string of data yourself you should simply pass the numpy array. The question is what should contain the array and what is the appropriate format.

svanan77, 2018-01-25:
In the manual, the data expected in a certain format as shown below

[image]
Refer: Page 230, http://literature.cdn.keysight.com/litweb/pdf/N5180-90074.pdf

Do you mean this format is automatically created when a write function is called?
I am new to pyVisa

MatthieuDartiailh, 2018-01-25:
The # header part is automatically created yes. You just need to find in the manual what values you are supposed to send to represent your complex waveform.

MatthieuDartiailh, 2018-01-25:
At least your message get through now.

svanan77, 2018-01-25:
Ok, the below works.

```inst_sg.write_binary_values(':MEM:DATA "WFM1:{}",'.format(filename.upper()), iq_str, datatype='c')```

Thanks Matthieu for your time.

closed 2018-01-25 by svanan77