pyvisa/pyvisa #310: ask vs. write command

aguileraGit, 2018-01-26:
I'm working on a script to capture 10 values from a Keithly 7510 DMM. If I use the write function, the values just appear on my computer screen and I can't save them. But they come back remarkably fast (which is something I need). If I change the write command to the ask command, I can save the data, however it takes about 1.2 seconds to get the data back. 

```python
print(inst.query('*RST'))
time.sleep(2)

inst.write(':TRACe:MAKE "voltMeasBuffer", 10000')
inst.write(':SENSe:FUNCtion "VOLTage"')
inst.write(':COUN 10')

timePast = datetime.datetime.now()

inst.write(':READ? "voltMeasBuffer"')

#Write vs. Ask
inst.write(':TRAC:DATA? 1, 10, "voltMeasBuffer"')
#vals = inst.ask(':TRAC:DATA? 1, 10, "voltMeasBuffer"')

timeNow = datetime.datetime.now()

print 'Time Diff: ', timeNow - timePast

#Time Diff:  0:00:00.000810 - Write
#Time Diff:  0:00:01.248488 - Ask
```
Also, with the ask command, there's an **Error 420 - Query Unterminated from USBTMC** on the DMM. I'm pretty confident that is causing the delay. I've simplified the code and just sent the DMM the read command using ask and write. Ask takes a second and throws up an error.

Any ideas? Thanks.

MatthieuDartiailh, 2018-01-26:
Ask (or query rather in recent versions) is a write folllowed by a read and should hence be used only when you expect an answer from the instrument.

aguileraGit, 2018-01-29:
@MatthieuDartiailh right. I use inst.query or inst.ask when I issue a READ? or TRAC:DATA? command to the DMM. I receive the data, but the DMM throws up an Error 420. It also delays the response from the DMM by ~1 second.

If I use inst.write command with READ? or TRAC:DATA?, the DMM doesn't show any errors. But I also can't save the data. I know the data is there, because I issue it inst.read() the data shows up on my computer. But again, the DMM throws up an Error 420 and delays the data by ~1 second.

In other words, sending it inst.read() causes an Error 420. And more importantly, it delays the data from getting back to the computer by ~1 second. 

Any ideas on how to avoid this error?

MatthieuDartiailh, 2018-01-29:
Have a look at the following document: download.tek.com/document/scpierrs.txt
Basically the error you see means that your instrument is not processing the command you sent it. This can be caused by an incorrectly set termination character (inst.write_termination = '\n'). You could ensure that your instrument does process your command by sending :DISPlay:LIGHt:STATe ON50 which should decrease the screen luminosity to 50%. Once this works try again reading.

aguileraGit, 2018-01-29:
So I followed your process. I was able to change the display intensity using inst.write with either '\r' or '\n' as the termination character. This didn't produce any errors on the DMM or python side.

I then tried to read the ID using inst.read('*IDN?'), inst.query('*IDN?'), and inst.ask('*IDN?') and inst.read_termination set to '\r', '\n', '\r\n', and '' as line ending. They all resulted in an Error 420 on the DMM. However, I did get a response using the query and ask commands. With the inst.read(), I saw the error UserWarning: read string doesn't end with termination characters regardless of the termination I used.

At this point, I think either the Library is sending something extra or the DMM isn't processing the data correctly. Is it possible to view the text sent to the DMM?

MatthieuDartiailh, 2018-01-29:
If you are on windows with NiVISA, one way is to use NiSpy that will record all communications. Otherwise if you use write_raw, PyVisa won't add anything to your message. Be careful that you need to encode your string message in this case (on Python 3).
By curiosity do you ever get a timeout error if you just read from the instrument (like read twice in a row).

aguileraGit, 2018-01-29:
I don't get any timeout errors even with two reads in a row.

I'm on Linux. I'll see if I can install NiSpy on a windows machine.

Again, the write is fine. The issue is with the read. No matter what I try, the read command always forces the error 420. Super frustrating.

MatthieuDartiailh, 2018-01-29:
What do you get out of the second read ? I mean the instrument buffer should be empty.

MatthieuDartiailh, 2018-01-29:
Also since you are on Linux, are you using pyvisa-py ? Asking because we know that the usb backend is not perfect so you may have it one of its bugs.

aguileraGit, 2018-01-30:
I'm pretty sure there's a bug. I started looking at the python-usbtmc library and switch to it. I get a different time out error when I try to read from the device. Out of luck (and the way I'm testing things using Jupyter), I open/closed the port with the py-visa library and then opened and ran commands with the usbtmc library. And I didn't see any errors - no timeout errors or 420 errors. No joke, the code below is working.
``` python
import visa
import usbtmc

rm = visa.ResourceManager('@py')
print rm.list_resources()

inst = rm.open_resource('USB0::1510::29968::04078027::0::INSTR')
inst.close()

inst = usbtmc.Instrument(1510, 29968)

timePast = datetime.datetime.now()

vals = inst.ask(':TRAC:DATA? 1, 10, "voltMeasBuffer"')
timeNow = datetime.datetime.now()

print type(vals), vals
print 'Time Diff: ', timeNow - timePast
```
Results:
```
<type 'unicode'> 1.298881E-01,1.298404E-01,1.298014E-01,1.298719E-01,1.297853E-01,1.297818E-01,1.296540E-01,1.298334E-01,1.299651E-01,1.297307E-01
Time Diff:  0:00:00.002787
```

This doesn't make sense to me. Maybe opening the device is setting some kind of config? I'm going to look it a bit longer. I hate not understanding why it works.

MatthieuDartiailh, 2018-01-30:
Try to see if somehow restarting the instrument allow you to reproduce the error. It is possible that python-usbtmc set a parameter that needs to be set only once and then we are good. From what I remember from python-usbtmc some settings are quite cryptic which is why I never took time to adapt them. I wish I add more time to spend on pyvisa-py, this backend could really use some fixes. If you figure it out please let me know.

aguileraGit, 2018-01-30:
I've tried making this code not work by rebooting, not using one library or the other, restarting the DMM. It really just comes down to opening/closing (visa lib) and then opening it with the usbtmc library. It always works. 

Can you point me to the function that sends the 'message' to the device? I'm looking at the Resource Class.
``` python
def write_raw(self, message):
        """Write a byte message to the device.

        :param message: the message to be sent.
        :type message: bytes
        :return: number of bytes written.
        :rtype: int
        """
        return self.visalib.write(self.session, message)
```
Where can I find the self.visalib.write function?

aguileraGit, 2018-01-30:
I could never find a way to see exactly what is being sent/received. So I used Wireshark to capture. I found two different things.

1. Line terminations. The usbtmc used two different line terminations. 0x0A for incoming text from the DMM. And no terminations for outgoing. I switched that on the pyvisa library. It didn't fix anything.

2. The configuration (when the port is opened) is different. There are probably 9 lines on Wireshark capture that are 'missing' when compared against the usbtmc capture. The problem is they are just hex characters. I don't have a way to make sense of them. I can send you those Wireshark packages if you want. That's about as much time as I can spend on this.

MatthieuDartiailh, 2018-01-31:
Yes please send those. Not sure when I will have time to look at this but it would be nice to have them anyway. Sorry for being unable to fix this issue.

MatthieuDartiailh, 2018-02-25:
Could you send me those Wireshark packages ?

clavay, 2018-04-10:
Hi,
I'm using pyvisa-py. (visa.ResourceManager('@py'))
I have a Keithley 2110 DMM and I have an error with a simple inst.query(':READ?')
It works with : 
inst.write(':READ?')
inst.read(termination='\n')

Is it ok for you ?

MatthieuDartiailh, 2018-04-10:
If you set `inst.read_termination = '\n'`, it should work with query.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh

bardgu, 2023-09-11:
Hi Matthieu, Im trying to understand what is read_termination = '\n' or write_termination = '\n' does. should I use it everytime I open resources?

thanks in adavance.