pyvisa/pyvisa #311: pyvisa-drivers repository proposal

dvincentwest, 2018-01-28:
I wonder if we could create a new pyvisa repository called pyvisa-drivers.  I have seen a number of attempts at creating standardized driver packages, with Lantz being the one most closely related to pyvisa.  As best I can tell they all fall short of critical mass in terms of traction, probably because instrument drivers are painfully complex to deal with.

I wonder if there is room for a different solution, namely a repository that hosts self-contained drivers that have no required relation to one another, other than that they use pyvisa.  Directory structure would look something like:

```
pyvisa-drivers/
    driverone.py
    drivertwo/  # this driver is a package
        drivertwo_fileone.py
        drivertwo_filetwo.py
    driverthree.py
```

By design you would want to take the consenting adults principle to the extreme here so that pyvisa maintainers would really have to spend very little time on this.  The main point is that pyvisa is visible and a pyvisa-drivers repo is just a visible place where pyvisa users post their drivers.  Then, when I run into the lab and need a quick driver (probably incomplete, but whatever, its something I can start with) maybe I have something.  With the goal being to get as many drivers in as possible, there would be little to no design philosophy other than each module / package should be self contained, that way users are encouraged to put up what they have rather than being put off by needing to conform their code to some standard.  The operative principle there being its better to have a mess of different drivers that I can maybe use and look at than it is to have nothing at all.

Thoughts?

MatthieuDartiailh, 2018-01-29:
The issue I see is that usually PyVISA is too low level to allow the implementation of drivers and usually people add an additional layer on top (such as Lantz) to overcome this limitation. On the other hand having examples, that people could run and in which basic functionalities of the instrument along with proper parametrization of pyvisa may prove helpful to a number of people.

However as the only currently active maintainer of pyvisa, I have no time to set this up and answer the questions that would inevitably pop up. Would you @dvincentwest be willing to manage such a repo ? If you are, I could create the repo and make you a team member of pyvisa (with at first access only to this repo).

dvincentwest, 2018-01-29:
I will maintain such a repo and answer questions as time allows. Ultimately if things go in a strange direction it can always be put on ice or disconnected from the project if need be.

DavidLutton, 2018-01-29:
I may have started something like this proposal intends

LabToolkit [usage examples](http://labtoolkit.readthedocs.io/en/latest/usage.html) where classes that perform as drivers are passed open resources and provide functions generally @property 's to make getters and setters with the same name like `SignalGenerator.frequency = 1e9` or `print(SignalGenerator.frequency)`  >>> 1000000000.0

The simplest driver is for an [ElectronicAttenuator](https://github.com/DavidLutton/LabToolkit/blob/trypip/labtoolkit/ElectronicAttenuator.py) this also uses the IEEE488 class to provide IEEE488 commands as functions

An example that uses class inheritance to handle model capability (Frequency Range) is class MarconiInstruments2031 and the parent class MarconiInstruments203N in [SignalGenerator](https://github.com/DavidLutton/LabToolkit/blob/trypip/labtoolkit/SignalGenerator.py)
A not as yet implemented idea is to have the class warn you that the hardware cannot operate out of its frequency range

I find the idea that you can invoke classes/hardware with out the library **magicaly** finding you hardware. As an example, HP 85nn spectrum analysers that only respond with their model with ID? and not at all to *IDN?

For example [Voltnuts logging topic on EEVBlog](http://www.eevblog.com/forum/metrology/raspberry-pi23-logging-platform-for-voltnuts/350/) have many [scripts](https://github.com/PlesaEEVBlog/RPi_LogNut/tree/master/python) that provide combinations of interfaces and drivers and modes. Where PyVISA and LabToolkit could simplify/manage the first two and help with the modes/measurements

[KE5FX](http://www.ke5fx.com/gpib/readme.htm) set of tools and audio analyser frequency response [hp8903](https://github.com/cosmonaut/hp8903). These tools are interesting but I think that Matplotlib or similar can produce better figures and also we can do things that that software does not support, like to get a frequency response as a single figure that combines multiple frequency ranges. re network analysers with limited points

The LabToolkit SignalGenerator, SpectrumAnalyser, NetworkAnalyser, PowerMeter, WaveformGenerator, ElectronicAttenuator and some of the DigitalMultimeter are working classes for what I needed so far. Most of the other classes are there to put the concept out there

LabToolkit works on Python 3.4 onwards on Windows with GPIB or USB devices with either Keysight or National Instruments VISA drivers and once when I had it working with Linux with GPIB-Linux, without issues relating to PyVISA

There are many things that we can control and use via PyVISA. I am interested in helping out with this, however we go about it

MatthieuDartiailh, 2018-01-30:
@DavidLutton this is an interesting idea and somehow match what we attempted at one point to do in LabPy, the sad story is that it failed. Implementing robust and generic driver is a complex problem (as I am sure you know) and can be tackled in many different ways. I have for example my own library (https://github.com/Ecpy/i3py) that actually does not implement yet any concrete driver because simply figuring out the tooling is a mess (I invite you to look at the docs (docs folder) as it may resonate with some of you ideas).
This is typically not something I would like to see in pyvisa-drivers because the maintenance burden is just much higher. However I see the interest in providing simple self contained pyvisa examples for specific instruments that can be used by people as basic. In addition to this maintaining a list of packages implementing drivers would be also interesting.

dvincentwest, 2018-01-30:
Indeed implementing a system of drivers is complex and what makes matters worse is that contributing drivers to such a system requires a fair bit of learning. For most of us we only need access to a handful of instruments and usually only a subset of features so the time investment in learning the system of drivers may not feel like it’s worth the payoff. But most of us need something so we write something simple to get done what we need. If we could have a collection of all of those things, I think it would be interesting to see how far it goes. 

As an aside the list of attempts at a driver system is enlighting. Aside from lantz, you have python-ivi and I think PyMeasure is still going. There is a python scpi module and a few others I  forgetting at the moment. Scikit-RF implements some network analyzer drivers that I contributed to last year, specifically a system of writing low level scpi commands in a consistent manner using a yaml parser. I think the reality is that the only way to build a large library of consistent drivers is to be National Instruments so maybe we should try to do something different, less ambitious in features in the hopes that people will post what they have. 

I don’t mind maintaining the repo as I envision it would be screening commits for malicious content and answering the few questions that I would be able to but otherwise would be a you’re mostly on your own kind of package

MatthieuDartiailh, 2018-02-04:
I created the repo and invited you as collaborator. I would suggest you start by adding a README describing clearly the intent of this repo. Also please go through pull request to update master. If I don't comment in a week, feel free to merge the changes you propose.

dvincentwest, 2018-02-05:
the README has been added, and as requested, I will submit code through
pull requests.  When I get a chance later this week I will upload some
drivers that I have made

MatthieuDartiailh, 2018-02-05:
Looks good !

closed 2018-02-05 by MatthieuDartiailh