pyvisa/pyvisa #312: Error during write_binary_values()

svanan77, 2018-01-30:
Hi All,

I am trying to 
a) connect to a signal generator (Agilent N5182A MXG) that supports ASCII range(256) using TCPIP. 
b) set a signal waveform to the signal generator (siggen) using IQ data
c) select the signal waveform from the siggen.

Step a) and b) in the above are successful. The _StatusCode Success: 0_ returned. Unfortunately the siggen is complaining when step c) was executed, as below,

[image]

At the pyVisa side _StatusCode Success: 0_ was returned.

According to the below link, error 230 indicates a new reading has started but not completed, pointing to possible incorrect header formation.
Refer: http://na.support.keysight.com/pna/help/latest/Support/SCPI_Errors.htm

No timeout occurred during Step b). Nevertheless increased the timeout to 1000ms, just to be on safe side.

At the siggen, I have validated the number of points, 510, and found it is accurate (i.e 2040/4 = 510). 4 bytes are used per IQ data and 2040 bytes of data string, _iq_str_ is transferred to the siggen.

[image]

Have manually also validated the content of the transmitted _iq_str_ variable. All seems good. 

Have anyone faced this problem?

The full code is provided as below and the std_out of some of the important variables are shown as comments including the data string of _iq_str_ variable. 2 bytes seems unaccounted, maybe additional white space in the syntax. Not sure of this could be a cause. 

Hope to get your opinion on this. 
Though strange, I am sure it must have a reasonable answer. Keen to understand the reason for this.

```
import visa
import numpy
import itertools
import pylab as pl
import math
import os

#######################################
# Variables
#######################################

IP_SG = '10.10.100.131'
siggen_filename ='myCombSig'
num_tone=16
num_smpl_tone=32 

#######################################
# Initialization
#######################################

rm = visa.ResourceManager('@py')

#######################################
# Helper Functions
#######################################

def df_open_connection(rm, ip):
	inst = rm.open_resource('TCPIP::'+ ip + '::INSTR')
	return inst

def df_close_connection(inst):
	inst.close()

def df_my_range(start, end, step):
	while start <= end:
		yield start
		start += step

def df_Comb_sig_gen(num_tone, num_smpl_tone):
	comb_real = []
	comb_imag = []
	comb_cmpx = []
	M = num_tone * num_smpl_tone
	N = 0
	C = num_tone
	k = pl.frange(float(-(C - 1)) / 2, float((C - 1)) / 2, 1)
	for n in df_my_range(-N + 1, M + N - 2, 1):
		cc = numpy.exp(2j * math.pi * k * numpy.array(n) / (C))
		comb_cmpx.append(numpy.sum(cc))  

	comb_cmpx = numpy.array(comb_cmpx)

	return (comb_cmpx)

def df_set_waveform(comb_signal, filename):

	scale  = numpy.amax([numpy.amax(comb_signal.real),numpy.amax(comb_signal.imag)])

	comb_signal_real_scale = numpy.array(comb_signal.real)/scale
	comb_signal_imag_scale = numpy.array(comb_signal.imag)/scale

	i = comb_signal_real_scale
	q = comb_signal_imag_scale

	i_dac = map(lambda x: int(32767 * abs(x)), i)
	q_dac = map(lambda x: int(32767 * abs(x)), q)

	#Big Endian: bit masking and shifting. 
	iq_str = ""
	for i_val, q_val in itertools.izip(i_dac, q_dac):
		iq_str += chr((i_val >> 8) & 0x00FF) #e.g 32767 (0x7FFF) -> 7F
		iq_str += chr(i_val & 0x00FF)		 #e.g 32767 (0x7FFF) -> FF
		iq_str += chr((q_val >> 8) & 0x00FF)
		iq_str += chr(q_val & 0x00FF)

	scpi_command = ':MEM:DATA "WFM1:{}",'.format(filename.upper())
	
	return scpi_command, iq_str

def df_timeout(inst):
	delay = 10000 	# in milisecond
	inst.timeout = delay

def df_select_waveform(filename):
	"""
	Selects a waveform file in the signal generator's volatile memory.
	After selecting a waveform file, it can be played by calling enable_waveform().
	"""

	scpi_command = ':RAD:ARB:WAV "WFM1:{}"\n'.format(os.path.basename(filename).upper())

	return scpi_command

#################################################
# Action: To Set a Waveform on Signal Generator
#################################################

inst_sg = df_open_connection(rm, IP_SG)
inst_sg.write(':FREQ 1.8825 GHz; POWer -30DBM')
df_timeout(inst_sg)

comb_signal = df_Comb_sig_gen(float(num_tone),float(num_smpl_tone))	

set_waveform, iq_str = df_set_waveform(comb_signal, siggen_filename)
select_waveform = df_select_waveform(siggen_filename) #Data

df_timeout(inst_sg)

response_write = inst_sg.write_binary_values(set_waveform, iq_str, datatype='c')
#> response_write -> (2075, <StatusCode.success: 0>)

#> Format -> MMEM:DATA “WFM1:my_file”,#3 240 12%S!4&07#8g*Y9@7...
#> len(set_waveform) -> 27
#> len('#') -> 1
#> len('4') -> 1
#> len('2040') -> 4
#> len(iq_str) -> 2040
#> 2075 - (27+1+1+4+2040) = 2 bytes unaccounted for. Maybe used to parse 'datatype', 'is_big_endian', 'termination' and 'encoding' default parameters to Signal Generator

# The complete iq_str's ASCII data is given below. 
#> iq_str=                                                             ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                              ÿ                                                          

response_select = inst_sg.write(select_waveform)
#> response_select -> (32, <StatusCode.success: 0>)

df_close_connection(inst_sg)

```

svanan77, 2018-01-31:
I have validated the additional 2 bytes are due to the termination characters (CR and LF) inserted to the SCPI command. Still trying to understand how the Header could be not recognized.

bjaraujo, 2018-02-02:
Hi,

I think pyVisa has some issue with reading/writing binary data and termination. 

In your case specifically I don't think you should be sending CRLF and the "reader" of this data should not "look" at termination characters either. The number of bytes is specified in the header so adding termination will only confuse the reader since it will stop when it sees a termination character. So might stop in the middle of binary data.

MatthieuDartiailh, 2018-02-04:
There is nothing pointing clearly to the termination character being the issue here, even though it may be. One thing you can easily try is to simply query the waveform you just set to check you get the proper values.
If you can, try to use the ni backend (which most likely mean you need to access a windows machine). The pure python backend still has some bugs. 
In any case, I believe you could simplify your code by letting the python do all the packing. To do so, you could create a numpy array with you interlaced 16-bit integers and then passing it to write_binary_values as follow:
```python
data = np.array(2*len(i_dac)
data[::2] = i_dac
data[1::2] = q_dac
inst_sg.write_binary_values(set_waveform, data, datatype='H', is_big_endian=True)
```
I am also surprised by the way you build your data. Why use abs ? Usually one shifts the data so that they all become positive and normalize them afterwards. Here your negative values become positive which mean that you distort your waveform, and I doubt it is what you want.

svanan77, 2018-02-08:
Hi,

Sorry for the late reply as I have been away. Thanks for your time. I will try to provide as complete information as possible.

@bjaraujo -- Referring to my code above, I did not send CR and LF termination characters in my code. When I validated the below line, the response returned 2075 bytes being sent. It had 2 byte unaccounted for which I later learned that pyVisa adds them automatically. Now all 2075 bytes are accounted for.

    response_write = inst_sg.write_binary_values(set_waveform, iq_str, datatype='c')
    #> response_write -> (2075, <StatusCode.success: 0>)

 @MatthieuDartiailh -- This could be a reason, let me elaborate in 3 parts.

a) I am keen to generate a 16 tone ideal flat comb signal where I feed this to the signal generator. 

    i_dac = map(lambda x: int(32767 * x), i)
    q_dac = map(lambda x: int(32767 * x), q)

The above creates the below signal stream

[image]

As can be seen in my data stream below,

    i_dac = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    q_dac = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

As I was looking for ideal flat comb signal with only positive tones, I included `abs()` as below,

    i_dac = map(lambda x: int(32767 * abs(x)), i)
    q_dac = map(lambda x: int(32767 * abs(x)), q)

The above creates the below signal stream,

[image]

As can be seen in my data stream below,

    i_dac = [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32766, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0,  [... cut ...]
    % Read back and display any instrument errors.
    instrumentError = query(signalGeneratorObject,'SYST:ERR?');
    while isempty(strfind(lower(instrumentError),'no error'))
        fprintf('\tINSTRUMENT ERROR: %s', instrumentError)
        instrumentError = query(signalGeneratorObject,'SYST:ERR?');
    end

    %%%%%%%%%%%%%%%%%%%%%Clean Up
    % Close the connection to the instrument and delete the object.
    fclose(signalGeneratorObject); delete(signalGeneratorObject);
    clear signalGeneratorObject`

MatthieuDartiailh, 2018-02-08:
I am not sure what is going there. Looking at the documentation from keysight I get the proper formatting for the values using PyVISA commands. Could you print the binary block before sending using both Matlab and Python (for both versions) (you can use the to_ieee_block from pyvisa.util to build the string), I am curious to see what differs between them.

svanan77, 2018-02-08:
I shall update you tomorrow when I get slot to test this. Thanks for your time.
I have another basic query which I will post separately to not mix with this issue. Just to learn your expertise.

MatthieuDartiailh, 2018-02-12:
Any news @svanan77 ?

svanan77, 2018-02-13:
Hi @MatthieuDartiailh -- I managed to the slot to test this. I have confirmed it is a hardware issue and not of pyVisa. As when I use a different siggen of same make and model the same script works. I believe there are some issue with the volatile memory location of the siggen. Have forwarded this finding to the team member who is responsible to check further on this. Will close this.

closed 2018-02-13 by svanan77