pyvisa/pyvisa #314: Challenges in query_binary_values()

svanan77, 2018-02-09:
Hi,

I am trying to retrieve IQ Data using pyVISA from N9010A Agilent EXA Signal Analyzer where I bumped into this situation. Have done comparison using Matlab to isolate its not a Signal Analyzer/hardware problem.

###    A. Using Matlab

The code:

    % Read the IQ data from Signal Analyzer
    fprintf(sigAnObject,':READ:WAV0?');
    data = binblockread(sigAnObject,'float');

    % Obtain inphase and quadrature component of the data
    inphase = data(1:2:end);
    quadrature = data(2:2:end);

The inphase and quadrature data (1st 30 samples):   

inphase | quadrature
-- | --
0.011705055 | 0.004768387
0.008743308 | 0.009115119
0.004254719 | 0.011908142
-0.000960059 | 0.012577926
-0.006008605 | 0.011135262
-0.010015562 | 0.007707035
-0.012288122 | 0.002975783
-0.01242462 | -0.002248236
-0.010483825 | -0.007151135
-0.00668062 | -0.010754864
-0.0017099 | -0.012506674
0.003542596 | -0.012129868
0.008226611 | -0.009639439
0.011367292 | -0.005496852
0.012626496 | -0.00036012
0.011727759 | 0.004814995
0.008736435 | 0.009154395
0.004256689 | 0.011886586
-0.000967312 | 0.012630761
-0.005985226 | 0.01113914
-0.009944156 | 0.007769966
-0.012286111 | 0.003028868
-0.012423206 | -0.002289723
-0.010420014 | -0.007153958
-0.006667234 | -0.01077261
-0.001754645 | -0.012473318
0.003515914 | -0.012151629
0.008169005 | -0.009651847
0.011414683 | -0.005466485

The charts:

[image]

[image]

The conclusion:
The results are as intended.

###    A. Using pyVisa

The code:

    #Read the IQ data from Signal Analyzer
    data =sigAnObject.query_binary_values(':READ:WAV0?', datatype='f')

    # Obtain inphase and quadrature component of the data
    inphase = data[0::2]
    quadrature = data[1::2]

The inphase and quadrature data (1st 30 samples):

inphase | quadrature
-- | --
-2.93E+17 | -3.96E-28
-1.79E-35 | 1.21E-19
-0.000383621 | 1.37E+17
-6.07E-38 | 2.33E-23
-1.28E-25 | -1352.054077
5.48E-13 | -1.02E+12
-1631.085327 | -1.75E-19
-4.04E-12 | -9.02E+12
1.38E-26 | -1.11E-32
4.52E+35 | -2.01E-21
-101486.4688 | 3.84E+14
-5.54E-17 | 1.18E+11
5.57034111 | -6.41E+12
7.26E+14 | 1.08E-24
1.58E-08 | 7.58E-06
1.18E-22 | nan
1.48E-15 | 9.46E+35
-0.000263458 | -1.94E-25
4.73E+27 | 5.48E+29
3.37E-10 | 1.25E-17
-8.11E+25 | 4.91E+17
1.26E-07 | 4.14E-22
3.38E+24 | -3.41E-35
-2.72E-20 | -8.32E+13
-2.56E-30 | -1.91E-25
7.28E+27 | -1.56E-19
2.71E+36 | 6.30E-13
-3.71E-09 | 2.11E-17
4.71E+29 | -1.69E+36

The charts:

[image]

[image]

The conclusion:
Though the same number of samples (n=12001) are retrieved from the Signal Analyzer, its values are weird. Though we are not expecting the exact same values as those produced by Matlab, it is expected to behave the same as in the Matlab charts.

Have anyone faced this problem? 
I am still studying if the datatype could be a cause.

MatthieuDartiailh, 2018-02-10:
It may be a big endian vs little endian issue. PyVisa may not use the same default as Matlab. Try adding is_big_endian=True to your call.

svanan77, 2018-02-13:
@MatthieuDartiailh -- I have tested your proposal. Yes. Confirm that is the reason. The default Matlab and pyVisa using different endness during binary read . Will close this.

closed 2018-02-13 by svanan77