pyvisa/pyvisa #315: Problem with binary communication

SengerM, 2018-02-14:
Hello,
I am using PyVISA to communicate with an HP 3458A multi-meter (and some other instruments). I am on a 
Windows 7 machine using Python 3.6.3, PyVISA 1.8 and the NI backend. 

Currently everything is working fine in ASCII format, using this line `float(HP3458A.query('RMEM {:d}'.format(k+1))` to get data from the voltmeter (the command `RMEM i` returns the value stored in position `i` of memory in ASCII or binary format, depending on previous configuration of the instrument). 

The issue arises when I try to read the memory in binary format (in order to speed things up). I have tried many solutions, none of them working. I have previously configured the instrument in order to answer in binary format (in particular I have used this: `HP3458A.write('OFORMAT SINT')` that, according to [the manual of the HP3458A](https://xdevs.com/doc/HP_Agilent_Keysight/3458A/03458-90014.pdf), configures it in order to send 2 bytes integers). After this configuration I have tried everything that is said [here](http://pyvisa.readthedocs.io/en/stable/rvalues.html#reading-binary-values), expecting an answer of two bytes. In particular I have tried the following two situations:

-  `values = inst.query_binary_values('RMEM 1', datatype='h')` with all kind of combinations in the arguments, without success. 
- `inst.write('RMEM 1')` and then `data = inst.read_raw()` or `inst.read_raw(1)`, `inst.read_raw(2)`, etc.

In all cases I am getting the error:
`Traceback (most recent call last):
  File "3458A sub-sampling test.py", line 48, in <module>
    print(HP3458A.read_raw())
  File "C:\Users\...\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.
py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\...\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py"
, line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\...\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py"
, line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before op
eration completed.`

On the other hand, if I open NI Max and then go to the "VISA test panel", set "bytes to read" to 2 and send the command `RMEM 1`, I get the expected answer from the instrument. I mean, this works:

[image]

The fact that on the NI Max this works makes me believe that this is a problem on the Python side and that the instrument is properly configured.

Any help will be appreciated.

Thanks in advance,
Matías.

MatthieuDartiailh, 2018-02-14:
Hi, 
Could you try reading more than two bytes using NI-VISA ? I would expect to see the termination character in the instrument response.

SengerM, 2018-02-15:
Hello,
If I try to read more than two bytes using NI-VISA I get the following:

[image]

If I query two bytes I get `\FEO` without any error.

According to the [HP3458A manual](https://xdevs.com/doc/HP_Agilent_Keysight/3458A/03458-90014.pdf) in page 210 we have this (I am using SINT):

[image]

so I understand that there is no termination char, instead I have to read 2 bytes from the buffer.

MatthieuDartiailh, 2018-02-15:
Thanks for testing. It is indeed unusual for instrument to not use a termination character but I guess that for an old model such as yours it can actually happen. At the time being there is no super easy way to go around this issue. However #301 introduce a read_bytes function that would do, what you were expecting read_raw to do. I was hoping to get some review on it before merging but as it worked well when I tested it, I will probably merge next week and release a new version hopefully shortly after. If you can try the branch or copy and paste the relevant part of the code for quick testing. Note however that the standard query_binary_values won't work in your case some you will have to implement the parsing of the message yourself.

SengerM, 2018-02-15:
Thank you for your replies. Do you know why `read_raw` is not working in this case? Isn't it supposed to do just what I want to do? I mean, read a fixed quantity of bytes.

MatthieuDartiailh, 2018-02-15:
No read_raw read bytes and does not decode them. But the detection of the end of message is the usual one. The size arguments simply set the chunz_size used for reading. I clarified this in the documentation in #301 and it is why I added read_bytes.

MatthieuDartiailh, 2018-02-25:
PyVISA 1.9 is out and implement `read_bytes` which fixes this issue.

closed 2018-02-25 by MatthieuDartiailh

SengerM, 2018-02-26:
Thank you very much, the problem was solved and the communication with the HP 3458A is working really good and fast!

MatthieuDartiailh, 2018-02-26:
Glad to hear it !