pyvisa/pyvisa #316: Query IQ Data from Signal Generator

svanan77, 2018-02-16:
Hi,

I would like to pull IQ Data from the Signal Generator (Agilent Technologies, N5182A, MY50140960, A.01.85).

According to the manual (as below), I can query using `:MEMory:DATA:IQ? "<file_name>"` syntax.

[image]

Have written the below code to pull the IQ of `waveform_file` already in the Signal Generator.

    IP_SG = '10.10.100.131'
    rm = visa.ResourceManager('@py')
    inst = rm.open_resource('TCPIP::'+ IP_SG + '::INSTR')
    data = inst.query_binary_values(:MEMory:DATA:IQ? "waveform_file")

Unfortunately getting timeout error, eventhough the timeout is 30s currently.

[image]

As the output is not entirely float, I dont think it require for a `datatype` or `is_big_endian` to be specificied in the ` inst.query_binary_values()`. I even tried  ` inst.query()` which results in the same timeout error.

Do you happen to face the same problem before?

Saravanan K

MatthieuDartiailh, 2018-02-16:
I note you are using the pyvisa-py backend. I identified recently a bug in the handling of the termination character for TCPIP. I fixed it in https://github.com/pyvisa/pyvisa-py/pull/126 but I did not merge yet. Could you try installing that branch ? Also can you retrieve any data from your instrument such as *IDN? ?

svanan77, 2018-02-16:
Hi @MatthieuDartiailh,

Yes I am using pyvisa-py backend. I installed it using `pip install pyvisa-py` (not `git clone`).

I went through #126, perhaps that could be the case. 

Would be glad to test the branch. Unfortunately not familiar on how. Could you please list out the steps I need to do to test the branch?

If I'd installed pyvisa-py via `git clone`, I would have `git pull` it. I unable to do it that way for the below reason,

    svanan@svds07:~/ubuntu/pyvisa/pyvisa-py-master$ git status
    fatal: Not a git repository (or any parent up to mount point /home/svanan)
    Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

Otherwise, 

    print("Session ID : %s" % (inst.session))
    print("Instrument identification information : %s" % (inst.query("*IDN?").rstrip()))

gives the below output, will get the same output again after installing the new patch from #126.

    Session ID : 5097654
    Instrument identification information	: Agilent Technologies, N5182A, MY50140960, A.01.85

MatthieuDartiailh, 2018-02-16:
To instal you can first do `pip uninstall pyvisa-py` and then `pip install https://github.com/MatthieuDartiailh/pyvisa-py/tarball/fixes`.

svanan77, 2018-02-16:
Hi @MatthieuDartiailh,

I have done the validation and would present it in 2 parts, a) the preparation, b) the test

** A. The Preparation**

Step 1:

    svanan@svds07:~/ubuntu/pyvisa$ python -m visa info
    Machine Details:
      Platform ID:    Linux-4.4.0-92-generic-x86_64-with-debian-stretch-sid
      Processor:      x86_64

    Python:
      Implementation: CPython
      Executable:     /opt/anaconda2/bin/python
      Version:        2.7.12
      Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
      Bits:           64bit
      Build:          Jul  2 2016 17:42:40 (#default)
      Unicode:        UCS4

    PyVISA Version: 1.8

    Backends:
      ni:
         Version: 1.8 (bundled with PyVISA)
         #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
            found by: auto
            bitness: 64
            Could not get more info:
               VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
      py:
         Version: 0.3.dev0
         ASRL INSTR: Available via PySerial (3.4)
         TCPIP INSTR: Available
         USB RAW: Available via PyUSB (1.0.2). Backend: N/A
         USB INSTR: Available via PyUSB (1.0.2). Backend: N/A
         GPIB INSTR:
            Please install linux-gpib to use this resource type.
            No module named gpib
         TCPIP SOCKET: Available
      sim:
         Version: 0.3
         Spec version: 1.1

Step 2:

    svanan@svds07:~/ubuntu/pyvisa$ pip uninstall pyvisa-py
    Uninstalling PyVISA-py-0.3.dev0:
     /opt/anaconda2/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg-info
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/__init__.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/__init__.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/common.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/common.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/gpib.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/gpib.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/__init__.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/__init__.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbraw.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbraw.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbutil.py
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/usbutil.pyc
     /opt/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/vxi11.py
[... 114 lines cut ...]
    Session ID : 1652253
    Instrument identification information : Agilent Technologies, N5182A, MY50140960, A.01.85

The same timeout problem
[image]

Step 2. Executed the  below using inst.query()

    IP_SG = '10.10.100.131'
    rm = visa.ResourceManager('@py')
    inst = rm.open_resource('TCPIP::'+ IP_SG + '::INSTR')
    inst.timeout = 30000
    print("Session ID : %s" % (inst.session))
    print("Instrument identification information : %s" % (inst.query("*IDN?").rstrip()))
    data = inst.query(:MEMory:DATA:IQ? "waveform_file")

Outcome

    Session ID :  8834162
    Instrument identification information : Agilent Technologies, N5182A, MY50140960, A.01.85

The same timeout problem
[image]

Do you need me to do more tests? Meanwhile I will also to continue investigate from my side.

MatthieuDartiailh, 2018-02-25:
Sorry for the delay, you can try using the latest Pyvisa version to use read_bytes(1) to try reading just a single byte. If this times out I would guess that your instrument either need more time to answer or did not understand your message.

svanan77, 2018-02-27:
Hi @MatthieuDartiailh ,

I managed to get slot to user the signal generator. 

1. I have uninstalled _pyvisa_ and _pyvisa-py_, then reinstalled both to get the latest version.  I am now using  `PyVISA Version: 1.9.0` .The `python -m visa info` output is below,

[image]

2. I have,
a) rerun my original script (as below) to connect to the signal generator to validate all my setup and script are fine. It has executed well at terminal

    IP_SG = '10.10.100.131'
    rm = visa.ResourceManager('@py')
    inst = rm.open_resource('TCPIP::'+ IP_SG + '::INSTR')
    inst.timeout = 10000
    print("\nConnecting to .... %s" % (IP_SG))
    print("Session ID : %s" % (inst.session))
    print("Instrument identification information : %s" % (inst.query("*IDN?").rstrip()))

b) added 	`read_bytes(1)` and my new script looks below,	

    IP_SG = '10.10.100.131'
    rm = visa.ResourceManager('@py')
    inst = rm.open_resource('TCPIP::'+ IP_SG + '::INSTR')
    inst.timeout = 10000
    print("\nConnecting to .... %s" % (IP_SG))
    print("Session ID : %s" % (inst.session))
    print("Instrument identification information : %s" % (inst.query("*IDN?").rstrip()))
    print("SK - Before excuting read_bytes(1)")
    data = self.conn.read_bytes(1)
    print("\nData %s" % (data))
    print ("SK - After excuting read_bytes(1)")

Despite the 10s timeout the excution somehow stopped as despite waiting for 12 minutes, it is still ongoing. The output of both are below,

[image]

I went ahead to get the `htop` output, as below
[image]

The status is SLEEP, does this mean it is awaiting for a response from the signal generator? This is the same despite 3 times excuting the script.

MatthieuDartiailh, 2018-02-27:
This is weird. You are using an old version of PyVISA-py.  First think you could do is update to the latest master from github, which includes some improvement as far as timeout are concerned. However to me it looks like either your instrument is very slow (in which case I would suggest to try to sleep manually before calling read bytes), or your command is somehow wrong. To test this you can check the error queue either manually if the instrument has a front panel or querying 'SYST:ERR?' if your instrument supports it (this is a standard SCPI command).

svanan77, 2018-02-27:
@MatthieuDartiailh 
a) I think I am using the latest version of pyVisa-py. As pyVisa, I had uninstalled pyVisa-py and reinstalled it again using `pip install pyvisa-py`. Is there a way to pull the version info as visa? `python -m pyvisa-py info` did not work. If yes do share it with me, I will provide the info to you.

b) Yes my instrument supports `SYST:ERR?`. There we not errors in the queue, as shown below,
[image]

Will investigate further.

MatthieuDartiailh, 2018-02-27:
You can install a yet unpublished version directly from github using:
```shell
pip install https://github.com/pyvisa/pyvisa-py/tarball/master
```
A lot has changed since 0.2 !

svanan77, 2018-02-27:
Sorry my bad.  I have installed from the master and am using the below version

`Version: 0.3.dev0`

Now the output is returning time out for read_bytes(1).

[image]

Perhaps the instrument dont understand the syntax.

MatthieuDartiailh, 2018-02-27:
But after sending the command you don't have any error from the instrument ? That is weird. May be worth posting it on the Keysight forum.

svanan77, 2018-02-27:
@MatthieuDartiailh 

1. I confirm there were no error at the siggen after the timeout due to instr.read_bytes(1) . Will raise this with Keysight 

2. I have changed ':MEM:DATA:IQ? "filename"' to ':MEM:DATA? "filename"' under the advice of Keysight expert. Manage to read the IQ data now though both is usuable syntaxes in the same manual. Would continue to discuss with them why the first one ought to fail when I am using the exact siggen supported by that syntax.

Can I go ahead to close this?

MatthieuDartiailh, 2018-02-27:
If everything works as expected with the other command I would say yes, since the issue appear to be specific to your instrument and command and not directly related to PyVISA or PyVISA-py.

closed 2018-02-27 by svanan77