pyvisa/pyvisa #319: Broken query_binary_values()

svanan77, 2018-02-27:
I have upgraded PyVISA Version: 1.8 to PyVISA Version: 1.9 today. The below binary read syntax from spectrum analyzer seems to fail.

data = self.conn.query_binary_values(':READ:WAV0?', datatype='f', is_big_endian=True)

[image]

It has been used many many times before without fail. 

Would the above syntax need to be parsed differently in the newer version of PyVISA? I am astill  using PyVISA Documentation, Release 1.9.dev0, 22 Aug 2016.

MatthieuDartiailh, 2018-02-27:
The error is occurring inside PyVISA-py  and is not related to the binary read at least it does not seem to be.  Could clone PyVISA-py, install it in develop mode and try to find at which commit the issue appears (I would guess that the last change I merged tonight may have caused it) ? Sorry about that.

MatthieuDartiailh, 2018-02-27:
I will try to run some more tests tomorrow on this. I am sorry I broke your code.

svanan77, 2018-02-27:
No bro. We are good.  Lets do tomorrow. Meanwhile if you like me to do any test, do inform. Would be glad to help. Its a plus if the steps could be given :-)

MatthieuDartiailh, 2018-02-27:
Are you familiar with using git ? or not at all ? 
If not try uninstalling Pyvisa-py and download the zip archive from https://github.com/pyvisa/pyvisa-py/archive/e91ce7ce81cadf0ab6eaaac11144257c6356a788.zip. Unzip it and then open a command line inside the unzipped folder and run python setup.py develop.
If this fix the issue it means my last changes broke something baddly and that will give me some idea where to look at.

svanan77, 2018-02-27:
@MatthieuDartiailh  Yes I am familiar. Will update you tomorrow with my findings. Have a rest

MatthieuDartiailh, 2018-02-28:
Testing on a Keysight PNA, I have no issue trasferring binary data. So the best you can do is try to bisect the offending git commit in pyvisa-py. I will try to run some additional tests on some other instruments but probably not today.

MatthieuDartiailh, 2018-03-11:
Any news on this @svanan77 ? As a lot has been improved recently I would like to make a new release soon. However, I would prefer to fix your issue before the release, but as I cannot reproduce I need some feedback from you.

svanan77, 2018-03-15:
@MatthieuDartiailh Sorry I was away. Back now. Will get this tested ASAP, awaiting slot on Spec Analyzer and revert soonest.

svanan77, 2018-03-19:
I have not got slot to the instrument as yet. Please bare with me.

MatthieuDartiailh, 2018-03-19:
No problem.

svanan77, 2018-03-21:
Hi @MatthieuDartiailh 

I cloned PyVISA-py on Monday (19 March 2018), install it in develop mode.

When I run my script today (21 March 2018), I still get the below error as it was before.

[image]

Pointing to `data = self.conn.query_binary_values(':READ:WAV0?', datatype='f', is_big_endian=True)`

svanan77, 2018-03-21:
@MatthieuDartiailh Ok will work on that and revert

svanan77, 2018-03-21:
@MatthieuDartiailh  Got it!

The culprit `[84e196634386cc4003da9a8f257de4a4832bf69f] tcpip: INSTR proper Visa timeout error on timeout`

[image]

MatthieuDartiailh, 2018-03-21:
Ok thanks for checking. I will try to understand what caused the regression. If you have more time feel free to experiment more.

MatthieuDartiailh, 2018-03-28:
I may have a fix please see https://github.com/pyvisa/pyvisa-py/pull/130

MatthieuDartiailh, 2018-03-30:
@svanan77 Could you please confirm the fix in https://github.com/pyvisa/pyvisa-py/pull/130 does correct you issue ?

svanan77, 2018-04-02:
@MatthieuDartiailh  Hi, Sorry we were on Easter Holidays.

Could inform if the below testing method is accurate.

I tried to checkout the commit that potentially fixed the problem but bumped into the below problem.

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git checkout ed044879027bcf34b8a408294679994be3f773bc .
    fatal: reference is not a tree: ed044879027bcf34b8a408294679994be3f773bc
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git reset --hard;
    HEAD is now at 5106ee1 Merge #127
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git pull
    Already up-to-date.

Believe it has not been merged yet hence unable to do a `git pull` too. Could you suggest the right way to test this commit?

Thanks

MatthieuDartiailh, 2018-04-02:
You can add my personal repo as remote , and switch to the fix_rpc_timeout branch:
```shell
git remote add mdartiailh https://github.com/MatthieuDartiailh/pyvisa-py.git
git fetch mdartiailh
git checkout -b fix_rpc_timeout mdartiailh/fix_rpc_timeout
```

I usually use a graphical git client so some commands may be slightly wrong.

svanan77, 2018-04-02:
The syntaxes are accurate. Do refer the output as below.

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git remote add mdartiailh https://github.com/MatthieuDartiailh/pyvisa-py.git
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git remote -v
    mdartiailh      https://github.com/MatthieuDartiailh/pyvisa-py.git (fetch)
    mdartiailh      https://github.com/MatthieuDartiailh/pyvisa-py.git (push)
    origin  https://github.com/pyvisa/pyvisa-py.git (fetch)
    origin  https://github.com/pyvisa/pyvisa-py.git (push)
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git fetch mdartiailh
    remote: Counting objects: 5, done.
    remote: Compressing objects: 100% (3/3), done.
    remote: Total 5 (delta 2), reused 5 (delta 2), pack-reused 0
    Unpacking objects: 100% (5/5), done.
    From https://github.com/MatthieuDartiailh/pyvisa-py
     * [new branch]      _original  -> mdartiailh/_original
     * [new branch]      fix_rpc_timeout -> mdartiailh/fix_rpc_timeout
     * [new branch]      master     -> mdartiailh/master
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git checkout ed044879027bcf34b8a408294679994be3f773bc .
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ python lib_sa_pyvisa.py

    Connecting to .... 10.10.100.101
    Session ID                                              : 5096395
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    Traceback (most recent call last):
      File "lib_sa_pyvisa.py", line 943, in <module>
        _fn_main()
      File "lib_sa_pyvisa.py", line 891, in _fn_main
        data= sa.fn_read_iqdata()
      File "lib_sa_pyvisa.py", line 394, in fn_read_iqdata
        data = self.conn.query_binary_values(':READ:WAV0?', datatype='f', is_big_endian=True)
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 653, in query_binary_values
    header_fmt)
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 467, in read_binary_values
        block = self._read_raw()
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
        chunk, status = self.visalib.read(self.session, size)
      File "/home/skandasamy/ubuntu/pyvisa/pyvisa-py/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
        pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (master)$ git checkout -b fix_rpc_timeout mdartiailh/fix_rpc_timeout
    Branch fix_rpc_timeout set up to track remote branch fix_rpc_timeout from mdartiailh.
    Switched to a new branch 'fix_rpc_timeout'
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$ python lib_sa_pyvisa.py

    Connecting to .... 10.10.100.101
    Session ID                                              : 8287459
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    Traceback (most recent call last):
      File "lib_sa_pyvisa.py", line 943, in <module>
        _fn_main()
      File "lib_sa_pyvisa.py", line 891, in _fn_main
        data= sa.fn_read_iqdata()
      File "lib_sa_pyvisa.py", line 394, in fn_read_iqdata
        data = self.conn.query_binary_values(':READ:WAV0?', datatype='f', is_big_endian=True)
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 653, in query_binary_values
    header_fmt)
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 467, in read_binary_values
        block = self._read_raw()
      File "/opt/anaconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
        chunk, status = self.visalib.read(self.session, size)
      File "/home/skandasamy/ubuntu/pyvisa/pyvisa-py/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
    pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$

MatthieuDartiailh, 2018-04-02:
Ok so we made some progress, but we are not there yet. I will try to give it another look tomorrow.

MatthieuDartiailh, 2018-04-02:
Out of curiosity how long does the instrument usually takes to answer ? Could you try increasing the timeout to see if it fixes the issue ?

svanan77, 2018-04-02:
By default, the timeout I have set as 10000 (ms). Its is usually very very fast. Unfortunately I have not timed it.
Retested by setting the timeout 3 times higher than default, 30000 (ms). Got the same error as above

MatthieuDartiailh, 2018-04-03:
Some additional questions:
- when you increase the timeout do you see the expected delay before the error ?
- could you add the following to the top of your script and post the result?
```python
from pyvisa import log_to_screen
log_to_screen()
```
- could you also post the result of your script with logging enabled on the last commit that worked ?

Sorry about not figuring this out faster but missing a local reproduction is a real killer.

svanan77, 2018-04-03:
@MatthieuDartiailh 

I have some updates

1. I have included a timer log in my script. I used a try/except catch where whenever there is a timeout the timer information is still produced measuring the time taken till the timeout occurred.

2. On the good commit i.e (b1d0c51), my instrument takes lesser than 5 sec to respond with 120,000 of IQ data samples. This tells, the timeout/delay set 10000 (10s) would be sufficient. Do refer the below,

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py ((no branch, bisect started on master))$ python lib_sa_pyvisa.py

   ---> time_start= 10.17.44

    Connecting to .... 10.10.100.101
    Session ID                                              : 8243518
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    ---> time_stop= 10.17.48

3.  Answering the below question, I have done 3 tests on the current failing commit (i.e fix_rpc_timeout), where the timeout set to 10000 (10s), 30000 (30s), and 300000 (300s=5m). 

> when you increase the timeout do you see the expected delay before the error ?

From the time logger I find this takes effect accurately. Refer below.

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$ python lib_sa_pyvisa.py
    **************************************************
    Test for delay = 10000 (10s)
    **************************************************

    ---> time_start= 09.49.48

    Connecting to .... 10.10.100.101
    Session ID                                              : 2600080
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    ---> time_stop= 09.50.02

    Traceback (most recent call last):
      File "lib_sa_pyvisa.py", line 961, in <module>
        _fn_main()
      File "lib_sa_pyvisa.py", line 902, in _fn_main
        data= sa.fn_read_iqdata()
      File "lib_sa_pyvisa.py", line 404, in fn_read_iqdata
        assert 0
    AssertionError
    
    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$ python lib_sa_pyvisa.py
    **************************************************
    Test for delay = 30000 (30s)
    **************************************************

    ---> time_start= 09.50.14

    Connecting to .... 10.10.100.101
    Session ID                                              : 2940565
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    ---> time_stop= 09.50.48

    Traceback (most recent call last):
      File "lib_sa_pyvisa.py", line 961, in <module>
        _fn_main()
      File "lib_sa_pyvisa.py", line 902, in _fn_main
        data= sa.fn_read_iqdata()
      File "lib_sa_pyvisa.py", line 404, in fn_read_iqdata
        assert 0
    AssertionError

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$ python lib_sa_pyvisa.py
    **************************************************
    Test for delay = 300000 (300s=5m)
    **************************************************

    ---> time_start= 09.51.00

    Connecting to .... 10.10.100.101
    Session ID                                              : 2420050
    Instrument identification information:                  : Agilent Technologies,N9010A,MY48030221,A.13.07
    Measurement process status:                             : 1

    ---> time_stop= 09.56.03

    Traceback (most recent call last):
      File "lib_sa_pyvisa.py", line 961, in <module>
        _fn_main()
      File "lib_sa_pyvisa.py", line 902, in _fn_main
        data= sa.fn_read_iqdata()
      File "lib_sa_pyvisa.py", line 404, in fn_read_iqdata
        assert 0
    AssertionError

    skandasamy@svds07:~/ubuntu/pyvisa/pyvisa-py (fix_rpc_timeout)$

Rushing to a meeting, will get back with more information

svanan77, 2018-04-03:
@MatthieuDartiailh 

Continuing the update,

Responding to the below,

> 
> - could you add the following to the top of your script and post the result?
> 
>     from pyvisa import log_to_screen
>     log_to_screen()
> 
> - could you also post the result of your script with logging enabled on the last commit that worked ?

Do refer my attached log files.

Thanks.

[bad_fixrpctimeout_log2screen.txt](https://github.com/pyvisa/pyvisa/files/1872717/bad_fixrpctimeout_log2screen.txt)

[good_master_b1d0c51_log2screen.txt](https://github.com/pyvisa/pyvisa/files/1872719/good_master_b1d0c51_log2screen.txt)

MatthieuDartiailh, 2018-04-04:
Thanks for your fast answers !

I pushed a two new commits that will allow me to dig deeper in the reasons of the failures. Could re-run the log to screen with these new commits ?

svanan77, 2018-04-05:
@MatthieuDartiailh 

I am still waiting for slot time to use the instrument. Will update once I get some results.
Hang on.

sibbi77, 2018-04-10:
I've the same problem.
I'm on 38e77ae93f871cc2850ba7b8e40f071ef4425594
PyVISA Version: 1.10.0.dev0
see attached log file
[log1.txt](https://github.com/pyvisa/pyvisa/files/1893599/log1.txt)

The received data starts correctly with #76201654. I expect 6201654 bytes to follow.

MatthieuDartiailh, 2018-04-10:
Thanks for providing the log I will try to look into it.

sibbi77, 2018-04-11:
The hardcopy I like to transfer indeed conatins many 0xffs:
42 4D 36 A1 5E 00 00 00 00 00 36 00 00 00 28 00 00 00 40 06 00 00 0C 05 00 00 01 00 18 00 00 00 00 00 00 A1 5E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF [...]

MatthieuDartiailh, 2018-04-11:
Can you try replacing `read_data = read_fun(exp_length)` by `read_data = read_fun(min(exp_length, 4096)` in rpc.py line 352 ? I would like to check if we fail because we try to read too many bytes at once.

sibbi77, 2018-04-11:
[log3.txt](https://github.com/pyvisa/pyvisa/files/1897566/log3.txt)
I had to add a closing brace to the replacement command in line 352.
No change...

MatthieuDartiailh, 2018-04-11:
Thanks for the quick feedback

svanan77, 2018-04-11:
Thanks @sibbi77 for coming in. I am still unable to get hold of my equipment.

As usual @MatthieuDartiailh , is as speed as lightning looking into the issue.

MatthieuDartiailh, 2018-04-11:
Sadly for the time being I have still no idea what may be really going on there... I need to sit down quietly to look at this and given my current schedule it may take some time. Anybody interested can investigate more deeply.

MatthieuDartiailh, 2018-04-15:
Since I really have a hard time figuring this out I pushed some changes to the debugging infos. Could one of you @svanan77 @sibbi77 run your script with logging at the debug level and post the result ? I will try to dedicate some time to this next week.

greyltc, 2018-04-16:
Is this the error you all are discussing here? I see this on the master branch now:

1) I fill my oscilloscope's (communication via vxi-11 over ethernet) buffer with 2.4 million samples
1) I can read 1.1 million samples out with no problem
1) Reading 1.2 million samples out gives me a timeout
1) Changing the `.timeout` value for the instrument object seems to do nothing

MatthieuDartiailh, 2018-04-16:
@sibbi77 see the same issue on much smaller dataset but I guess it is related. Could you try pyvisa/pyvisa-py#130 with `pyvisa.log_to_screen()` (see above) and post the output ? I would like to know if this has something to do with using multiple fragment per record ?

greyltc, 2018-04-17:
With https://github.com/MatthieuDartiailh/pyvisa-py/tree/fix_rpc_timeout 

My test script: https://gist.github.com/greyltc/33e51700bbe79eaf822a72cf37d7407d
which I run like this: `python timeoutTest.py &> testLog.txt`
giving [this result](https://gist.githubusercontent.com/greyltc/c1c324cd5d927f1aa5b60c43d8dd56c5/raw/603cbab2a317410c392d505b5d7021ca813411fc/testLog.txt).

MatthieuDartiailh, 2018-04-17:
Thanks @greyltc, I will give a look at what you sent.

MatthieuDartiailh, 2018-04-19:
Looking at your code @greyltc there is actually a problem in the way you read the data, in that you forget to read the termination character of the first large transfer. This may cause some unexpected issue. Could you run the same test using `query_binary_values` (and possibly also your method but reading the termination character) ?
@svanan77 @sibbi77 I am still interested in seeing what you get with last commit I pushed. 
Just so that you know I am in holidays next week, so I probably won't answer as quickly as usual.

greyltc, 2018-04-21:
I'm pretty sure I'm not missing a term char because trying to read one single byte after I read the 1.1 million samples times out.

greyltc, 2018-04-23:
So after looking more closely at this, I can't recreate the error with pyvisa-py. I was getting a timeout (as described above) only because I was illegally requesting too many bytes from my instrument. After reading its documentation more closely, it doesn't support reading 1.2 million samples at once. So, for me and my oscilloscope comms over ethernet, there is no error.

MatthieuDartiailh, 2018-04-24:
Thanks for looking into thus @greyltc. I will have some free times in two weeks so I will try to cut a release at this time. If we do not manage to fix this by then I will list it as a known issue.

MatthieuDartiailh, 2018-05-02:
@svanan77 @sibbi77 any chance you could try running with the additional debug logs ?

sibbi77, 2018-05-04:
I was on a business trip; sorry for the delay.
I cannot get my hands on the RS FSV13 (where I could see the error) right now, but for RS SMJ100A and RS FSQ26 it works.
I use pyvisa-py 8aa0ffd3ae5981cf4a110dd3c8ab856ca7af9312
[RS_FSQ26.log](https://github.com/pyvisa/pyvisa/files/1973975/RS_FSQ26.log)
[RS_SMJ100A.log](https://github.com/pyvisa/pyvisa/files/1973976/RS_SMJ100A.log)
[untitled1.py.txt](https://github.com/pyvisa/pyvisa/files/1973985/untitled1.py.txt)

MatthieuDartiailh, 2018-05-04:
Thanks for testing @sibbi77. I will probably have to remove part of the debugging  code before merging but if you manage to reproduce please send me the log using 8aa0ffd3ae5981cf4a110dd3c8ab856ca7af9312 so that we get any many info to work with as possible.

closed 2018-07-04 by MatthieuDartiailh