pyvisa/pyvisa #321: Remove VISA as a requirement from Readme.md

ClundXIII, 2018-03-22:
Hi,

I forced myself onto windows for my latest few experiments. I have been painfully trying to get any proprietary visa library running under linux.

Just to find out that you can use a simple '@py' with pyvisa-py for USB devices. Please make that clear in the docs and the readme to prevent idiots like me from wasting time ;-)

Simon

MatthieuDartiailh, 2018-03-22:
You are welcome to do a pull request, otherwise I will try to do so as soon as I find some time. Note that the py backend is not totally bugfree. Under linux (Arch), I recently tested the ni-visa AUR package that appears to work (but I gave it only very limited testing).

MatthieuDartiailh, 2018-03-28:
Closed by #323

closed 2018-03-28 by MatthieuDartiailh