pyvisa/pyvisa #322: Disable timeout (not zero, not infinite, but disable it)

SengerM, 2018-03-27:
Hello,

I would like to know if it is possible to use `instrument.write(...)` and continue the execution of the program without waiting for any kind of response from the instrument. 

I need this because I am using two voltmeters that need to be driven from the same trigger signal. Thus I 
1) Disable trigger from the source.
2) Set trigger arm in voltmeter one.
3) Set trigger arm in voltmeter two.
4) Activate trigger signal from the source.
Under this conditions both voltmeters should trigger at the same time. But it happens that in step 2 I am getting a timeout and the program stops.

The relevant part of the code is as follows:

HP33120A.write('OUTP:SYNC OFF') # Disable the sync terminal from the function generator
HP3458A_1.write('TARM SGL') # Enable sampling, single shot when trigger.
HP3458A_2.write('TARM SGL') # Enable sampling, single shot when trigger.
HP33120A.write('OUTP:SYNC ON') # Enable the sync terminal so the voltmeters trigger.

and the program is ending in the second line due to the timeout.

Thanks in advance,
Matías

MatthieuDartiailh, 2018-03-27:
I have several questions. 
First can you actually communicate at all with your instrument ? For example does your instrument answers to `instr.query('ID?')`  (as it should per the documentation https://www.keysight.com/upload/cmc_upload/All/EPSG084364.pdf) ? If it does not first we need to fix this, and the answer may be to properly set the termination characters used for reading and writing. Also could you check your command syntax ? According to the documentation the TARM command may expect number_arms following SGL.
Second could you post the output of `python -m visa info`, so that I know on which OS and with which version of pyvisa you are working.
Third could you post your full script ? It may help debugging this issue.

Basically I believe you are mostly seeing the result of a real issue in communicating with the device and that even if pyvisa was not emitting a timeout, your instrument would not behave as you expect.

SengerM, 2018-03-27:
Thanks for the quick answer. Here is the code I am using (cannot find how to attach it as a file):

import visa
import numpy as np
import math
import matplotlib.pyplot as plt
plt.rc('axes', axisbelow=True) # This is for plotting data in front of the grid.
plt.rcParams.update({'font.size': 8}) # Changes the default font size for the plots.
from time import sleep
import struct # This is used in parsing bynary data.
import scipy.optimize as opt # This is used for fiting models and data.
import pymagpy as mg # This is a custom class I have created.
from utils.fit_bootstrap import *
from utils.HP3458A_lib import *
from utils.timestamp import *

def HP3458A_configure_DCV_digitalizing(HP3458A, samples_per_burst, effective_sampling_frequency):
	HP3458A.write('PRESET DIG')
	HP3458A.write('MEM FIFO') # ENABLE READING MEMORY, FIFO MODE.
	HP3458A.write('MFORMAT SINT') # SINT READING MEMORY FORMAT.
	HP3458A.write('OFORMAT SINT') # Set the output format when reading the samples in memory.
	HP3458A.write('TIMER ' + str(1/effective_sampling_frequency)) # Specifies sampling frequency.
	HP3458A.write('APER 500e-9') # Set aper time to 500 ns which gives a resolution of 16 bits (see http://literature.cdn.keysight.com/litweb/pdf/03458-90014.pdf table 5-2).
	HP3458A.write('NRDGS ' + str(samples_per_burst) + ', 6') # Specifies the number of samples to be recorded and the event in which to sample (6 is for "timer").
	HP3458A.write('SSRC EXT') # Set external trigger.

# REFERENCES: 
# 	[1] Keysight 3458A Multimeter, https://xdevs.com/doc/HP_Agilent_Keysight/3458A/03458-90014.pdf

# PROGRAM PARAMETERS ----------------------
GENERATOR_FREQUENCY = 1e3 # Hertz
GENERATOR_AMPLITUDE = 1 # Volt
SAMPLES_PER_BURST = 10000 # Integer between 1 and ?? (have not found yet the size of the memory.
CYCLES_PER_BURST = 100 # Number of signal cycles to be recorded in each burst.
EFFECTIVE_SAMPLING_FREQUENCY = SAMPLES_PER_BURST*GENERATOR_FREQUENCY/CYCLES_PER_BURST
SAVE_PLOTS = 0
# -----------------------------------------

# print('\t|\t|\t|\t|\t|\n'*10)

timestamp = get_timestamp()
print('Timestamp: ' + timestamp)

set_frequency = mg.magnitude(GENERATOR_FREQUENCY, GENERATOR_FREQUENCY*20e-6) # The value for the uncertainty in the generator frequency was taken from https://www.keysight.com/upload/cmc_upload/All/6C0633120A_USERSGUIDE_ENGLISH.pdf
set_amplitude = mg.magnitude(GENERATOR_AMPLITUDE, GENERATOR_AMPLITUDE*1e-2) # The value for the uncertainty in the generator amplitude was taken from https://www.keysight.com/upload/cmc_upload/All/6C0633120A_USERSGUIDE_ENGLISH.pdf

rm = visa.ResourceManager()

HP3458A_1 = rm.open_resource('GPIB0::22::INSTR') # Multimeter.
HP3458A_1.read_termination = '\r' # Each reading output to the GPIB in ASCII format is normally followed by '\r\n' (see [1], page 99).
HP3458A_2 = rm.open_resource('GPIB0::23::INSTR') # Multimeter.
HP3458A_2.read_termination = '\r' # Each reading output to the GPIB in ASCII format is normally followed by '\r\n' (see [1], page 99).
HP33120A = rm.open_resource('GPIB0::10::INSTR') # Function generator.

# Check that the instruments are responding:
print('HP3458A_1.query(\'ID?\') returns --> ' + HP3458A_1.query('ID?'))
print('HP3458A_2.query(\'ID?\') returns --> ' + HP3458A_2.query('ID?'))
print('HP33120A.query(\'*IDN?\') returns --> ' + HP33120A.query('*IDN?'))
# Reset the instruments:
HP3458A_1.write('RESET')
HP3458A_2.write('RESET')
HP33120A.write('*RST')
print('Setting generator output to ' + str(set_amplitude) + ' V and frequency to ' + str(set_frequency) + ' Hz')
HP33120A.write('APPL:SIN {}, {}, 0'.format(GENERATOR_FREQUENCY, GENERATOR_AMPLITUDE)) # Set a sine wave in the generator.
# HP33120A.write('OUTP:SYNC OFF') # Disable the sync terminal (so the voltmeters do not trigger until they are configured).
print('Configuring voltmeters...')
HP3458A_configure_DCV_digitalizing(HP3458A_1, SAMPLES_PER_BURST, EFFECTIVE_SAMPLING_FREQUENCY)
HP3458A_configure_DCV_digitalizing(HP3458A_2, SAMPLES_PER_BURST, EFFECTIVE_SAMPLING_FREQUENCY)
HP3458A_1.write('TARM SGL') # ENABLE SAMPLING (SGL specifies to trigger a single shot, see [1] pag. 251).
HP3458A_2.write('TARM SGL') # ENABLE SAMPLING (SGL specifies to trigger a single shot, see [1] pag. 251).
[... 104 lines cut ...]
# PLOT DATA USING CORRECTION ON fs1 AND fs2 ----------------------------------------
FIG_WIDTH = 130e-3 # Figure width in meters (size of the chart+axis_ticks, not counting legend, title, etc.).
FIG_RATIO = [1, 0.4] # XY ratio aspect of figures.
f, (ax1) = plt.subplots(1, sharex=True, figsize=(FIG_WIDTH*FIG_RATIO[0]/25.4e-3, FIG_WIDTH*FIG_RATIO[1]/25.4e-3)) # Create the figure for plotting.
f.subplots_adjust(hspace=0.3) # Fine-tune figure; make subplots close to each other and hide x ticks for all but bottom plot.
ax1.plot(np.linspace(0, SAMPLES_PER_BURST/fs1.val, SAMPLES_PER_BURST), samples_1, marker='.', label='V1')
ax1.plot(np.linspace(0, SAMPLES_PER_BURST/fs2.val, SAMPLES_PER_BURST), samples_2, marker='.', label='V2')
ax1.plot(np.linspace(0, SAMPLES_PER_BURST/fs1.val, SAMPLES_PER_BURST), sinusoidal_signal(times_1, Vp1.val, f1.val, phi1.val), label='Fit for V1')
ax1.plot(np.linspace(0, SAMPLES_PER_BURST/fs2.val, SAMPLES_PER_BURST), sinusoidal_signal(times_2, Vp2.val, f2.val, phi2.val), label='Fit for V2')
ax1.set_xlabel('Time (s)')
ax1.set_ylabel('Voltage (V)')
ax1.grid(b=True, which='major', color='#aaaaaa', linestyle='-', linewidth=0.5) # Configure the grid in a more fachera way.
[... cut ...]
if SAVE_PLOTS:
	plt.savefig(timestamp+'_corrected_signals.pdf', bbox_inches='tight')
# ----------------------------------------------------

HP3458A_1.close()
HP3458A_2.close()
HP33120A.close()

When the line "HP33120A.write('OUTP:SYNC OFF')" is commented it works pretty fine, because the voltmeter triggers instantly after writing the command TARM SGL. But in this case the two voltmeters measure at different times. When that line is not commented I get a timeout in the line "HP3458A_1.write('TARM SGL')".

MatthieuDartiailh, 2018-03-27:
Looking at the documentation of your instrument (following the link in your code https://xdevs.com/doc/HP_Agilent_Keysight/3458A/03458-90014.pdf) p254, it is mentioned (under Multiple arming) that `TARM SGL` blocks the GPIB line till all the triggers events have been received which means you cannot use it in the way you are trying to (and this is not a PyVISA limitation). For your application a trigger EXT may be more appropriate.

I am closing since there is nothing to be done on PyVISA end. Feel free to re-open should you have any related question ?

closed 2018-03-27 by MatthieuDartiailh

SengerM, 2018-03-27:
Okay, thanks anyway. I have passed that paragraph away when configuring the voltmeter.