pyvisa/pyvisa #325: WindowsError: exception: access violation reading 0x0000000000000000

cullenmcmahon, 2018-04-04:
I'm getting the following error:

  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1634, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1594, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Anaconda2\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1383, in parse_resource_extended
    alias_if_exists)
WindowsError: exception: access violation reading 0x0000000000000000

I was able to get it to run by changing byref to ctypes.pointer in ctwrapper\functions.py in the call on line 1380.

$ python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 70 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda2\python.exe
   Version:        2.7.13
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          May 11 2017 13:17:26 (#default)
   Unicode:        UCS2

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928

MatthieuDartiailh, 2018-04-04:
Thanks for reporting and with a possible fix on top of that. I will try to look into it. If you do not hear from me in a week or so do not hesitate to ping me.

MatthieuDartiailh, 2018-04-05:
@cullenmcmahon Could you perform some additional tests for me (I cannot reproduce this locally) ?
- creating a new environment with conda can you test under Python 3 ?
- creating a new environment with conda can you test with the latest Python 2 (2.14) ?
- can you check what happens if you remove the byref ?
- can you check if it works with ViPUint16 instead of pointer (I would expect it works too) ?
- can you test some other functions, I am interested in particular in viWrite because of #271 ?

MatthieuDartiailh, 2018-04-10:
@cullenmcmahon can you perform my suggested tests ?

cullenmcmahon, 2018-04-13:
Haven't had time this week, will address next week.

MatthieuDartiailh, 2018-04-28:
ping @cullenmcmahon

Oscilllator, 2018-07-09:
I also have this issue, and have tried is using both python 2.7.10:
################################################
>python -m visa info                                Machine Details:                                                                                                           Platform ID:    Windows-8-6.2.9200                                                                                      Processor:      Intel64 Family 6 Model 78 Stepping 3, GenuineIntel                                                                                                                                                                           Python:                                                                                                                    Implementation: CPython                                                                                                 Executable:     C:\WinPython-64bit-2.7.10.3\python-2.7.10.amd64\python.exe                                              Version:        2.7.10                                                                                                  Compiler:       MSC v.1500 64 bit (AMD64)                                                                               Bits:           64bit                                                                                                   Build:          May 23 2015 09:44:00 (#default)                                                                         Unicode:        UCS2                                                                                                                                                                                                                         PyVISA Version: 1.9.0                                                                                                                                                                                                                           Backends:                                                                                                                  ni:                                                                                                                        Version: 1.9.0 (bundled with PyVISA)                                                                                    #1: C:\WINDOWS\system32\visa32.dll:                                                                                        found by: auto                                                                                                          bitness: 64                                                                                                             Vendor: National Instruments                                                                                            Impl. Version: 16777216                                                                                                 Spec. Version: 5244672                                                                                               #2: C:\WINDOWS\system32\visa64.dll:                                                                                        found by: auto                                                                                                          bitness: 64                                                                                                             Vendor: National Instruments                                                                                            Impl. Version: 16777216                                                                                                 Spec. Version: 5244672
#############################################

As well as python 3.7 (fresh install of pyvisa):
#############################################
Machine Details:                                                                                                           Platform ID:    Windows-10-10.0.17134-SP0                                                                               Processor:      Intel64 Family 6 Model 78 Stepping 3, GenuineIntel                                                                                                                                                                           Python:                                                                                                                    Implementation: CPython                                                                                                 Executable:     C:\WinPython-64bit-3.6.5.0\python-3.6.5.amd64\python.exe                                                Version:        3.6.5                                                                                                   Compiler:       MSC v.1900 64 bit (AMD64)                                                                               Bits:           64bit                                                                                                   Build:          Mar 28 2018 17:00:18 (#v3.6.5:f59c0932b4)                                                               Unicode:        UCS4                                                                                                                                                                                                                         PyVISA Version: 1.9.0                                                                                                 [... cut ...]                                                                                      ret = library.viWrite(session, data, len(data), byref(return_count))                                                WindowsError: exception: access violation reading 0x000000000905266C

I tried replacing byref with ctypes as @cullenmcmahon suggested, as well as ViPUint16. This did not work either, same error message (although the memory address changes a bit each time I run the program).

ping @MatthieuDartiailh

MatthieuDartiailh, 2018-07-09:
This is weird. Do you have an anti-virus that may try to prevent the acces to the dll ? Also can you test that other National Instruments programs such as NIMax wirk as expected ?

Oscilllator, 2018-07-09:
That's a good point about the antivirus. I disabled it as well as my firewall since I am accessing the device over LAN, but this did not resolve the issue. I have been able to manually open a telnet port to the device as well as to access its web interface however, so I don't think that's the problem either although I could be wrong about that.

I haven't tried running other national instruments programs but I did install pyvisa-py as an alternative to the NI library, and this did not appear to change anything.

MatthieuDartiailh, 2018-07-10:
To actually use pyvisa-py you need to instantiate the resource manager using `ResourceManager ('@py')`. And since we then use simple sockets I do not expect such errors. 
I would really suggest to validate your installation using NIMax. One thing that can cause issue is if you prevented the installer to disable Windows fast start-up or did not restart after installing.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Based on what I have seen in other setup, it may have been a bad install of Python with a corrupted ctypes. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh