pyvisa/pyvisa #326: How to specify parity and stop bits?

wendlerj, 2018-04-05:
I am trying to control a slightly offbeat device.
I am using Python 3.65 with pyVISA 1.9.0 and the latest NI VISA under Windows 10.
The device would like a serial port at 9600 baud, 7 data bits, odd parity, 1 stop bit.  The flow control is rtscts
Incrementally building up the open_resource string, I can get this far without error:
rm=visa.ResourceManager()
doohickey = rm.open_resource('ASRL5::INSTR',
			    baud_rate=9600,
			    data_bits=7,
			    flow_control=2)

when I try to add parity and stop_bits into the rm.open line, I get errors of the flavor "should be a <enum 'StopBits'>" and "should be an <enum Parity>" I have tried a wide variety of things without success and have not been able to find a pyvisa example that fully specifies a serial port.

What is the correct way to add odd parity and 1 stop bit to the above code?

Thanks in advance

wendlerj, 2018-04-05:
There is a typo in the above - the other error is "should be a <enum 'Parity'>

MatthieuDartiailh, 2018-04-05:
You need to import the matching enumeration from the `pyvisa.constants` module (ie StopBits and Parity) and then do 
```python
from pyvisa.constants import StopBits, Parity
rm=visa.ResourceManager()
doohickey = rm.open_resource('ASRL5::INSTR', 
    baud_rate=9600, data_bits=7, flow_control=2, 
    parity=Parity.odd, stop_bits=StopBits.one)
```

I have not tested because I do not have an instrument close by but it should work.

wendlerj, 2018-04-05:
Matthieu, thanks, that worked great!

closed 2018-04-05 by wendlerj