pyvisa/pyvisa #327: FutureWarning: values_format is deprecated and will be removed in 1.10

greyltc, 2018-04-07:
What should I use instead?

The docs still seem to suggest I use `values_format`:
http://pyvisa.readthedocs.io/en/stable/rvalues.html?highlight=values_format#preconfiguring-the-transfer-format

MatthieuDartiailh, 2018-04-07:
I am sorry I missed this part of the docs I will try to update them as soon as possible. Basically read_values is deprecated and you should use read_ascii_values or read_binary_values as appropriate. The way to preconfigure the transfer is actually gone. I may consider revive it bit I would need some strong arguments.

greyltc, 2018-04-07:
So if I'm using non-default values for datatype, is_big_endian, or container I should pass them in as kw arguments whenever I call `query_binary_values`?

Having `values_format` to preconfigure was convenient for avoiding that I guess.

MatthieuDartiailh, 2018-04-08:
Adding a `binary_values_format` and a `ascii_values_format` may make sense to allow preconfiguring. And in any case the docs need to be updated. @greyltc would you have the time to put up a PR ?

MatthieuDartiailh, 2018-04-28:
@greyltc would you be interested in `binary_values_format` and `ascii_values_format`  ? (just to know if I should add it on my todo list)

greyltc, 2018-04-29:
I do think it makes sense to be able to preconfigure these, but it's not terribly inconvenient to pass the configuration parameters to `query_binary_values` every time.

marccheneau, 2018-10-02:
@MatthieuDartiailh Because of the deprecation warning I went for storing these format settings in my wrapper class and pass them in all calls to the write/read/query methods. It definitely makes sense to provide a way to set these parameters in the `MessageBasedResource` class. Why do you want to get rid of 'ValueFormats' in version 1.10? I could spend some time there to propose a patch but I would need to know what your preferred option is.

MatthieuDartiailh, 2018-10-02:
As mentioned above `read_values` is deprecated and new methods replace it, as I cannot see an easy backward compatible way to re-use `ValuesFormat` I am planning to suppress at the same time `read_values` goes away.
Since there seems to be an interest in pre-configuring the transfer parameters (personally I never encountered the need for more than a couple of occurrences of those methods per driver and hence it did not seem crucial to me to provide a resource level configuration) , my suggestion would be to get two new attributes `binary_values_format` and `ascii_values_format` backed by matching classes. I would have to look in more details to check if we need different parameters for reading and writing but it may make sense.

marccheneau, 2018-10-03:
If it makes sense I could start implementing something along those lines soon. May I ask why you would store the settings in separate classes? Why not treating them the same way as the write/read terminations or the encoding?

MatthieuDartiailh, 2018-10-03:
Both ascii and binary values require multiple parameters, to avoid crowding the root of the resource with those, I prefer to group them within a dedicated object. As mentioned an open question is do we need separate parameters for reading and writing.

MatthieuDartiailh, 2020-09-06:
As mentioned in that thread I do see a compelling need for such configuration. Feel free to re-open if this requires more discussion.

closed 2020-09-06 by MatthieuDartiailh