pyvisa/pyvisa #328: any ResourceManager command gives VI_ERROR_INV_OBJECT

sparrowCE, 2018-04-10:
import visa
rm = visa.ResourceManager()
visa_address = 'USB::0x0699::0x040F::C010583::INSTR'
scope = rm.open_resource(visa_address)

**gives error:**

2018-04-10 16:59:27,570 - pyvisa - DEBUG - No visa libaray specified and environment variable PYVISA_LIBRARY is unset. Using NI library
2018-04-10 16:59:27,579 - pyvisa - DEBUG - No user defined library files
2018-04-10 16:59:27,598 - pyvisa - DEBUG - Automatically found library files: ['C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa64.dll']
2018-04-10 16:59:27,599 - pyvisa - DEBUG - Reusing ResourceManager with session 460489664
2018-04-10 16:59:27,601 - pyvisa - DEBUG - viParseRsrcEx(460489664, 'USB::0x0699::0x040F::C010583::INSTR', 'c_ushort(0)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x000001CA1A5E0D48>, <ctypes.c_char_Array_256 object at 0x000001CA1A5E0BC8>, <ctypes.c_char_Array_256 object at 0x000001CA1A2F5DC8>) -> -1073807346
Traceback (most recent call last):

  File "<ipython-input-70-dba0729c117a>", line 4, in <module>
    scope = rm.open_resource(visa_address)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 1634, in open_resource
    info = self.resource_info(resource_name, extended=True)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 1594, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1383, in parse_resource_extended
    alias_if_exists)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

**and visa info:**

H:\>python -m visa info
Machine Details:
   Platform ID:    Windows-10-10.0.14393-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

i've also tried

import visa
rm = visa.ResourceManager()
scope = rm.open_resource('USB::0x0699::0x040F::C010583::INSTR')

**and many many other combinations**

MatthieuDartiailh, 2018-04-11:
I do not know what VISA library you have installed, but I have seen a similar issue with Ni-VISA when preventing the installer from disabling windows fasr startup. The fix was to reinstall and let it disable the fast startup. If everything goes well the infos about the library should not contain any error message.

sparrowCE, 2018-04-11:
fast startup was already disabled on my computer; so does this still apply?

MatthieuDartiailh, 2018-04-12:
I have no idea. The only time I have seen this kind of failure was when keeping the fast startup. You could try to re-install Visa. Simply checking the output of the visa infos will tell you if everything is alright or not.

sparrowCE, 2018-04-13:
i was using TekVISA with a Tektronix  MDO4054-3 oscilloscope.  i uninstalled TekVISA and installed NI-VISA as per the recommendation of the Tektronix tech support guy and everything works beautifully!