pyvisa/pyvisa #330: PyVisa generating lots of errors/exceptions - Still working though

JacobEFO, 2018-04-17:
Hello PyVisa community

First of all I love this piece of software, it is absolutely fantastic and works splendidly.

However I get quite some errors when I interface with my lab equipment, even though it works.

See the following link for a screenshot: https://imgur.com/a/o1gjt

I have been looking a little around, but I can't seem to find the issue. 

Best regards Jacob

MatthieuDartiailh, 2018-04-17:
I suspect that those errors show up because you do not explicitly close the connection to the instruments. As a consequence we try to close them when the object is deleted, however at that time the ResourceManager may have already been destroyed which I suspect leads to the error you are seeing. You can either close explicitly the connection at the end of your scripts or use a context manager:
```python
with rm.open_resource('TCPIP::192.168.0.10::INSTR) as my_instrument:
    my_instrument.query('*IDN?')
```

JacobEFO, 2018-04-17:
I do believe I am using a context manager, looks similar to yours. I do not explicitly close the connection to the device though, might be that is the issue.

The following link shows an example of what I am currently doing: https://pastebin.com/3dZpgYGe

MatthieuDartiailh, 2018-04-17:
Sorry when I meant a context manager I was not referring to the pyvisa resource manager but the Python with syntax, which takes care of cleaning up when you leave its scope (you can do the same for files using open). Using this of explicitly closing the connection should allow you to get rid of the error.

JacobEFO, 2018-04-17:
Ah okay, I'll be trying that tomorrow then. Thanks for the feedback!

JacobEFO, 2018-04-19:
Ah fantastic, thanks for your help.

I solved the issue by explicitly closing the connection using before_close() and close(). That solved the issue.

Thanks again for realizing Pyvisa!

JacobEFO, 2018-04-19:
Issue resolved:

Close connection using:

my_instrument.before_close()
my_instrument.close()

When you are done with the instrument.

closed 2018-04-19 by JacobEFO