pyvisa/pyvisa #332: PyVisa cannot connect to DPO7104 Tektronix scope on GPIB

Mehran1424, 2018-05-01:
I tried to connect to DPO7104 using PyVisa and either TekVisa backened or NI-VISA 17.0. Using TekVisa I can actually see the instrument using list_resources() method, but when trying to connect I get a "nsufficient location information or the device or resource is not present in the system" error. When I use Ni-VISA backend, I can's even see the device.
On both instances I can communicate with the scope through TekVisa GUI or NI MAX. I'm using PyVISA 1.9 with 32 bit Python.

denimdragon, 2018-05-01:
A few questions and maybe I can help.
  

  
What GPIB interfaces adapters are you using? NI-USB-GPIB, Agilent, etc. Also if you have NI and Agilent VISA drivers that can cause problems, so it could be the same situation with TekVISA.
  

  
And lastly how are you connecting and and which OS? I went through similar issues last week on Linux and W7. I was finally able to connect but I use an Ethernet to GPIB adapter (Prologix).

Mehran1424, 2018-05-01:
Thanks! I'm using NI-USB-GPIB, and Windows 10.
I have an update: I downgraded both NI-VISA (from 17.5 to 16.0) and NI 488.2 (from 17.6 to 17.0) drivers and it finally worked. Not sure which one did the trick but NI 488.2 17.6 was released only in February so it might've been because of that.