pyvisa/pyvisa #334: pyvisa-py as fallback implementation

bilderbuchi, 2018-05-09:
To use pyvisa, one needs a backend, by default the NI-VISA library. Unfortunately, the binary nowadays seems to be behind a [login wall](http://lumen.ni.com/nicif/US/GB_NIDU/content.xhtml?du=http://www.ni.com/download/ni-visa-5.4.1/4626/en/), so there is definitely a barrier to get going.

If no NI-VISA library is found installed, I think it would make sense to automatically fall back to the open [pyvisa-py](https://github.com/pyvisa/pyvisa-py) implementation. This way, nothing changes for users having NI-VISA installed, and it's still used by default if people can install the library, but the procedure is smoother than now for people who don't have access to that library.

In particular, it would make it easier to use packages like pymeasure where you are [not necessarily using](http://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html) a `ResourceManager` to configure the backend choice. Yes I know that it's possible to select the backend via environment variable, but that's another extra step that users have to do every time they use pyvisa.

Does the proposal to automatically fall back to pyvisa-py if NI-VISA is not installed sound good/smart/acceptable?

MatthieuDartiailh, 2018-05-14:
I believe it would make sense yes to use pyvisa-py as a fallback backend. I have no time to do myself at the time being but I will be happy to review and merge a PR in that direction.

bilderbuchi, 2018-05-15:
Ok, do you have any pointers/tips where to start and how to approach implementing this?

MatthieuDartiailh, 2018-05-15:
The function that needs to be updated is open_visa_library in highlevel.py
You could wrap the library instantiation and decide to fall back if an error occurs.

bilderbuchi, 2018-05-20:
Thanks, I'll see what I can do. :-)

commit f2d69892 (2018-06-01): For default backend, use pyvisa-py if the NI binary cannot be found.

Fixes #334.
commit c1d2f901 (2018-06-01): Merge #341

341: For default backend, use pyvisa-py if the NI binary cannot be found. r=MatthieuDartiailh a=bilderbuchi

This fixes #334.

I also have a commit prepared that adds `pyvisa-py` as a dependency to setup.py, but I don't know if you want that. If we include it, it's guaranteed that people will always have a backend available even if the NI one isn't installed, so i lean towards doing that. Please comment on that @MatthieuDartiailh (*edit: I have pushed that commit to the PR now, it should be easy to not included it if desired*)

I tested this on my local machine (without the NI libr [... cut]
PR #341 (2018-06-01): For default backend, use pyvisa-py if the NI binary cannot be found.
This fixes #334.

I also have a commit prepared that adds `pyvisa-py` as a dependency to setup.py, but I don't know if you want that. If we include it, it's guaranteed that people will always have a backend available even if the NI one isn't installed, so i lean towards doing that. Please comment on that @MatthieuDartiailh (*edit: I have pushed that commit to the PR now, it should be easy to not included it if desired*)

I tested this on my local machine (without the NI library). pyvisa-py gets selected successfully:

```
imp [... cut]
closed 2018-06-01 by bors[bot]