pyvisa/pyvisa #335: WPF + IronPython+Pyvisa

vita0206, 2018-05-11:
when i click the button, VS2013 show the below error:

[image]

[image]

1.  **WPF code is shown as below:**
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows;
using System.Windows.Controls;
using System.Windows.Data;
using System.Windows.Documents;
using System.Windows.Input;
using System.Windows.Media;
using System.Windows.Media.Imaging;
using System.Windows.Navigation;
using System.Windows.Shapes;
using IronPython.Hosting;
using Microsoft.Scripting.Hosting;

namespace Tek
{
    /// <summary>
    /// Interaction logic for MainWindow.xaml
    /// </summary>
    public partial class MainWindow : Window
    {
        ScriptRuntime pyRuntime = null;
        dynamic Tek = null;
        public MainWindow()
        {
            InitializeComponent();
        }
        private void Dir_Open_Bt_Click(object sender, RoutedEventArgs e)
        {
            pyRuntime = Python.CreateRuntime();
            Tek = pyRuntime.UseFile(AppDomain.CurrentDomain.BaseDirectory + @"Tek.py");
            Tek.Devices(); //show errors
        }
    }
}

2.python file: Tek.py
#coding=utf-8
def Devices():
     import sys
     sys.path.append('D:\\Python27')
     sys.path.append('D:\\Python27\\Lib')
     sys.path.append('D:\\Python27\\Lib\\site-packages')
     sys.path.append('D:\\Python27\\Lib\\site-packages\\six-1.11.0.dist-info')
     sys.path.append('D:\\Python27\\Lib\\site-packages\\PyVISA-1.9.0-py2.7.egg-info')
     sys.path.append('D:\\Python27\\Lib\\site-packages\\enum34-1.1.6.dist-info')
     import six # import successful
     import visa #import fail,ironpython show errors: index out of range: -1
     
     return 0
     #rm=visa.ResourceManager() #
     #return (rm.list_resources())#
     #inst = rm.open_resource('GPIB0::1::INSTR') #

MatthieuDartiailh, 2018-05-14:
I have no experience with IronPython I must say. However I remember reading somewhere that IronPython does not support ctypes, however pyvisa uses it to interface with the VISA dll. You should probably check whether or not this is the case.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh