pyvisa/pyvisa #337: PyVisa cannot find GPIB devices

GnakIewiy, 2018-05-18:
My software details for PyVisa are:
Machine Details:
   Platform ID:    Windows-10-10.0.16299-SP0
   Processor:      Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\python.exe
   Version:        3.6.5
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 28 2018 17:00:18 (#v3.6.5:f59c0932b4)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
------------------------------------------------------------------------------------------------
Note : I use the  ADLINK USB-3488A,   a usb-gpib Adapter cable
------------------------------------------------------------------------------------------------
NI-VISA : 17.5
The GPIB device can be found in NI software (VISA Interactive Control).

USB0::0x0699::0x0415::C031425::INSTR
GPIB0::1::INSTR
GPIB0::21::INSTR
GPIB0::22::INSTR
GPIB0::23::INSTR
ASRL1::INSTR
ASRL3::INSTR

But I cannot find the GPIB device using the following pyvisa code

pyvisa code:
`import visa 
rm = visa.ResourceManager('C:\\Windows\\System32\\nivisa64.dll')
print(rm.list_resources('?*'))`

('USB0::0x0699::0x0415::C031425::INSTR', 'ASRL1::INSTR', 'ASRL3::INSTR')

How can I find the GPIB devices by pyvisa?

MatthieuDartiailh, 2018-05-19:
I am not sure what may be the issue here and I am a bit too busy right now to look into the details. Out of curiosity can you open the resources even if they are not listed ?

GnakIewiy, 2018-05-21:
I also cannot open the gpib resources.

I will try the issue in win32 OS......

MatthieuDartiailh, 2018-05-22:
Not sure what is happening here (I do not have experience with this adaptor, I only tested the Keysight 82357B). Out of curiosity what happens if you do not manually select the visa library. Also could you post the full output of  the `pyvisa-info` info command (or `python -m visa info` for PyVISA < 1.9).

cphenicie, 2019-02-02:
Was there any conclusion on how to make this work? Am having the same problem using an ADLINK LPCI-3488A card. (Running Windows 7 64 bit, using python 2.7 via Anaconda)

MatthieuDartiailh, 2019-02-05:
Could you post the output of `pyvisa-info` ? I got no further answer so no we did not figure out what could be the problem.

cphenicie, 2019-02-05:
Thanks for following up, Mattieu! Here's the output:

(base) C:\Users\thompsonlab>pyvisa-info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\thompsonlab\anaconda2\python.exe
   Version:        2.7.15
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 10 2018 21:57:18 (#default)
   Unicode:        UCS2

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672

MatthieuDartiailh, 2019-02-05:
Ok, among the following list what does work and what does not ?
- finding your instrument in NI-Max ?
- communicating with your instrument using NI-Max ?
- listing your instrument using pyvisa ? (I guess this one fails)
- communicating with your device using pyvisa and the known address ?

If NI does not work as expected, I would suggest you contact the board manufacturer. If it does and Pyvisa does not, we will need to dig more.

cphenicie, 2019-02-06:
- NI Visa can find and communicate with the board (will try NI MAX tomorrow if that's an important data point) 

- does not show up on pyvisa

- cannot communicate with known address via pyvisa 

I would also add that Matlab's instrument explorer can also find and communicate with the board, so this gives me hope that the problem isn't laying too deep.

MatthieuDartiailh, 2019-02-06:
I am a bit perplex. Pyvisa basically calls the NI-Visa library to perform the job of finding the resources and then translate the result (you can look at `highlevel.py` and `ctwrapper/highlevel.py`) . The only thing that sometimes surprises people is that the default filter is `'?*:INSTR'` matching only INSTR resources. I would be curious to know what you see if you use `rm.list_resources('?*')`.

Also how did you test bare NI-Visa ? I am curious.

hgrecco, 2019-02-06:
The ability to discover and to connect are two different things. A lot of USB-GPIB adapters fail to tell  NI-VISA that something is plugged but connecting and talking to them works perfectly. In this cases,  VISA Interactive Control (VISAIC) discovers connected devices that NI-MAX does not. I am not sure why, I guess VISAIC calls other underlying libraries.

MatthieuDartiailh, 2019-02-06:
Thanks for the insight @hgrecco. In this case the OP is using a PCIe card not a USB adapter but I guess there are similarities.

hgrecco, 2019-02-06:
It might be the same thing. What I have seen is that certain companies provide their own gpib dll which is binary compatible with the one for National Instrument. Therefore you can use it a drop in replacement. I am not sure how, but VISAIC sometimes is able to discover and query this library (or maybe the installation was just replacing the old one). 

`pyvisa-ni` calls the `nivisa` which calls gpib which defaults to the NI `gpib-32.dll` (this is the old name, I do not remember the name now)

Additionally, matlab has a `gpib` function that might be going directly to gpib dll without calling visa dll. This could explain why matlab works.

MatthieuDartiailh, 2019-02-08:
Looking at https://github.com/tivek/gpib_ctypes which we use in pyvisa-py it seems the dll is still named the same. @cphenicie could you experiment replaing the default dll with the manufacturer specific one and confirm it fixes the issue. If it does we will investigate how to make happen in a bit more automatic manner (either through pyvisa-py where we could load a custom dll, or by providing ni with the proper information).

dcribeiro, 2019-05-09:
Hi all! I do not know if this issue was fixed or not, but I am now having a very similar problem.
In the past I had pyvisa working perfectly fine and showing all GPIB instruments. But after a Python update through Anaconda, I can no longer see my GPIB instruments.
I can see them and control them in NI MAX, but not through pyvisa.
In an attempt to solve the issue, I already updated NI MAX, NIvisa and NI 488.2. I have also tried older python and pyvisa versions without success.
Currently this is what I get from the pyvisa info:

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.8
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 21 2019 18:30:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928

These are the resources I can see: 
 'PXI0::1::BACKPLANE',
 'USB0::0x05FF::0x1023::3808N60697::INSTR',
 'TCPIP0::192.168.0.11::inst0::INSTR',
 'ASRL1::INSTR',
 'ASRL10::INSTR'

Further, even if I put the visa address directly and try to open the GPIB visa resource, I get the error:
VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

I have also noticed that in NI MAX the NI visa path is C:\Windows\SysWOW64\visa32.dll, ans not C:\Windows\System32\visa32.dll as it is described in the pyvisa info. However, if I try to do:
rm = visa.ResourceManager(r'C:\Windows\SysWOW64\visa32.dll')
I get:
LibraryError: Error while accessing C:\Windows\SysWOW64\visa32.dll: [WinError 193] %1 is not a valid Win32 application

Any idea on what the problem might be?

Regards,
Diogo

MatthieuDartiailh, 2019-05-11:
Hi, I don't have a good idea of what may have caused such a change (in particular why updating Python would cause such a change). Are you using a GPIB adapter or a card ? The fact that NI-MAX picks this dll is probably related to the fact that it is still (?) a 32-bit application (Yes on Windows 64bits dll in System32 are 64 bits and the ones in WOW64 are 32, it is confusing). When you updated did you fully uninstall National instrument components and re-installed after ? if not, I would try to fully uninstall all National instrument related software and then cleanly re-install, but it is mostly because I don't have any better idea.

dcribeiro, 2019-05-13:
Hi Matthieu,
Thanks for your help. Indeed, visa32.dll in WOW64 is 32bit, I did not know that.
I also do not know what caused the issue, but visa32.dll or visa64.dll in System32 were not returning any GPIB connections.
It is not a pyvisa issue, it is a problem of these dlls, I cannot understand why that is.

For now, to save myself the trouble of a 'clean' full reinstalation of all the NI applications, I have installed the Keysight IO libraries and I can see the GPIB instruments again.

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.8
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 21 2019 18:30:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1376087058
         Spec. Version: 5243136
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1376087058
         Spec. Version: 5243136

 Instruments:
(visa) list
( 0) GPIB0::24::INSTR
( 1) RSNRP::0x0070::101472::INSTR
( 2) USB0::0x05FF::0x1023::3808N60697::INSTR
     alias: USBInstrument1
( 3) TCPIP0::192.168.0.11::inst0::INSTR
( 4) ASRL1::INSTR
( 5) ASRL10::INSTR
     alias: LPT1

I am happy again :)

Also, to answer your question, the GPIB is not an USB adapter, it is a native GPIB connection from the PXI computer on the PXI chassis I am using (probably kind of a PCI card GPIB implementation).

Once again, thank you for all the help.

MatthieuDartiailh, 2019-05-13:
@dcribeiro I am happy you fixed your issue. I guess somehow the install got corrupted or something, thanks for letting us know.

arthur00, 2019-05-30:
Sorry about that SyntaxError, that was me trying to replace the library_path variable on the debugger. I wasn't aware ResourceManager took a dll path as an argument - never had to use it before..  Same issue, different stack:

rm = visa.ResourceManager('C:\\Windows\\SysWOW64\\visa32.dll')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1526, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1498, in open_visa_library
    return cls(argument)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 108, in __new__
    obj._init()
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 139, in _init
    functions.set_signatures(self.lib, errcheck=self._return_handler)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 168, in set_signatures
    apply("viOpen", [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession], required=True)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 95, in _internal
    set_signature(library, function_name, argtypes, restype, errcheck_)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 241, in set_signature
    func = getattr(library, function_name)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\ctypes\__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\ctypes\__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'viOpen' not found

MatthieuDartiailh, 2019-05-30:
What seems weird to me is that usually one gets a different error on a bitness mismatch. If you get the occasion I would be curious to know if you get the same issue with National Instrument or if this is a Keysight only issue.

If you have the proper tooling installed (Visual Studio or http://www.dependencywalker.com/ for example), you could inspect the functions exported by the dll (using `DUMPBIN -EXPORTS XXX.DLL` if you have windows compiler).

arthur00, 2019-05-30:
File Type: DLL

  Section contains the following exports for visa32.dll

    00000000 characteristics
    5BCE1D30 time date stamp Mon Oct 22 11:55:44 2018
        0.00 version
         128 ordinal base
         168 number of functions
         107 number of names

    ordinal hint RVA      name

        215    0          viAssertIntrSignal (forwarded to visa64.viAssertIntrSignal)
        258    1          viAssertTrigger (forwarded to visa64.viAssertTrigger)
        214    2          viAssertUtilSignal (forwarded to visa64.viAssertUtilSignal)
        203    3          viBufRead (forwarded to visa64.viBufRead)
        202    4          viBufWrite (forwarded to visa64.viBufWrite)
        260    5          viClear (forwarded to visa64.viClear)
        132    6          viClose (forwarded to visa64.viClose)
        136    7          viDisableEvent (forwarded to visa64.viDisableEvent)
        137    8          viDiscardEvents (forwarded to visa64.viDiscardEvents)
        135    9          viEnableEvent (forwarded to visa64.viEnableEvent)
        130    A          viFindNext (forwarded to visa64.viFindNext)
        129    B          viFindRsrc (forwarded to visa64.viFindRsrc)
        268    C          viFlush (forwarded to visa64.viFlush)
        133    D          viGetAttribute (forwarded to visa64.viGetAttribute)
        128    E          viGetDefaultRM (forwarded to visa64.viGetDefaultRM)
        212    F          viGpibCommand (forwarded to visa64.viGpibCommand)
        210   10          viGpibControlATN (forwarded to visa64.viGpibControlATN)
        208   11          viGpibControlREN (forwarded to visa64.viGpibControlREN)
        213   12          viGpibPassControl (forwarded to visa64.viGpibPassControl)
        211   13          viGpibSendIFC (forwarded to visa64.viGpibSendIFC)
        261   14          viIn16 (forwarded to visa64.viIn16)
        224   15          viIn16Ex (forwarded to visa64.viIn16Ex)
        281   16          viIn32 (forwarded to visa64.viIn32)
        226   17          viIn32Ex (forwarded to visa64.viIn32Ex)
        220   18          viIn64 (forwarded to visa64.viIn64)
        228   19          viIn64Ex (forwarded to visa64.viIn64Ex)
        273   1A          viIn8 (forwarded to visa64.viIn8)
        222   1B          viIn8Ex (forwarded to visa64.viIn8Ex)
        139   1C          viInstallHandler (forwarded to visa64.viInstallHandler)
        144   1D          viLock (forwarded to visa64.viLock)
        263   1E          viMapAddress (forwarded to visa64.viMapAddress)
        242   1F          viMapAddressEx (forwarded to visa64.viMapAddressEx)
        216   20          viMapTrigger (forwarded to visa64.viMapTrigger)
        291   21          viMemAlloc (forwarded to visa64.viMemAlloc)
        243   22          viMemAllocEx (forwarded to visa64.viMemAllocEx)
        292   23          viMemFree (forwarded to visa64.viMemFree)
        244   24          viMemFreeEx (forwarded to visa64.viMemFreeEx)
        200   25          viMove (forwarded to visa64.viMove)
        201   26          viMoveAsync (forwarded to visa64.viMoveAsync)
        241   27          viMoveAsyncEx (forwarded to visa64.viMoveAsyncEx)
        240   28          viMoveEx (forwarded to visa64.viMoveEx)
        285   29          viMoveIn16 (forwarded to visa64.viMoveIn16)
        234   2A          viMoveIn16Ex (forwarded to visa64.viMoveIn16Ex)
        287   2B          viMoveIn32 (forwarded to visa64.viMoveIn32)
        236   2C          viMoveIn32Ex (forwarded to visa64.viMoveIn32Ex)
        230   2D          viMoveIn64 (forwarded to visa64.viMoveIn64)
        238   2E          viMoveIn64Ex (forwarded to visa64.viMoveIn64Ex)
        283   2F          viMoveIn8 (forwarded to visa64.viMoveIn8)
        232   30          viMoveIn8Ex (forwarded to visa64.viMoveIn8Ex)
        286   31          viMoveOut16 (forwarded to visa64.viMoveOut16)
        235   32          viMoveOut16Ex (forwarded to visa64.viMoveOut16Ex)
        288   33          viMoveOut32 (forwarded to visa64.viMoveOut32)
        237   34          viMoveOut32Ex (forwarded to visa64.viMoveOut32Ex)
        231   35          viMoveOut64 (forwarded to visa64.viMoveOut64)
        239   36          viMoveOut64Ex (forwarded to visa64.viMoveOut64Ex)
        284   37          viMoveOut8 (forwarded to visa64.viMoveOut8)
        233   38          viMoveOut8Ex (forwarded to visa64.viMoveOut8Ex)
[... 30 lines cut ...]
        134   57          viSetAttribute (forwarded to visa64.viSetAttribute)
        267   58          viSetBuf (forwarded to visa64.viSetBuf)
        142   59          viStatusDesc (forwarded to visa64.viStatusDesc)
        143   5A          viTerminate (forwarded to visa64.viTerminate)
        140   5B          viUninstallHandler (forwarded to visa64.viUninstallHandler)
        145   5C          viUnlock (forwarded to visa64.viUnlock)
        264   5D          viUnmapAddress (forwarded to visa64.viUnmapAddress)
        217   5E          viUnmapTrigger (forwarded to visa64.viUnmapTrigger)
        294   5F          viUsbControlIn (forwarded to visa64.viUsbControlIn)
        293   60          viUsbControlOut (forwarded to visa64.viUsbControlOut)
        270   61          viVPrintf (forwarded to visa64.viVPrintf)
        280   62          viVQueryf (forwarded to visa64.viVQueryf)
        205   63          viVSPrintf (forwarded to visa64.viVSPrintf)
        207   64          viVSScanf (forwarded to visa64.viVSScanf)
        272   65          viVScanf (forwarded to visa64.viVScanf)
        209   66          viVxiCommandQuery (forwarded to visa64.viVxiCommandQuery)
        138   67          viWaitOnEvent (forwarded to visa64.viWaitOnEvent)
        257   68          viWrite (forwarded to visa64.viWrite)
        278   69          viWriteAsync (forwarded to visa64.viWriteAsync)
        218   6A          viWriteFromFile (forwarded to visa64.viWriteFromFile)

  Summary

        2000 .rdata
        1000 .rsrc

MatthieuDartiailh, 2019-05-30:
Is that the DLL from system32 or SysWoW64 ? (just to know)

arthur00, 2019-05-30:
system32

arthur00, 2019-05-30:
I will just set PYVISA_LIBRARY to '@py' on my runtime - that is enough for me. :)

MatthieuDartiailh, 2019-05-30:
Thanks for your help. I will try to investigate a bit more but I am really puzzled by why we cannot get a function that appears to be there.

arthur00, 2019-05-30:
Here is my take on it: visa32 is actually a 32-bit library, which is why it doesn't fail bitness check. However attempting to call it's functions reroutes it visa64.dll. That's when the call fails, but at this point the bitness check already passed. Thus we only see the issue when an exception is thrown when trying to actually call viopen on visa64.

I'm not too familiar with forwarding DLLs myself, so that's my best guess.

MatthieuDartiailh, 2020-09-06:
Closing since it appears that the issue is on the VISA side in this case.  Feel free to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh