pyvisa/pyvisa #339: Mac OS 10.13.1 - Abort trap: 6

unvieilhomme, 2018-05-26:
Hi together

I tried to install and run pyvisa on my macbook (Mac OS 10.13.1).
`import visa` works fine but as i try to open a ResourceManager it aborts with "Abort trap: 6". Same happens if I try `python -m visa info`. 
I tried this with python 3.6.5 and 2.7.15, same result. I installed pyvisa with pip. I got NI-VISA 18.0 from ni.com. I configured the library path with the .pyvisarc file with content
"[Paths]
VISA library: /Library/Frameworks/VISA.framework/VISA". 

I googled but did not find much. Any idea how to solve this?

Thanks in advance

MatthieuDartiailh, 2018-05-26:
I do not usually work on Mac and I never heard about this. I will try to look into it. In the meantime you may find related issues by googling issues about ctypes and Mac.

philexfan, 2018-07-13:
I have this problem too. 

Before installing NI-Visa_18.0.0 for macOS 10.13.4, the information shows below after tying `python -m visa info`.

> Machine Details:
   Platform ID:    Darwin-17.5.0-x86_64-i386-64bit
   Processor:      i386
   Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.10
   Compiler:       GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)
   Bits:           64bit
   Build:          Oct  6 2017 22:29:07 (#default)
   Unicode:        UCS2
PyVISA Version: 1.9.0
Backends:
ni:
      Version: 1.9.0 (bundled with PyVISA)
      Binary library: Not found
py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

However, after NI-Visa_18.0.0 installation, after tying `python -m visa info` it shows abort trap:6. It looks like it is the problem of NI-VISA (/Library/Frameworks/nipalu.framework/Versions/1/nipalu).

Any idea to resolve this problem?

The full problem report:

Process:               Python [1593]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        bash [790]
Responsible:           Python [1593]
User ID:               783402280

PlugIn Path:             /Library/Frameworks/nipalu.framework/Versions/1/nipalu
PlugIn Identifier:       com.ni.framework.nipalu
PlugIn Version:          18.0.0 (18.0.0f0)

Date/Time:             2018-07-13 13:25:24.114 +0100
OS Version:            Mac OS X 10.13.4 (17E202)
Report Version:        12
Bridge OS Version:     3.0 (14Y664)
Anonymous UUID:        48737E74-FF6F-D986-6AA0-02111A2BD070

Time Awake Since Boot: 1000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

[... 360 lines cut ...]
TOTAL, minus reserved VM space   446.5M      585 

Model: MacBookPro14,2, BootROM MBP142.0173.B00, 2 processors, Intel Core i7, 3.5 GHz, 16 GB, SMC 2.44f1
Graphics: Intel Iris Plus Graphics 650, Intel Iris Plus Graphics 650, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x171), Broadcom BCM43xx 1.0 (7.77.37.29.1a7)
Bluetooth: Version 6.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci1b73,1100, USB eXtensible Host Controller, Thunderbolt@196,0,0
PCI Card: ethernet, Ethernet Controller, Thunderbolt@195,0,0
PCI Card: pci8086,15d4, USB eXtensible Host Controller, Thunderbolt@197,0,0
USB Device: USB 3.0 Bus
USB Device: iBridge
USB Device: USB 3.0 Bus
USB Device: USB audio CODEC
USB Device: USB Optical Mouse
USB Device: USB 3.1 Bus
USB Device: 4-Port USB 2.0 Hub
USB Device: BillBoard Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 33.1
Thunderbolt Device: Thunderbolt 3 Express Dock HD, Belkin International, Inc., 1, 19.1

Philex

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity and because it seems the issue lies with NI-VISA and not in PyVISA itself. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh