pyvisa/pyvisa #342: pip install on Redhat

tomcap0618, 2018-06-04:
Hello,

I am getting the following error when I am running pip install pyvisa.
$ ./pip install pyvisa  
Collecting pyvisa
  Using cached https://files.pythonhosted.org/packages/c2/bb/9f3885442308b5a3aa916d0426113ae2034949a2acc4756a27f0a3a6ca8a/PyVISA-1.9.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/tmp/pip-build-77saj9kt/pyvisa/setup.py", line 23, in <module>
        long_description = '\n\n'.join([read('README'),
      File "/tmp/pip-build-77saj9kt/pyvisa/setup.py", line 20, in read
        return f.read()
      File "<File Path>/python-3.5.1/lib/python3.5/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 964: ordinal not in range(128)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-77saj9kt/pyvisa

The files being read are the README, AUTHORS, and LICENSE File.
The 964th and 1061st characters  (LabVIEW's and vendor's) in the README are apostrophes. These are different than the 683rd character (90'). I think that if the apostrophe style is changed, the bug will be fixed.

Regards,
Tom

tomcap0618, 2018-06-04:
This error also exists in AUTHORS in Tobias's last name.

tomcap0618, 2018-06-04:
I modified the 3 locations locally, and I was able to successfully install using pip and a downloaded clone of master.

MatthieuDartiailh, 2018-06-05:
I find that a bit surprising since we set manually the default encoding to utf-8 to avoid that kind of issue I believe. 
Could you try opening with 'rb' option and decoding using utf-8 manually ?

tomcap0618, 2018-06-05:
Hi Matthieu,
I think you are correct.
My default OS user language was C.
When I set LANG="en_US.UTF-8", the pip install pyvisa command worked as expected.
Sorry about the mix-up.
At least it is documented now, in this thread for anyone in the future.
Regards,
Tom

MatthieuDartiailh, 2018-06-05:
Could you see if #343 fix the issue for you ? I don't think we should/need to mess up with the encoding used by sys.

MatthieuDartiailh, 2018-06-08:
ping @tomcap0618

MatthieuDartiailh, 2018-07-04:
ping @tomcap0618

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh