pyvisa/pyvisa #344: How to reopen visa resource manager after connection of instruments lost suddently?

isanmu, 2018-06-20:
I init the visa instrument with bellow codes:
`self.rm = visa.ResourceManager()
self.addr_res = self.rm.list_resources()
for addr in self.addr_res:
      try:
            name_res = self.rm.open_resource(addr).query('*IDN?')
            ......`

Here's the question:
      When i disconnect(artificial or non-artificial reason) the GPIB(usb) connection with computer, the visa instruments was closed(did them really closed?) incorrectly, so, I want to reopen the ResourceManager again to refresh the Instruments after reconnecting the GPIB instruments. But it didn't work,  self.rm.list_resources() cannot regain the instruments info.

Before reopen ResourceManager , if i close the instruments that open before, errors will caused as bellow:  
`<class 'pyvisa.errors.VisaIOError'>, VisaIOError('VI_ERROR_CONN_LOST (-1073807194): The connection for the given session has been lost.',`

I also try the code as below to close ResourceManager before reopenning it.
`self.rm.close()`
but it didn't work also.

Asking for help:
        How to reopen visa resource manager after the connection of instruments lost suddently?

isanmu, 2018-06-20:
@MatthieuDartiailh 
### **visa info:**
```
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 142 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\user\PycharmProjects\Test\venv32\Scripts\python.exe
   Version:        3.6.5
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 28 2018 16:07:46 (#v3.6.5:f59c0932b4)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
```
### **Reply:**

- > if I disconnect an instrument it will keep showing up in the list

I use PYQT5 treeWidget to list my instruments:
```
def update_instr_list(window, instr_name_list):
    window.treeWidget.clear()
    root = QtWidgets.QTreeWidgetItem(window.treeWidget)
    root.setText(0, 'Instruments')
    # all_instr.update()
    print(instr_name_list)
    for num in range(len(instr_name_list)):
        child_ele = "child" + str(num)
        # print(child_ele)
        child_ele = QtWidgets.QTreeWidgetItem(root)
        child_ele.setText(0, instr_name_list[num])
```
So, the instruments will keep showing up in the treewidget if i didn't update resource lists and recall the update_instr_list() function after detaching the USB cable. 

- > Out of curiosity can you connect to the instrument even though it does not show up in the list_resources

No, i can not connect the instruments, that's my i want to reopen the `ResourceManager` to regain `list_resource `and reopen the instruments (`self.rm.open_resource(addr)`), than i can recall the update_instr_list() function to list my instruments and control them. **But, if i disconnect the USB cable, and connect again without close the application**, i have no idea how to reopen the instruments.

MatthieuDartiailh, 2018-06-20:
I do not have a National Instrument VISA at hand but I will try to test later.
In the meantime, just to be sure we understand each other, this works for me:
```python
rm = visa.ResourceManager()
addr = rm.list_resources()[0]  # I have a single USB instrument connected
i = rm.open_resource(addr)
print(i.query('*IDN?')
# Unplug the device, replug the device
print(i.query('*IDN?')

# Unplug again, replug
i = rm.open_resource(addr)
print(i.query('*IDN?')
```

isanmu, 2018-06-20:
Yes, the code you test is exactly the function that i want to realize!
I will try the `rm.open_resource(addr)` directly later... since i use
```
addr = rm.list_resources()[0]
i = rm.open_resource(addr)
```
instead but didn't work. 
Thanks, wait for your test result.

isanmu, 2018-09-07:
Hi, @MatthieuDartiailh: 
### Here I feedback some updated information.

**Update**： 
This problem would **occur** with _GPIB-Instruments_ connected, while **disappear** with the _USB-Instruments_ connected.
So, I think that's why your test result was different from mine.  I look forward to your feedbacks and suggestions soon.

MatthieuDartiailh, 2018-09-08:
The GPIB can be a bit painful in that kind of situation. You could try to not only close the resource manager but also close the underlying link to the VISA library. However there is no exposed way to do that. So try before closing the resource manager, get a reference to the library (library attribute), then close the resource manager, and next clear the `_registry` dict of the library and delete the library. Check that the logger shows again `'Created library wrapper ...'` when you re-create the resource manager. If that works it may be worth adding a utility somewhere to do that.

isanmu, 2018-09-10:
@MatthieuDartiailh 
### Update Infromation:
```python
>>>import visa
>>>rm = visa.ResourceManager()
>>>addr = rm.list_resources()[1]
>>>i = rm.open_resource(addr)
>>>print(i.query('*IDN?'))
CHROMA,62012P-100-50,03.20,02633
```
**# Unplug the device, replug the device**

```python
>>>rm.close()
>>>rm.visalib._registery.clear()
>>>del rm.visalib
>>>del rm
>>>rm = visa.ResourceManager()
>>>rm.list_resources()
('Some USB/TCPIP-instruments', )
```
**Result:**
The logger did show the strings: 'Created library wrapper ...', **but the rm.list_resources() doesn't contain the GPIB instrument that i unpluged and repluged before**.
 Is there anything that i did't do or did wrong?

MatthieuDartiailh, 2018-09-10:
Could you post the full tracebacks to help me see if we can do better ? And the log too ?

isanmu, 2018-09-11:
@MatthieuDartiailh 
### Update Information Here:
```python
>>>import visa
>>>visa.log_to_screen()
>>>rm = visa.ResourceManager()
>>>rm.list_resources()
('TCPIP0::C302153-7KC::inst0::INSTR', 'GPIB0::1::INSTR')
>>>i = rm.open_resource(rm.list_resources()[1])
```
- Unplug the device, replug the device
```python
>>>rm.list_resources()
('TCPIP0::C302153-7KC::inst0::INSTR', )
>>>rm.close()
2018-09-11 08:56:11,793 - pyvisa - DEBUG - Closing ResourceManager (session: 72735592)
2018-09-11 08:56:11,803 - pyvisa - DEBUG - viClose(72735592,) -> -1073807338
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\user\PycharmProjects\Test\venv32\lib\site-packages\pyvisa\highlevel.py", line 1559, in close
    self.visalib.close(self.session)
  File "C:\Users\user\PycharmProjects\Test\venv32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 394, in close
    return library.viClose(session)
  File "C:\Users\user\PycharmProjects\Test\venv32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_CLOSING_FAILED (-1073807338): The VISA driver failed to properly close the session or object reference. This might be due to an error freeing internal or OS resources, a failed network connection, or a lower-level driver or OS error.
2018-09-11 08:56:11,833 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B670>) -> -1073807346
2018-09-11 08:56:11,833 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B6C0>) -> -1073807346
```

- I can understand this error here for disconnecting the  instrument without closing normally. 
```python
>>>rm.visalib._registry.clear()
2018-09-11 08:56:20,267 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B760>) -> -1073807346
2018-09-11 08:56:20,267 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B8F0>) -> -1073807346
>>>rm = visa.ResourceManager()
2018-09-11 08:56:28,802 - pyvisa - DEBUG - No visa libaray specified and environment variable PYVISA_LIBRARY is unset. Using NI library
2018-09-11 08:56:28,802 - pyvisa - DEBUG - No user defined library files
2018-09-11 08:56:28,812 - pyvisa - DEBUG - Automatically found library files: ['C:\\Windows\\system32\\visa32.dll']
2018-09-11 08:56:28,812 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2018-09-11 08:56:28,812 - pyvisa - DEBUG - Created library wrapper for C:\Windows\system32\visa32.dll
2018-09-11 08:56:28,812 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x07A4B850>',) -> 0
2018-09-11 08:56:28,812 - pyvisa - DEBUG - Created ResourceManager with session 71424704
2018-09-11 08:56:28,842 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B710>) -> -1073807346
2018-09-11 08:56:28,842 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B670>) -> -1073807346
>>>rm.list_resources()
('TCPIP0::C302153-7KC::inst0::INSTR', 'GPIB0::1::INSTR')
2018-09-11 08:56:37,536 - pyvisa - DEBUG - viFindRsrc(71424704, '?*::INSTR', '<ViObject object at 0x07A4B850>', 'c_ulong(2)', <ctypes.c_char_Array_256 object at 0x07A4B8F0>) -> 0
2018-09-11 08:56:37,536 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x07A4B850>, <ctypes.c_char_Array_256 object at 0x07A4B8F0>) -> 0
2018-09-11 08:56:37,536 - pyvisa - DEBUG - viClose(<ViObject object at 0x07A4B850>,) -> 0
2018-09-11 08:56:37,556 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B850>) -> -1073807346
2018-09-11 08:56:37,556 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B620>) -> -1073807346
>>>i = rm.open_resource(rm.list_resources()[1])
2018-09-11 08:56:48,541 - pyvisa - DEBUG - viFindRsrc(71424704, '?*::INSTR', '<ViObject object at 0x07A4B850>', 'c_ulong(2)', <ctypes.c_char_Array_256 object at 0x07A4B710>) -> 0
2018-09-11 08:56:48,541 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x07A4B850>, <ctypes.c_char_Array_256 object at 0x07A4B710>) -> 0
2018-09-11 08:56:48,541 - pyvisa - DEBUG - viClose(<ViObject object at 0x07A4B850>,) -> 0
2018-09-11 08:56:48,541 - pyvisa - DEBUG - viParseRsrcEx(71424704, 'GPIB0::1::INSTR', 'c_ushort(1)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x07A4B8F0>, <ctypes.c_char_Array_256 object at 0x07A4B670>, <ctypes.c_char_Array_256 object at 0x07A4B620>) -> 0
2018-09-11 08:56:48,541 - pyvisa - DEBUG - GPIB0::1::INSTR - opening ...
2018-09-11 08:56:48,541 - pyvisa - DEBUG - viOpen(71424704, 'GPIB0::1::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x07A4B620>') -> 0
2018-09-11 08:56:48,541 - pyvisa - DEBUG - GPIB0::1::INSTR - is open with session 71380768
2018-09-11 08:56:48,551 - pyvisa - DEBUG - GPIB0::1::INSTR - closing
2018-09-11 08:56:48,551 - pyvisa - DEBUG - viDisableEvent(71412064, 1073709055, 65535) -> -1073807346
Exception ignored in: 2018-09-11 08:56:48,551 - pyvisa - DEBUG - viGetAttribute(71412064, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B620>) -> -1073807346
<object repr() failed>
Traceback (most recent call last):
  File "C:\Users\user\PycharmProjects\Test\venv32\lib\site-packages\pyvisa\resources\resource.py", line 108, in __del__
    self.close()
[... cut ...]
2018-09-11 08:56:55,469 - pyvisa - DEBUG - viDisableEvent(71380768, 1073709055, 65535) -> 1073676291
2018-09-11 08:56:55,469 - pyvisa - DEBUG - viDiscardEvents(71380768, 1073709055, 65535) -> 1073676292
2018-09-11 08:56:55,469 - pyvisa - DEBUG - viClose(71380768,) -> 0
2018-09-11 08:56:55,469 - pyvisa - DEBUG - GPIB0::1::INSTR - is closed
2018-09-11 08:56:55,499 - pyvisa - DEBUG - viGetAttribute(71503848, 3221159938, <ctypes.c_char_Array_256 object at 0x07A4B710>) -> 0
>>>
```

- **At last, i need to open instrument 'GPIB::1::INSTR' twice then this GPIB-Instrument could open correctly.**

MatthieuDartiailh, 2018-09-16:
Out of curiosity can you communicate with the instrument after the first re-opening. I am asking because the actual error is ignored and only show up in the log. There may be a good reason to ignore it, and it may not cause any actual issue.

isanmu, 2018-09-17:
@MatthieuDartiailh 
Unfortunately, after the first re-opening, i can not control the instrument normally!(**Sometimes**)
For Example, if i just open once, when i control instrument[0], actually the cmd may send to instrument[1] or other, so, it will be disordered.

edupo, 2020-07-10:
We are receiving this error on our Windows 10 machines after the machine goes to sleep.
We are using a USB to GPIB interface from NI.
To fix it we must disconnect and then connect the device and then the error is gone.

Just in case this gives any additional clue.

MatthieuDartiailh, 2020-07-10:
Letting windows goes to sleep may have some weird effect on the VISA library for sure. I would try to avoid having this happening while connections are opened if possible.

MatthieuDartiailh, 2020-09-06:
The overall take of the issue is that violent disconnection of GPIB connection can be hard to recover and may require a computer/instrument restart. Windows sleep also interact poorly with VISA opened connections.

Closing for now, feel free to reopen.

closed 2020-09-06 by MatthieuDartiailh