pyvisa/pyvisa #346: Unknown Visa IO error code

tddaniel, 2018-06-29:
Not sure if this is the right place but maybe someone can point me in the right direction.

Attempting to control a Lecroy 9430 scope via gpib-usb-hs. This scope has been controlled 
this way before. Python sees the resource, connects, but any query, write, or read throws

```
VisaIOError: ? (-1073807190): Unknown code.
```

minimal code to replicate, the scope is at board 0 address 6.
```
import pyvisa 
rm = pyvisa.ResourceManager()
scope = rm.open_resource('GPIB0::6::INSTR')
scope.query('*IDN?')
```

system info
```
Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 78 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\timothy\Anaconda3\python.exe
   Version:        3.6.5
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 29 2018 13:32:41 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 18874368
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 18874368
         Spec. Version: 5244928
```

MatthieuDartiailh, 2018-07-04:
I have no real idea what is happening here... Did you restart after installing National Instrument VISA ? Also did you let the installer disable the fast Windows startup ? The fast startup is known to mess up with NI VISA. 
Can you try uninstalling VISA and re-installing it ?

tddaniel, 2018-07-09:
The issue turns out to be on the National Instruments side of things. The NI back end thinks my GPIB-USB device is not a genuine NI device. I believe it is genuine but from what I can tell they wont do anything about it. I am not sure if there is a way for me to ignore this error.

Thanks for the help.

MatthieuDartiailh, 2020-09-06:
Closing since there is nothing we can do on PyVISA side to improve the situation.

closed 2020-09-06 by MatthieuDartiailh

Frank-Duffy, 2022-04-07:
> The issue turns out to be on the National Instruments side of things. The NI back end thinks my GPIB-USB device is not a genuine NI device. I believe it is genuine but from what I can tell they wont do anything about it. I am not sure if there is a way for me to ignore this error.
> 
> Thanks for the help.
> 
> Edit: I can still use the device on linux via linux-gpib so that is what I am doing for now.

How is it that you came to this conclusion? I am receiving the same error code from pyvisa, but I have two other gpib adapters of the same make and model which are working. So it seems like it's a hardware issue for me but I'm not sure.