pyvisa/pyvisa #347: Only reading every second character

hemanti0503, 2018-07-10:
Hi,

I'm trying to control a IFR 2026B signal generator using the Agilent 82357B USB-GPIB adapter on Raspbian Stretch. After compiling `linux-gpib`, I can confirm that the connection works using `ibtest`:

> $ ibtest
> [...]
> enter primary gpib address for device you wish to open [0-30]: 2
> trying to open pad = 2 on /dev/gpib0 ...
> [...]
> enter a string to send to your device: *IDN?
> sending string: *IDN?
> 
> gpib status is: 
> ibsta = 0x2100  < END CMPL >
> iberr= 0
> 
> ibcntl = 6
> [...]
> enter maximum number of bytes to read [1024]: 
> trying to read 1024 bytes from device...
> received string: 'IFR,2026B,202601/844,44540/044/01.08
> '
> Number of bytes read: 37
> gpib status is: 
> ibsta = 0x2100  < END CMPL >
> iberr= 0
> 
> ibcntl = 37

However, using `pyVisa`'s `query` or `read` method, only every second character is read:

```
import visa
visa.log_to_screen()
rm = visa.ResourceManager('@py')
rm.list_resources()
2018-07-10 16:58:49,100 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-07-10 16:58:49,326 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-07-10 16:58:49,384 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2018-07-10 16:58:49,422 - pyvisa - DEBUG - GPIBSession was correctly imported.
2018-07-10 16:58:49,429 - pyvisa - DEBUG - Created library wrapper for unset
2018-07-10 16:58:49,441 - pyvisa - DEBUG - Created ResourceManager with session 9227923
('ASRL/dev/ttyAMA0::INSTR',)

sg = rm.open_resource('GPIB::02::INSTR')
2018-07-10 16:58:52,241 - pyvisa - DEBUG - GPIB::02::INSTR - opening ...
2018-07-10 16:58:52,248 - pyvisa - DEBUG - GPIB::02::INSTR - is open with session 5808513

print(sg.query('*IDN?'))
2018-07-10 16:58:59,162 - pyvisa - DEBUG - GPIB.write b'*IDN?\r\n'
2018-07-10 16:58:59,171 - pyvisa - DEBUG - GPIB::02::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
IR22B220/4,44/4/10

sg.chunk_size = 102400
sg.write('*IDN?')
2018-07-10 16:59:00,853 - pyvisa - DEBUG - GPIB.write b'*IDN?\r\n'
<StatusCode.success: 0>

sg.read_raw()
2018-07-10 16:59:02,414 - pyvisa - DEBUG - GPIB::02::INSTR - reading 102400 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
b'IR22B220/4,44/4/10\n'
```

`rm.list_resources` doesn't list my device, but the Keysight manager in Windows doesn't either, so I'm not too concerned about it as long as I can open it directly by specifying the address. However, you can see the problem about the reads above.

Here is some more information about my setup:

> $ python3 -m visa info
> Machine Details:
>    Platform ID:    Linux-4.14.50-v7+-armv7l-with-debian-9.4
>    Processor:      
> 
> Python:
>    Implementation: CPython
>    Executable:     /usr/bin/python3
>    Version:        3.5.3
>    Compiler:       GCC 6.3.0 20170124
>    Bits:           32bit
>    Build:          Jan 19 2017 14:11:04 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.9.0
> 
> Backends:
>    ni:
>       Version: 1.9.0 (bundled with PyVISA)
>       Binary library: Not found
>    py:
>       Version: 0.2
>       TCPIP INSTR: Available 
>       USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
>       GPIB INSTR: Available 
>       ASRL INSTR: Available via PySerial (3.2.1)
>       TCPIP SOCKET: Available 
>       USB RAW: Available via PyUSB (1.0.2). Backend: libusb1

Is there some configuration, that I've missed or some parameter I should set? Seems really weird to me, since in `ibtest` everything works correctly without any extra configuration. Please let me know if you need any further information.

Thank you very much for your help!

Henrik

MatthieuDartiailh, 2018-07-10:
I have no real idea about what is going on here. Could you try reading a byte at a time using the `read_bytes` method ?

hemanti0503, 2018-07-11:
Fist of all, thank you for your help!

Unfortunately, this also leaves out every second character:

```
>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-07-11 09:43:43,539 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-07-11 09:43:43,660 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-07-11 09:43:43,686 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2018-07-11 09:43:43,695 - pyvisa - DEBUG - GPIBSession was correctly imported.
2018-07-11 09:43:43,697 - pyvisa - DEBUG - Created library wrapper for unset
2018-07-11 09:43:43,698 - pyvisa - DEBUG - Created ResourceManager with session 6814469
>>> sg = rm.open_resource('GPIB::02::INSTR')
2018-07-11 09:43:52,415 - pyvisa - DEBUG - GPIB::02::INSTR - opening ...
2018-07-11 09:43:52,424 - pyvisa - DEBUG - GPIB::02::INSTR - is open with session 4015412
>>> sg.write('*IDN?')
2018-07-11 09:44:01,550 - pyvisa - DEBUG - GPIB.write b'*IDN?\r\n'
<StatusCode.success: 0>
>>> sg.read_bytes(1)
2018-07-11 09:44:06,126 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'I'
>>> sg.read_bytes(1)
2018-07-11 09:44:07,630 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'R'
>>> sg.read_bytes(1)
2018-07-11 09:44:08,606 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'2'
>>> sg.read_bytes(1)
2018-07-11 09:44:09,406 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'2'
>>> sg.read_bytes(1)
2018-07-11 09:44:10,174 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'B'
>>> sg.read_bytes(1)
2018-07-11 09:44:10,894 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'2'
>>> sg.read_bytes(1)
2018-07-11 09:44:11,630 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'2'
>>> sg.read_bytes(1)
2018-07-11 09:44:12,286 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'0'
>>> sg.read_bytes(1)
2018-07-11 09:44:12,894 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'/'
>>> sg.read_bytes(1)
2018-07-11 09:44:13,582 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'4'
>>> sg.read_bytes(1)
2018-07-11 09:44:14,286 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b','
>>> sg.read_bytes(1)
2018-07-11 09:44:14,895 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'4'
>>> sg.read_bytes(1)
2018-07-11 09:44:15,486 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'4'
>>> sg.read_bytes(1)
2018-07-11 09:44:15,950 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'/'
>>> sg.read_bytes(1)
2018-07-11 09:44:16,398 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'4'
>>> sg.read_bytes(1)
2018-07-11 09:44:16,926 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'/'
>>> sg.read_bytes(1)
2018-07-11 09:44:17,582 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'1'
>>> sg.read_bytes(1)
2018-07-11 09:44:18,334 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'0'
>>> sg.read_bytes(1)
2018-07-11 09:44:19,102 - pyvisa - DEBUG - GPIB::02::INSTR - reading 1 bytes (last status None)
b'\n'
```

hemanti0503, 2018-07-12:
As a side note, using the Python module installed with linux-gpib, it works flawlessly. So I'm using that for now. If you need any more info, I will be happy to provide you.