pyvisa/pyvisa #348: query_binary_values times out in pyvisa 1.9.0 but does not in pyvisa 1.8.0

astafan8, 2018-07-10:
## Introduction

In [QCoDeS](https://github.com/QCoDeS/Qcodes), we use `pyvisa` to implement communication with different instruments. We are currently using `pyvisa 1.8`.

Since `pyvisa 1.9.0` has been out for quite a long time, we have decided to start migrating to it. But during the process of testing `pyvisa 1.9.0`, I have found a bug.

## Problem description

One of our drivers uses `query_binary_values` method of the `MessageBasedResource` class, [see the line with the call in the code of our driver here](https://github.com/QCoDeS/Qcodes/blob/fb574ef1cf2b7a76cdab5d6591cb4f7a5eced376/qcodes/instrument_drivers/stanford_research/SR86x.py#L467). This method works well with `pyvisa 1.8`, while it times out for `pyvisa 1.9.0` with the following call stack (this in an excerpt from the pyvisa log):
```
...
2018-07-10 17:27:08,701 - pyvisa - DEBUG - viWrite(2, b'CAPTUREGET? 0, 2\n', 17, 'c_ulong(17)') -> 0
2018-07-10 17:27:08,701 - pyvisa - DEBUG - GPIB0::4::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-10 17:27:08,711 - pyvisa - DEBUG - viRead(2, <ctypes.c_char_Array_20480 object at 0x00000116117AFF48>, 20480, 'c_ulong(155)') -> 0
2018-07-10 17:27:08,711 - pyvisa - DEBUG - GPIB0::4::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-10 17:27:08,714 - pyvisa - DEBUG - viRead(2, <ctypes.c_char_Array_20480 object at 0x00000116117AFF48>, 20480, 'c_ulong(1899)') -> 0
2018-07-10 17:27:08,714 - pyvisa - DEBUG - GPIB0::4::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-10 17:27:18,735 - pyvisa - DEBUG - viRead(2, <ctypes.c_char_Array_20480 object at 0x00000116117AFF48>, 20480, 'c_ulong(0)') -> -1073807339
2018-07-10 17:27:18,736 - pyvisa - DEBUG - GPIB0::4::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
...
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 653, in query_binary_values
    header_fmt)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 482, in read_binary_values
    block.extend(self._read_raw())
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
...
```

## Possible cause

I debugged my code with both versions of `pyvisa`, and here is what I've found.

The `query_binary_values` method in `pyvisa 1.8` has a slightly different implementation than in `pyvisa 1.9.0`. In `1.8`, the method is self-sufficient, while in `1.9.0`, a new method has been introduced called `read_binary_values`. When comparing the "functionality" of the code in `query_binary_values` from `1.8` with the `read_binary_values` from `1.9.0`, I noticed that the latter [does append the length of the read terminator](https://github.com/pyvisa/pyvisa/blob/1363f48b9356a912fc84dd8af2f276ba43dfb8a2/pyvisa/resources/messagebased.py#L479) to the `expected_length` of the bytes array that is being read. In order to compare, just open [pyvisa/resources/messagebased.py from pyvisa 1.9.0 at line 467](https://github.com/pyvisa/pyvisa/blob/1363f48b9356a912fc84dd8af2f276ba43dfb8a2/pyvisa/resources/messagebased.py#L467) next to [pyvisa/resources/messagebased.py from pyvisa 1.8 at line 481](https://github.com/pyvisa/pyvisa/blob/a5c3cbf9b15ec3f61547dc5ba6d1d3c427c29f70/pyvisa/resources/messagebased.py#L481). While debugging, I was going line by line, and I was also monitoring what is written to the log, and I found out that indeed, the instrument makes `N` bytes available for reading, and then `pyvisa` reads all of them, but then, when it tries to read the remaining `1` byte (because the `expected_length == N + 1`), the instrument times out.

## Proposed solution

My suggestion for the fix is to remove the following lines from the `read_binary_values` method in `pyvisa 1.9.0`:
```python
if self._read_termination is not None:
    expected_length += len(self._read_termination)
```
So far as I understand, the termination characters are not involved when the binary data is read. On the other hand, I obviously do not know and do not have the competence to understand why would these two lines be introduced in the `pyvisa 1.9.0`; thus, if there is a valid reason for this, then the fix might be different.

## Conclusion

As a temporary fix in my code, I am going to change `_read_termination` to `None` before calling `query_binary_values`, and revert it back after `query_binary_values` returns (only if `pyvisa.__version__ == '1.9.0'`).

Please, let me know when the fix is going to be released. Thank you in advance!

## Debug info

Debug info generated by `pyvisa` for `pyvisa==1.9.0`:
```
Machine Details:
   Platform ID:    Windows-10-10.0.14393-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel
Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4
PyVISA Version: 1.9.0
Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
```

Debug info generated by `pyvisa` for `pyvisa==1.8`:
```
Machine Details:
   Platform ID:    Windows-10-10.0.14393-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel
Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4
PyVISA Version: 1.8
Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
```

MatthieuDartiailh, 2018-07-10:
Thanks for the report. It turns out that this line was added because many instruments actually return the termination character even after a binary block, and not reading it was causing some weird issues (when the last part of the binary block is also a termination character in particular). But since yours does not, it is probably worth adding a `expect_termination` optional argument True by default controlling this. Could you write a PR ?

astafan8, 2018-07-10:
@MatthieuDartiailh Thanks for the response.

But what about `read_binary` function? it contains very similar implementation, but it does not account for the read termination neither in pyvisa 1.8 nor in 1.9.0.

And what are those instruments that you are mentioning? Do you mean that they do not work with pyvisa 1.8? (I also expected to read a note about this in the CHANGES file)

Yes, I can make a PR once we define what to do exactly.

MatthieuDartiailh, 2018-07-10:
The change was added by @thliebig (I do not know what instrument he was using. But I believe I tested the change on a Keysight PNA) and I did not notice the inconsistency. I propose the following:
- move the code from `query\_binary\_values` `read\_binary\_values`
- add expect_termination (on both methods) but with False as default to recover the old behavior.
- add an entry to the changelog under 1.9.1
If you can make such a PR I will do my best to review it, merge it and make a bugfix release shortly (possibly before next week).
As a side note PyVISA is usually short on testers and since QCoDeS us a large user it would be nice to have one person from QCoDeS providing feedback.

MatthieuDartiailh, 2018-07-11:
Going through both implementations, I am bothered by the following points. In both implementations we use `_read_raw` which keeps going till it encounters a termination character (since the term char is enabled in VISA). So if your instrument does not send one, I am not sure to get why the reading complete. Furthermore it means that if the instrument never send a termination character we can never read the binary message which breaks the assumption under which this code is written.

astafan8, 2018-07-16:
Hmm. Could it be that the parsing functions (for example, `util.from_ieee_block`) are smart enough to exclude the read termination from the read data block?

Also note that in `read_binary_values`, an `expected_length` is defined via parsing (for example, `util.parse_ieee_block_header`) the very first block that is read. Since `pyvisa` `1.8` worked for me, I guess, that the parsing functions are smart enough to account for read termination. Do you think this makes sense?

MatthieuDartiailh, 2018-07-24:
Hi @astafan8,
Sorry for the delay I was on holiday last week. 
Giving a closer look at the parsing functions, it turns out that when we compute the array size we round the result (https://github.com/pyvisa/pyvisa/blob/master/pyvisa/util.py#L457) which means that if we expect data which are longer than a single byte, we can easily get rid of the termination character.
But, since this just guessing, could you check on your end that the length of the block is not a multiple of the size of the datatype ? If you can send me the raw data block it would be perfect.

astafan8, 2018-07-25:
Dear @MatthieuDartiailh , no worries :)

So, I debugged again the code with `pyvisa` `1.9.0`. Here is the summary of what I see with my comments (you will also find the binary block there):

```python
# this is the binary block that is read by the first `_read_raw` call
# By coincidence, this is just the only data that my instrument is supposed to return.
block_read_raw =bytearray(
    b'#42048gff\xb2\x9a\x99\xc3\xb243\x08\xb3\x00\x00+\xb3\x01'
    b'\x00L\xb3\x01\x00j\xb3\x00\x80\x82\xb3\xcd\xcc\x8e\xb3\xcdL'
    b'\x99\xb3\xcdL\xa2\xb3\x9a\x19\xaa\xb3\x9a\x19\xb0\xb3\x9a\x99'
    b'\xb4\xb3\x9a\x99\xb7\xb3\xcd\xcc\xb8\xb3\x00\x80\xb8\xb34\xb3'
    b'\xb6\xb3gf\xb3\xb3\xcdL\xae\xb34\xb3\xa7\xb3\xcdL\x9f\xb34\xb3'
    b'\x95\xb3\xcdL\x8a\xb3gf{\xb3\x9a\x99^\xb3\xcd\xcc>\xb3\x9a\x99'
    b'\x1c\xb3\xcd\xcc\xee\xb2\x9a\x99\x9f\xb2\xcd\xcc\x14\xb2\x9a\x99'
    b'\xf90\xcd\xcc\\243\xd32\xcd\xcc\x1d3\xcd\xccS3\x9a\x19\x86343\xa33'
    b'\x01\x80\xc13gf\xe03g\xe6\xff3\xcdL\x104\xcd\xcc 4\x00\xc014\x9a\xd9B4'
    b'\x01@T4\xcd\xcce4g\xa6w44\xd3\x8444\xd3\x8d4g\xe6\x964\xcd\x0c\xa0443'
    b'\xa94gF\xb24\xcdl\xbb4\x01\x80\xc444\x93\xcd44\x93\xd64\x01\x80\xdf4'
    b'\x9aY\xe84\x01 \xf14\x01\xc0\xf94g&\x0154c\x0554\x83\t5\x9a\x99\r5\xcd'
    b'\x9c\x115\xcd\x8c\x155\x00`\x195\x00 \x1d5\xcd\xcc 5\xcd\\$5\x9a'
    b'\xd9\'54C+5g\x86.5g\xb615\x9a\xc945\x00\xc0754\xa3:5\x9ai=54\x13@5'
    b'\x01\xa0B5\x9a\x19E5\xcdlG5\xcd\xacI5\x01\xd0K5g\xd6M5\x01\xc0O5g'
    b'\x96Q5\x01PS5\xcd\xecT5gvV54\xe3W5\xcd<Y5\x9ayZ54\xa3[5\x01\xb0\\5\x9a'
    b'\xa9]5\x01\x90^5\x9aY_5\x9a\x19`5\xcd\xbc`5gVa54\xd3a5gFb5\xcd\x9cb5'
    b'\x9a\xe9b54#c5\x9aIc5gfc5\x01pc5\x01pc5gfc5\x01@c5\x9a\x19c5\x01\xe0b5'
    b'\xcd\x9cb5\x01Pb5\x9a\xf9a5\x01\x90a5g&a54\xb3`5\xcd,`5g\xa6_5g\x16_5g'
    b'\x86^54\xe3]5\x01@]5\xcd\x9c\\5g\xe6[5\x9a9[5\x9ayZ54\xc3Y5\x9a\xf9X5'
    b'\x9a9X5\x01pW5g\xa6V54\xd3U5\x9a\tU5g6T54cS5\x9a\x99R5g\xc6Q54\xf3P5'
    b'\x01 P5\xcdLO54\x83N5\x9a\xb9M5\x01\xf0L5g&L5gfK5g\xa6J5g\xe6I5\x010I5'
    b'\x9ayH5\xcd\xccG5\x9a)G5g\x86F54\xe3E5\x9aIE5\x9a\xb9D543D5\xcd\xacC5'
    b'\x9a9C5\xcd\xbcB54SB5\x9a\xe9A54\x93A5\xcd<A5g\xe6@54\xa3@5\x01`@5g&@5g'
    b'\xf6?5\x00\xd0?5\x9a\xa9?5\xcd\x8c?5\x9ay?5gf?5\xcd\\?5\xcd\\?5\xcd\\?5'
    b'\xcd\\?5\xcd\\?5gf?5\x9ay?54\x83?5g\x96?5\x00\xa0?54\xb3?5\xcd\xbc?5\x00'
    b'\xd0?5\x9a\xd9?54\xe3?54\xe3?54\xe3?5\x9a\xd9?5g\xc6?54\xb3?5\xcd'
    b'\x8c?5gf?5g6?54\xf3>5g\xa6>5\x00P>5g\xe6=54s=5\xcd\xec<5\xcd\\<5\x9a'
    b'\xb9;54\x03;5\x9a9:5\xcd\\95\xcdl85\x9ai754S65\x00 554\xe335\x9a\x8925'
    b'\xcd\x1c154\x93/5g\xf6-5gF,54\x83*54\xa3(5\x00\xb0&5\x9a\xa9$5g\x86"5'
    b'\x9aY 5\x00\x10\x1e5\x9a\xa9\x1b5\x9a9\x195g\xb6\x165\x00 \x145gv\x115'
    b'\x9a\xb9\x0e5\x9a\xe9\x0b5\x00\x10\t54#\x065\xcd,\x035\xcd,\x00543\xfa4'
    b'\x9a\xf9\xf34\x01\xc0\xed4\x01`\xe74\x01\x00\xe14\xcd\x8c\xda4\x9a\x19'
    b'\xd44g\xa6\xcd443\xc74\xcd\xac\xc04\x9a9\xba4g\xc6\xb344S\xad44\xf3'
    b'\xa64g\xa6\xa04\xcdl\x9a443\x944\x00 \x8e4\x00 \x88443\x824\x9a\xd9x4'
    b'\x9a\x99m4\x01\x80b44\xb3W443M4\x01\x00C4\x00@94g\xa6/4\x00\x80&4\xcd'
    b'\xcc\x1d4gf\x154\xcdL\r4g\xa6\x054g\xe6\xfc3\x9a\x19\xef343\xe2343\xd63'
    b'\x9a\x19\xcb3\x9a\x99\xc03\x00\x00\xb73\xcdL\xae3\x00\x80\xa63\x9a\x99'
    b'\x9f3\xcdL\x993\x9a\x99\x933\x9a\x19\x8f343\x8b3g\xe6\x873\x00\x80\x853'
    b'\x00\x00\x843\xcd\xcc\x823\xcd\xcc\x823\xcd\xcc\x8234\xb3\x83343\x853'
    b'\xcdL\x8734\xb3\x8934\xb3\x8c3\x00\x00\x903g\xe6\x933\x9a\x19\x983g\xe6'
    b'\x9c34\xb3\xa13\xcd\xcc\xa6343\xac3g\xe6\xb13\x9a\x99\xb73\x9a\x99\xbd3'
    b'\x9a\x99\xc33\x9a\x99\xc93\xcdL\xcf3\xcdL\xd53\xcdL\xdb3\x01\x00\xe134'
    b'\xb3\xe63\x9a\x19\xec343\xf13\x01\x00\xf63\xcd\xcc\xfa3\x01\x00\xff3\x9a'
    b'\x99\x014gf\x034g\xe6\x044\x00@\x0644s\x074\x9aY\x0844\xf3'
    b'\x084gf\t4gf\t4\x00@\t4\xcd\xcc\x084g\xe6\x074\x9a\xd9\x064\x9aY\x054'
    b'\xcd\x8c\x0344s\x014\xcd\xcc\xfd3gf\xf83\x9a\x19\xf23g\xe6\xea3\x9a\x19'
    b'\xe334\xb3\xda3gf\xd13\x01\x80\xc73\x00\x00\xbd3\x9a\x99\xb13\x9a\x99'
    b'\xa534\xb3\x98343\x8b343z3\xcd\xcc\\3\x9a\x99=343\x1d343\xf72\x9a\x99'
    b'\xb12\xcd\xccP2\x01\x00p143\xbb\xb1\x01\x00|\xb2\x9a\x99\xcf\xb2\xcd'
    b'\xcc\x11\xb3gf<\xb343h\xb3\x9a\x99\x8a\xb3\x9a\x19\xa1\xb343\xb8\xb3g'
    b'\xe6\xcf\xb3\x9a\x99\xe7\xb3\x9a\x99\xff\xb34\xf3\x0b\xb4\x00@\x18\xb4'
    b'\xcd\x8c$\xb4\x00\x001\xb4\x9a\x99=\xb4\xcd\x0cJ\xb4g\xa6V\xb4\x9a\x19c'
    b'\xb44\xb3o\xb4g&|\xb4\x9a9\x84\xb4\x00`\x8a\xb44s\x90\xb4g\x86\x96\xb44s'
    b'\x9c\xb4\xcdL\xa2\xb4\x00\x00\xa8\xb4\x00\xa0\xad\xb4\x9a\x19\xb3\xb4\x00'
    b'\x80\xb8\xb4\xcd\xac\xbd\xb4g\xc6\xc2\xb44\x93\xc7\xb4\xcdL\xcc\xb4\xcd'
    b'\xcc\xd0\xb44\x13\xd5\xb443\xd9\xb4g\x06\xdd\xb4\x01\xa0\xe0\xb4\x01\x00'
    b'\xe4\xb4g&\xe7\xb4\x01\x00\xea\xb4\xcd\x8c\xec\xb4\x01\xe0\xee\xb44\xd3'
    b'\xf0\xb4\xcd\x8c\xf2\xb4\x9a\xf9\xf3\xb4\x9a\x19\xf5\xb4\xcd\xec\xf5\xb4'
    b'\x01`\xf6\xb4g\x86\xf6\xb4\x01`\xf6\xb4\xcd\xec\xf5\xb4\x9a\x19\xf5\xb4'
    b'\x9a\xf9\xf3\xb4\xcd\x8c\xf2\xb4\x01\xc0\xf0\xb4g\xa6\xee\xb4\x01@\xec'
    b'\xb4\x9ay\xe9\xb4gf\xe6\xb4g\x06\xe3\xb4gF\xdf\xb4\xcdL\xdb\xb4g\x06\xd7'
[... 32 lines cut ...]
# length is equal to the block length value that has been extracted from the
# block; hence the while loop will work just fine because we have already 
# read all the data from the instrument. If _read_termination was accounted 
# for, then we would get the timeout exception because there is simply no 
[... cut ...]
# inside from_binary_block, I get the following values of variables:
element_length = 4
array_length = 512

# And here is what is returned by the `struct.unpack_from` call:
# result = container(struct.unpack_from('<512f', block, offset)
# result = <class 'list'>: <Too big to print. Len: 512>
# 000 = {float} -1.3411045607369942e-08
# 001 = {float} -2.2770837659891185e-08
# 002 = {float} -3.1711536507827986e-08
# ...
```

In conclusion, I'm more and more convinced that adding the length of the `_read_termination` to the `expected_length` is not necessary at all.

MatthieuDartiailh, 2018-08-06:
Sorry again, after my holidays I moved from Europe to the USA. But I managed to get some feedback from my colleagues. And they did confirm that in the case of a Keysight PNA on ethernet the termination character is indeed added at the end of the message and is not counted in the declared length in the header.
So I will review your patch this week, and do my best to release a bug-fix release. But since I am still convinced we need to account for this when pertinent, I will change the default in 1.10 and add documentation about reading binaries data and how to debug whether or not you should set `expect_termination` to `True` or `False`.

astafan8, 2018-08-06:
Just for my understanding, do your colleagues confirm that for that particular instrument the `block` (!) that is read by `_read_raw` method inside `read_binary_data` method DOES contain the termination? Also, when is this termination at the very end of the block get removed? (I couldn't find any relevant code in `from_**_block` functions in `pyvisa.util`) (and [rounding the length on this line in `from_binary_block`](https://github.com/pyvisa/pyvisa/blob/master/pyvisa/util.py#L457) does not seem very convincing because it is not explicit).

About the default, let's make it clear. Which of the two cases is the most frequent for `read_binary_values`: `expect_termination=False` or `expect_termination=True`? (I also do not prefer to change the default between pyvisa `1.9.1` and `1.10`, let's choose a good one now, and forget about it)

What I observe is the following. `pyvisa 1.8` did not account for the termination in `read_binary_values`, everything was fine. `pyvisa 1.9` started accounting for the termination in `read_binary_values` due to the case with Keysight PNA instuments. This in turn broke the functionality for (in my case) Standford Research SR860 lock-in amplifier. So, the fix in `1.9` for Keysight PNA users changes the behavior, and there are no means of getting it back (other than hacking). From my understanding of the visa infrastructure, in case the binary data is being read, there is no need to account for the termination because (1) the block header contains the length of the data, (2) terminations seem to be handled on the lower level (e.g. `_read_raw` or `.dll`). Your argument about [rounding the length on this line in `from_binary_block`](https://github.com/pyvisa/pyvisa/blob/master/pyvisa/util.py#L457) does not convince me because it is not explicit, which may lead to confusion and bugs. But then I see that after consultation with your colleagues, you still insist on sticking to `1.9`'s behavior as the default, right? I do not want this discussion to go into the direction of defending needs of particular groups of users using particular instruments. Instead, I would like to find out what is a meaningful default for *all* `pyvisa` users. 

So far, to my understanding, sticking to the `1.8`'s behavior (`expect_termination=False`) in `1.9.1` and further on makes more sense because:
- lines of code that account for read termination in `read_binary_values` don't seem to be justified, apart from the Keysight PNA case
- it is what pyvisa `1.8` users expect since 2015. Although, this argument is vague because I do not know the adoption rate of new pyvisa versions. In QCoDeS, we were strict in using `1.8` because VISA-related stuff is an area of high risk for us.

MatthieuDartiailh, 2018-08-09:
Another of my colleague confirmed he sees the same behavior as the Keysight PNA on a R&S VNA (over ethernet once again).

astafan8, 2018-08-13:
Dear @MatthieuDartiailh, thank you for your thorough response.

Let me mention that I was able to contact one of our QCoDeS users who work with Keysight PNA (Keysight Technologies, N9913A, MY56072788, A.09.60) (and also with Agilent Technologies, N5230C, MY48100481, A.09.42.22), and they confirmed that their code (that is based on QCoDeS drivers) that uses `query_binary_data` works well with *both* `pyvisa` `1.8` and `1.9`. Read at the end of the message for details (code+plots).

So, with this information from the user (he's also a colleague of mine), could it be that in your colleagues' setting the communication with the device was not correctly set up, hence the accounting for the read termination was needed? (please, do not get me wrong, I'm not blaming, but just checking)

Finally, you generally convinced me about the default (btw, please, have a look at the updated PR #349). Still, if you can check one last time with your colleagues about the Keysight PNA case (perhaps, get the exact model/firmware), it would be great 😺 

### More details on Keysight PNA test by QCoDeS user

`query_binary_values` method was tested for the two Keysight PNA devices below, and for two `pivsa` versions, `1.8` and `1.9`.

#### Keysight Technologies, N9913A, MY56072788, A.09.60

```python
import pyvisa
import matplotlib.pyplot as plt
import numpy as np

rm = pyvisa.ResourceManager()
ff = rm.open_resource('TCPIP0::10.66.42.172::inst0::INSTR')
print(ff.query('*IDN?'))

ff.write('FORM:DATA REAL,32')
ff.write(f'CALC:FORM MLOG')
data = np.array(ff.query_binary_values('CALC:DATA:SDATA?', datatype='f', is_big_endian=False))
I = data[::2]
Q = data[1::2]
RAW = I +1j*Q

LMAG = 20*np.log10(np.abs(RAW))
start = float(ff.query('SENS:FREQ:STAR?'))
stop = float(ff.query('SENS:FREQ:STOP?'))
freq = np.linspace(start, stop, len(LMAG))

plt.figure()
plt.legend(['S21'])
plt.xlabel('Frequency [MHz]')
plt.ylabel('Magnitude [dB]')
plt.plot(freq * 1e-6, LMAG)
plt.show()
```

N9913A with pyvisa 1.8 (left) and 1.9.0 (right):
[image]

#### Keysight Technologies, N5230C, MY48100481, A.09.42.22

```python
# -*- coding: utf-8 -*-
import pyvisa
import matplotlib.pyplot as plt
import numpy as np

rm = pyvisa.ResourceManager()
pna = rm.open_resource('TCPIP0::10.66.42.15::inst0::INSTR')
print(pna.query('*IDN?'))

pna.write('FORM:DATA REAL,32')
pna.write('FORM:BORD NORM')

pna.write(f'CALC:FORM MLOG')
data = np.array(pna.query_binary_values('CALC:DATA? SDATA', datatype='f', is_big_endian=True))
I = data[::2]
Q = data[1::2]
RAW = I +1j*Q

LMAG = 20*np.log10(np.abs(RAW))
start = float(pna.query('SENS:FREQ:STAR?'))
stop = float(pna.query('SENS:FREQ:STOP?'))
freq = np.linspace(start, stop, len(LMAG))

plt.figure()
plt.legend(['S21'])
plt.xlabel('Frequency [MHz]')
plt.ylabel('Magnitude [dB]')
plt.plot(freq * 1e-6, LMAG)
plt.show()
```

N5230C with pyvisa 1.8:
[image]

N5230C with pyvisa 1.9.0:
[image]

astafan8, 2018-08-13:
@MatthieuDartiailh yeah.. I was just wondering :)

Thanks for the tip!

I'm looking forward to your comments on the PR.

MatthieuDartiailh, 2018-08-14:
Closed by #349

closed 2018-08-14 by MatthieuDartiailh