pyvisa/pyvisa #351: Pyvisa USB Instrument wait_on_event() argument error

daydin, 2018-08-03:
Hello,

I am trying to follow [this example](https://pyvisa.readthedocs.io/en/1.4/pyvisa.html) of reading 10 voltage measurements and calculating their average. My code is below:

`import visa
import matplotlib.pyplot as plt
import numpy

class MDO3014:
	def __init__(self,address):
		self.address = address
		self.rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
		self.inst = self.rm.open_resource(address)
		print(self.inst.query("*IDN?"))

	def get_trace(self):
		# values = list(range(100))
		# self.inst.write_binary_values('WLISt:WAVeform:DATA square,', values, datatype='d', is_big_endian=True)
		self.conf = self.inst.write("*rst; status:preset; *cls") 
		interval_in_ms = 500
		number_of_readings = 1
		self.inst.write("status:measurement:enable 512; *sre 1")
		self.inst.write("sample:count %d" % number_of_readings)
		self.inst.write("trigger:source bus")
		self.inst.write("trigger:delay %f" % (interval_in_ms / 1000.0))

		self.inst.write("trace:points %d" % number_of_readings)
		self.inst.write("trace:feed sense1; feed:control next")
		self.inst.write("initiate")
		self.inst.assert_trigger()
		self.inst.wait_on_event("srq",10000)
		#self.inst.wait_for_srq()
		voltages = self.inst.ask_for_values("trace:data?")
		print ("Average voltage: ", sum(voltages) / len(voltages))

osc = MDO3014('USB0::0x0699::0x0408::C049183::INSTR')
osc.get_trace()`

I get the following ArgumentError: line 1848, in wait_on_event
    byref(out_event_type), byref(out_context))
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type.

I also tried to input two integers for the call to wait_on_event():

`		self.inst.wait_on_event(0,10000)`

but this throws another error: pyvisa.errors.VisaIOError: VI_ERROR_INV_EVENT (-1073807322): Specified event type is not supported by the resource.

Not sure what is the correct input format there, any help is appreciated!

MatthieuDartiailh, 2018-08-03:
I have little experience with events but I think you need to use the `EventType` enumeration found in `pyvisa.constants` which has a `service_request` member.

MatthieuDartiailh, 2018-08-03:
Feel free to submit a documentation update if you have a little time for it.

daydin, 2018-08-13:
Hello, sorry I've been on holidays. In the meanwhile, someone else resolved this using a different code segment, but thank you for your help.

MatthieuDartiailh, 2020-09-06:
Closing since the user found a workaround. As a side note the support for event based programming has been improved with #511.

closed 2020-09-06 by MatthieuDartiailh