pyvisa/pyvisa #354: read_bytes() fails with 20020 bytes; RG1054Z oscilloscope

jp3141, 2018-09-01:
import visa
import numpy as np
import sys

print(sys.version)
print("visa", visa.__version__)
#visa.log_to_screen()

DS1054Z = visa.ResourceManager().open_resource('USB0::0x1AB1::0x04CE::DS1ZA201003553::INSTR') 
print("DS1054Z:", DS1054Z.query("*IDN?"))

DS1054Z.write(":WAV:FORMAT BYTE;:WAV:MODE RAW;:SYSTEM:BEEPER OFF")
DS1054Z.write(":ACQUIRE:MDEPTH 30000")
DS1054Z.write(":STOP")

for NPOINTS in range(20010,20100):# will fail at 20020
  print(NPOINTS, end='')  
  DS1054Z.write(":WAV:START 1;:WAV:STOP %i;:WAV:DATA?" % NPOINTS)
  print(">",DS1054Z.read_bytes(11),"<", end='') # Contains '#90000ddddd'
  CURVE1 = np.frombuffer(DS1054Z.read_bytes(NPOINTS), dtype='b')
  print(len(CURVE1), "chars")
  DS1054Z.read_bytes(1) # trailing '\n'

My Rigol RG1054Z & pyvisa fail on certain (random ?) lengths of read_bytes() commands. It also fails on query_binary_values(). This example code shows 1 sample of the problem -- there are other magic values besides 20020 that fail (greater than 20020). It appears to be a memory issue because sometimes depending on the codes used before this one, the failing magic number is different.

3.6.5 |Anaconda, Inc.| (default, Mar 29 2018, 13:32:41) [MSC v.1900 64 bit (AMD64)]
visa 1.9.1
DS1054Z: RIGOL TECHNOLOGIES,DS1104Z,DS1ZA201003553,00.04.04.03.05

20010> b'#9000020010' <20010 chars
20011> b'#9000020011' <20011 chars
20012> b'#9000020012' <20012 chars
20013> b'#9000020013' <20013 chars
20014> b'#9000020014' <20014 chars
20015> b'#9000020015' <20015 chars
20016> b'#9000020016' <20016 chars
20017> b'#9000020017' <20017 chars
20018> b'#9000020018' <20018 chars
20019> b'#9000020019' <20019 chars
20020> b'#9000020020' <Traceback (most recent call last):
  File "bug.py", line 20, in <module>
    CURVE1 = np.frombuffer(DS1054Z.read_bytes(NPOINTS), dtype='b')
  File "C:\Projects\Miniconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 342, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Projects\Miniconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Projects\Miniconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2018-09-01:
Honestly I do not know (but I suspect the instrument). I suggest you try to do the same using NI MAX (or something equivalent) that should allow us to see if the issue is on the instrument side or on the PyVISA side.

jp3141, 2018-09-01:
I looped for all lengths -- it starts failing at length 8244, and therafter at increments of 64; however after every 7 of these, it has a fail at an increment of 128:

Here's a segment of the log:
```
9074> b'#9000009074' <9074 chars
9075> b'#9000009075' <9075 chars
9076> b'#9000009076' <
Error at 9076  after  64
9075 chars
9077> b'#9000009077' <9077 chars
9078> b'#9000009078' <9078 chars
9079> b'#9000009079' <9079 chars
```

and filtered for 'Error':
```
Error at 8244  after  8243
Error at 8308  after  64
Error at 8372  after  64
Error at 8436  after  64
Error at 8500  after  64
Error at 8564  after  64
Error at 8628  after  64
Error at 8756  after  128
Error at 8820  after  64
Error at 8884  after  64
Error at 8948  after  64
Error at 9012  after  64
Error at 9076  after  64
Error at 9140  after  64
Error at 9268  after  128
Error at 9332  after  64
Error at 9396  after  64
```

Code segment:
```
for NPOINTS in range(2,30100):
  print(NPOINTS, end='')
  err+=1
  DS1054Z.write(":WAV:START 1;:WAV:STOP %i;:WAV:DATA?" % NPOINTS)
  print(">",DS1054Z.read_bytes(11),"<", end='') # Contains '#90000ddddd'
  try:
    CURVE1 = np.frombuffer(DS1054Z.read_bytes(NPOINTS), dtype='b')
  except:
    print("\nError at", NPOINTS, " after ", err)
    err=0
  print(len(CURVE1), "chars")
  DS1054Z.read_bytes(1) # trailing '\n'

```

for now, I'll just read in 8000 byte segments and join the arrays. I'll also search in Rigol forums for someone with the same setup who could replicate.

jp3141, 2018-09-04:
Actually since I have at most a 60,000-byte read, it's faster to read in the whole 60k bytes and truncate the array. 60,000 doesn't appear to have a timeout problem.

closed 2020-09-06 by MatthieuDartiailh