pyvisa/pyvisa #356: TypeError: super(type, obj): obj must be an instance or subtype of type

maquiero, 2018-09-04:
This is the output of python -m visa info:

Machine Details:
   Platform ID:    Linux-4.15.0-33-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.15rc1
   Compiler:       GCC 7.3.0
   Bits:           64bit
   Build:          Apr 15 2018 21:51:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      Binary library: Not found

MatthieuDartiailh, 2018-09-04:
Could you post the full traceback please ?

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel fee to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh