pyvisa/pyvisa #358: Trouble connecting to virtual COM port device

benaxelrod, 2018-09-18:
I have a BK Precision 2831E Digital Multimeter that I am trying to use with pyvisa.  Unfortunately, I am unable to get pyvisa to connect to it.  I have been using pyvisa for a while with another instrument (a spectrum analyzer over LAN) so I know my install is working.

I installed the virtual COM port drivers from BK Precision.  When I plug in the DMM, it shows up as:
Silicon Labs CP210x USB to UART Bridge (COM59)

Here is the error i get:

``` 
Python 2.7.10 (default, May 23 2015, 09:40:32) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.__version__
'1.9.1'
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
()
>>> rm.open_resource('COM59')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1703, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1663, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1383, in parse_resource_extended
    alias_if_exists)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.
```

If I try with a different resource name, I get this:
```
>>> rm.open_resource('USB::10c4::2831E::123B18126::RAW')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1703, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1663, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1383, in parse_resource_extended
    alias_if_exists)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INV_RSRC_NAME (-1073807342): Invalid resource reference specified. Parsing error.
```
I am not exactly sure how to get the USB board number, manufacturer ID, or USB interface number.  

If I use PuTTY to open COM59 I can control it simply over serial.  Any ideas?

benaxelrod, 2018-09-19:
The ID of the COM port or the manufacturer?

MatthieuDartiailh, 2018-09-19:
ID of the com port. The fact that you see it as a COM port makes me believe it is an emulated serial resource. So it should show up in NI-MAX as a serial resource (Starting with ASRL).

benaxelrod, 2018-09-25:
I installed Keysight Connection Expert and got this screen:
[image]

And if I click the "Add Instrument" button it lets me test the VISA address:
[image]

But the "ASRL59::INSTR" string doesn't work.  
[image]

MatthieuDartiailh, 2018-09-26:
This is weird. Could you post the output of the following command `pyvisa-info` ? By curiosity could you try to install pyvisa-py and run the same script but with `rm = visa.ResourceManager('@py')`.

benaxelrod, 2018-09-26:
First, the info:
```
C:\error_model>pyvisa-info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\python27\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          May 23 2015 09:40:32 (#default)
   Unicode:        UCS2

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: 5244164
         Spec. Version: 5243904
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP INSTR: Available
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available
```
I did this to install pyvisa-py:
pip install pyvisa-py
and got:
Successfully installed pyvisa-py-0.3.1

Then when i tried with "@py", i got a different error:
```
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> instrument = rm.open_resource('ASRL59::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mod
e, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa-py\sessions.py", line 213, in __init__
    self.after_parsing()
  File "C:\Python27\lib\site-packages\pyvisa-py\serial.py", line 65, in after_parsing
    self.interface = cls(port=self.parsed.board, timeout=self.timeout, write_timeout=self.timeout)
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 38, in __init__
    SerialBase.__init__(self, *args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'write_timeout'
```

MatthieuDartiailh, 2018-09-26:
I will look into the pyvisa-py error, sorry for the mess. I did not know that R&S had their own VISA implementation. Could you try uninstalling it and running from the Keysight one that (or just specify you want the Keysight one when opening the resource manager)  ?

MatthieuDartiailh, 2018-09-26:
For pyvisa-py to work you need to update pyserial (requires >= 3.0)

benaxelrod, 2018-09-26:
OK, updated pyserial.  Different error now:
```
C:\error_model>python
Python 2.7.10 (default, May 23 2015, 09:40:32) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial
>>> import visa
>>> visa.__version__
'1.9.1'
>>> serial.__version__
'3.4'
>>> rm = visa.ResourceManager('@py')
>>> instrument = rm.open_resource('ASRL59::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mod
e, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa-py\sessions.py", line 213, in __init__
    self.after_parsing()
  File "C:\Python27\lib\site-packages\pyvisa-py\serial.py", line 65, in after_parsing
    self.interface = cls(port=self.parsed.board, timeout=self.timeout, write_timeout=self.timeout)
  File "build\bdist.win32\egg\serial\serialwin32.py", line 31, in __init__
  File "build\bdist.win32\egg\serial\serialutil.py", line 240, in __init__
  File "build\bdist.win32\egg\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port u'59': WindowsError(2, 'The system cannot fin
d the file specified.')
```
This is strange because I can open up PuTTY on COM59 and control the device.

MatthieuDartiailh, 2018-09-26:
Are you sure that no other application holds an open connection to the device ? (putty is closed, etc)

benaxelrod, 2018-09-26:
Very sure.  I just did a reboot to make sure.

I wasn't sure how to specify which visa library at first.  It took some time for me to figure out I need to specify the path to the DLL and it must use forward slashes, not backslashes (even though I am on Windows).  

Is there a way to list available visa libraries?  I searched my machine and found lots of visa looking DLLs.  But I think these are the main ones:

- C:/Program Files (x86)/IVI Foundation/VISA/WinNT/ktvisa/ktbin/visa32.dll
- C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll
- C:/Program Files (x86)/IVI Foundation/VISA/WinNT/RsVisa/bin/visa32.dll

If I use either the Tektronix or Keysight library, it works!

```
>>> import visa
>>> rm = visa.ResourceManager(visa_library='C:/Program Files (x86)/IVI Foundation/VISA/WinNT/ktvisa/
ktbin/visa32.dll')
>>> instrument = rm.open_resource('ASRL59::INSTR')
>>> instrument.query('*IDN?')
u'*IDN?\r2831E  Multimeter,Ver1.4.14.06.18,123B18126\n'
>>> rm.close()
```

MatthieuDartiailh, 2018-09-27:
I do not know of any way to list all visa implementations no. Some can cooperate (NI and Keysight) for the others I do not know. Usually it is safer to stick to NI or Keysight implementation, I had issue in the past with Tektronix and it seems R&S is not perfect either. If you can uninstall the one you do not need/use.
Anyway I am happy we figured this out. I am going to rename the issue and close it. Feel free to re-open if needs be.

closed 2018-09-27 by MatthieuDartiailh

bilderbuchi, 2020-05-10:
So, this might be useless by now, but using pyvisa-py I just had a similar error like the OP had here:
```
>>> instrument = rm.open_resource('ASRL59::INSTR')
serial.serialutil.SerialException: could not open port u'59': WindowsError(2, 'The system cannot fin
d the file specified.')
```
and found that if I specify the name as `'ASRLCOM59::INSTR'`, the connection started to work. Is that intended or a bug I should report against pyvisa-py?

MatthieuDartiailh, 2020-05-10:
Since you are on windows what behavior do you see using NI or Keysight or another vendor VISA implementation ? If `ASRL59::INSTR` works with any of those, please report this to pyvisa-py with as many details as possible.

bilderbuchi, 2020-05-10:
sure!

bilderbuchi, 2020-05-11:
Done (https://github.com/pyvisa/pyvisa-py/issues/230), I hope this is enough details to troubleshoot this.