pyvisa/pyvisa #359: .stb does not throw a time-out exception when timed out

Dennis-van-Gils, 2018-09-19:
Is this by design? I would welcome being able to catch a time-out, instead of .stb returning silently after the time-out period expired.

MatthieuDartiailh, 2018-09-19:
This indeed does not seem right. Could you post the output of python -m visa info ? I would like to know with which backend you are seeing this behavior.

Dennis-van-Gils, 2018-09-19:
Thanks for looking into it. FYI: both the Keysight and NI VISA libraries are installed on that computer. Here is the info you requested:

Machine Details:
   Platform ID:    Windows-10-10.0.15063-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.dev0

Backends:
   ni:
      Version: 1.9.dev0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672

Dennis-van-Gils, 2018-09-19:
On further inspection. I might have installed the PyVisa library incorrectly. I had it installed via 'pip install pyvisa', but conda actually recommends 'conda install -c conda-forge pyvisa'. Running new tests.

'python -m visa info' now reports:
Machine Details:
   Platform ID:    Windows-10-10.0.15063-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672

MatthieuDartiailh, 2018-09-19:
Could you explain to me how you reach the conclusion that the operation time-out ? Going through the code I do not see how this would fail to raise.

Dennis-van-Gils, 2018-09-19:
No improvement unfortunately. The problem of .stb timing out silently persists.

Dennis-van-Gils, 2018-09-19:
```python
if (self.device.stb & 0b100) == 0b100:
    # There are unread errors in the queue available. Retrieve all.
    while True:
        if self.query_error():   # Will perform query("err?") and stores reply in 'state.error'. Returns True when query operation ended successful.
            if self.state.error is None:
                break
            else:
                self.state.all_errors.append(self.state.error)
        else:
            break
```

Mmmm. I think I see my problem now. The `while True` mechanism is causing me to believe that `.stb` is timing out, whereas it might be my `.query_error()` method that is causing the time-out. I thought my `.query_error()` method would catch a time-out as well, but I will have to scrutinize my code once more.

Dennis-van-Gils, 2018-09-19:
No, I think I am still right in my assumption that `.stb` times out silently. I wrote a small program to isolate the problem:
```python
def stb_test(self):
    return ((self.device.stb & 0b100) == 0b100)
```

`stb_test` is being called from another thread (PyQt5 QThread running on a QTimer) every second, like this:
```python
mutex.lock()
tick = time()
my_device.stb_test()
print("done in %.3f" % (time() - tick))
mutex.unlock()
```

Output on the terminal would read:
```
done in 0.001
done in 0.001
done in 0.001
done in 4.001   <----- occasional time out, very intermittent 
done in 0.001

MatthieuDartiailh, 2018-09-19:
Since I am likely to have difficulties reproducing this one could try the following. In your pyvisa installation (under site-packages) could try adding `print(retcode)` below this line https://github.com/pyvisa/pyvisa/blob/master/pyvisa/resources/messagebased.py#L701.

I would like to make sure that we do see a timeout return code.

Dennis-van-Gils, 2018-09-19:
Okay done. And I made a minimal test:
```python
import visa
import time

rm = visa.ResourceManager()
psu = rm.open_resource("USB0::0x0957::0x8707::US15M3727P::INSTR", timeout=4000)

while True:
    psu.query("err?")   # Some stuff to do
    psu.query("*idn?")  # Some stuff to do
    tick = time.time()
    print ((psu.stb & 0b100) == 0b100)
    print("Done in %.3f" % (time.time() - tick))
```

Terminal output
```
False
Done in 0.001
StatusCode.success
False
Done in 0.001
StatusCode.success
False
Done in 0.000
StatusCode.success
False
Done in 4.001                  <----- time out
StatusCode.success
False
Done in 0.001
StatusCode.success
```

MatthieuDartiailh, 2018-09-19:
I am very perplexed. Could you enable logging (`visa.log_to_screen()`) and run this again ?

Dennis-van-Gils, 2018-09-19:
New minimal test:
```python
import sys
import visa
import time

visa.log_to_screen()
rm = visa.ResourceManager()
psu = rm.open_resource("USB0::0x0957::0x8707::US15M3727P::INSTR", timeout=4000)

sys.exit()

while True:
    psu.query("err?")   # Some stuff to do
    psu.query("*idn?")  # Some stuff to do
    tick = time.time()
    print ((psu.stb & 0b100) == 0b100)
    done_in = time.time() - tick
    print("Done in %.3f" % done_in)
    if (done_in) > 1: sys.exit(0)
```

Start terminal output:
```
2018-09-19 19:38:25,978 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2018-09-19 19:38:25,979 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2018-09-19 19:38:25,981 - pyvisa - DEBUG - No user defined library files
2018-09-19 19:38:25,985 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2018-09-19 19:38:25,987 - pyvisa - DEBUG - The NI implementation available
2018-09-19 19:38:25,990 - pyvisa - DEBUG - No user defined library files
2018-09-19 19:38:25,997 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa64.dll']
2018-09-19 19:38:26,150 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2018-09-19 19:38:26,153 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
2018-09-19 19:38:26,219 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x000002180A6686C8>',) -> 0
2018-09-19 19:38:26,220 - pyvisa - DEBUG - Created ResourceManager with session 1
2018-09-19 19:38:26,223 - pyvisa - DEBUG - viParseRsrcEx(1, 'USB0::0x0957::0x8707::US15M3727P::INSTR', 'c_ushort(7)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x000002180A668648>, <ctypes.c_char_Array_256 object at 0x000002180A6687C8>, <ctypes.c_char_Array_256 object at 0x000002180A668748>) -> 0
2018-09-19 19:38:26,224 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - opening ...
2018-09-19 19:38:26,237 - pyvisa - DEBUG - viOpen(1, 'USB0::0x0957::0x8707::US15M3727P::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x000002180A54F4C8>') -> 0
2018-09-19 19:38:26,238 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - is open with session 2
2018-09-19 19:38:26,238 - pyvisa - DEBUG - viSetAttribute(2, 1073676314, 4000) -> 0
```

Last terminal output:
```
False
Done in 0.003
False
Done in 0.003
False
Done in 0.004
2018-09-19 19:35:39,988 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1CC8>, 20480, 'c_ulong(57)') -> 0
2018-09-19 19:35:39,988 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1CC8>, 20480, 'c_ulong(57)') -> 0
2018-09-19 19:35:39,988 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1CC8>, 20480, 'c_ulong(57)') -> 0
2018-09-19 19:35:39,993 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
2018-09-19 19:35:39,993 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
2018-09-19 19:35:39,993 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
2018-09-19 19:35:39,996 - pyvisa - DEBUG - viWrite(4, b'err?\r\n', 6, 'c_ulong(6)') -> 0
2018-09-19 19:35:39,996 - pyvisa - DEBUG - viWrite(4, b'err?\r\n', 6, 'c_ulong(6)') -> 0
2018-09-19 19:35:39,996 - pyvisa - DEBUG - viWrite(4, b'err?\r\n', 6, 'c_ulong(6)') -> 0
2018-09-19 19:35:39,999 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-09-19 19:35:39,999 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-09-19 19:35:39,999 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-09-19 19:35:40,002 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1E48>, 20480, 'c_ulong(6)') -> 0
2018-09-19 19:35:40,002 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1E48>, 20480, 'c_ulong(6)') -> 0
2018-09-19 19:35:40,002 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1E48>, 20480, 'c_ulong(6)') -> 0
2018-09-19 19:35:40,004 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,004 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,004 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,006 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-09-19 19:35:40,006 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
[... 36 lines cut ...]
2018-09-19 19:35:40,167 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B1E48>, 20480, 'c_ulong(6)') -> 0
2018-09-19 19:35:40,172 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,172 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,172 - pyvisa - DEBUG - viWrite(4, b'*idn?\r\n', 7, 'c_ulong(7)') -> 0
2018-09-19 19:35:40,177 - pyvisa - DEBUG - USB0::0x0957::0x8707::US15M3727P::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
[... cut ...]
2018-09-19 19:35:40,230 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B19C8>, 20480, 'c_ulong(57)') -> 0
2018-09-19 19:35:40,230 - pyvisa - DEBUG - viRead(4, <ctypes.c_char_Array_20480 object at 0x000001E9643B19C8>, 20480, 'c_ulong(57)') -> 0
2018-09-19 19:35:44,239 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
2018-09-19 19:35:44,239 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
2018-09-19 19:35:44,239 - pyvisa - DEBUG - viReadSTB(4, 'c_ushort(32)') -> 0
False
Done in 4.013
An exception has occurred, use %tb to see the full traceback.

SystemExit: 0
```

MatthieuDartiailh, 2018-09-19:
I do not know what is going with your instrument but as you can see the return code of viReadSTB does not indicate an error (error codes are negative). At this point if NI-VISA does not report the error I am not sure what pyvisa can do.

Dennis-van-Gils, 2018-09-19:
I understand. That sets me thinking that the problem might lie in the USB isolator that is in between the computer and the VISA device. I can't take it out easily though for further testing as the rest of the facility needs to stay on a floating ground. Have you heard about USB isolators causing problems with VISA before perhaps? 

The problem is hardware related, it seems. Many thanks for your effort and time!

MatthieuDartiailh, 2018-09-19:
I have no experience with usb isolators but it may cause some issues. Could you not bypass it just for this device ?

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh