pyvisa/pyvisa #360: GPIB bad settings

hedva92, 2018-09-21:
When I call X device in the labview, X will respond immediately. When I call X device in the Python, 
X device responds to the query_delay argument at the set time. If the query delay turns off, python reports  time expired. I think, that instrument under python doesn't able to read a termination from X device. Do not you know how it could be set in Python? Thanks.

```
import visa
rm = visa.ResourceManager()
x = rm.list_resources()
...
X = rm.open_resource(x[int(n)], query_delay=15 )
```

MatthieuDartiailh, 2018-09-21:
You can set the read and write termination character using the `read_termination` and `write_termination` attributes of the object returned by `rm.open_resources`.

hedva92, 2018-09-21:
python returns..
`UserWarning: read string doesn't end with termination characters return self.read()`

with read_termchar="\n" or "\r" or "\r\n" still waiting for full time of set query_delay

MatthieuDartiailh, 2018-09-23:
`query_delay` is the time the program wait between the write and the read operation when using the `query` method. I would suggest trying to experiment with different values of `query_delay` (shorter) and find what minimal value your instrument requires.

hedva92, 2018-09-23:
So what's different?

hedva92, 2018-09-23:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "xxx\v2.py", line 771, in vykon
    checkOK()
  File "xxx\v2.py", line 55, in checkOK
    if (X.query('SYST:ERR?') == '0,"No error"\n'):
  File "C:\Python\lib\site-packages\pyvisa\resources\messagebased.py", line 569, in query
    return self.read()
  File "C:\Python\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "C:\Python\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Python\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
with `set query_delay=0.1`

but when I call common command ddd.. `"*IDN?"` so device response immediately. The problem is that the calibrator needs more time than setting all the values over a simple `"*IDN?"`

MatthieuDartiailh, 2020-09-06:
Closing since the root problem of the issue what the finite time the instrument needed to handle some particular command.  Feel free to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh