pyvisa/pyvisa #361: VI_ERROR_TMO Problem with Read_Bytes() on DS1052

chrisjn52, 2018-09-21:
Hello,
I am using PyVISA to communicate with a Rigol DS1052E oscilloscope in Windows 10, Python Anaconda Python 3.6.5 and PyVisa 1.9.1 with the NI-VISA backend.

The issue is that a VI_ERROR_TMO error occurs when retrieving a large waveform file of size 1048576 bytes. I still get timeouts even when changing the chunk_size to 1048576. Smaller waveform files of 600 and 16384 bytes are transferred ok. Very occasionally the full 1048576 bytes are received.
I have attached two log files, the first shows a successful retrieval and the second, (executed directly after with no changes to the code), a failing scenario.

I am issuing the following command to read the waveform data after issuing ':WAV:DATA? CHAN1'

- return np.frombuffer(scope.read_bytes(1048576, break_on_termchar=False)[10:])

The command works flawlessly in NI-MAX using the identical scope setup:

[image]

In Wireshark NI-MAX retrieves four chunks:
[image]

Whilst pyVisa times out at various points, sometimes after receiving a single small chunk, sometimes after receiving several pairs of smaller chunks, with each pair constrained by the specified chunk size:
[image]

[success.log](https://github.com/pyvisa/pyvisa/files/2405248/success.log)
[failure.log](https://github.com/pyvisa/pyvisa/files/2405252/failure.log)

MatthieuDartiailh, 2018-09-23:
Happy you figured it out. I was quite perplexed by your issue.

chrisjn52, 2018-09-23:
Hi  Matthieu,
yes it was confusing to me also.

I had the timeout set according to the docs here.
<https://pyvisa.readthedocs.io/en/stable/resources.html?highlight=timeout> (I
used *scope.timeout = 60000*).

However, today the NI-MAX Visa Test Panel failed for the first time ever
and so I looked at the default in the configuration settings and it was set
to 2000mS.
Setting to 3000 did not help but 5000 worked ok. This was very odd since I
hadn't even looked for this timeout value in NI-MAX Visa Test Panel before
because it always worked flawlessly before today.

However, this new problem highlighted the fact that the timeout value must
have been on the borderline and so I looked at the timeout value settings
in my code again.
I had the following line in my code:

*scope =
visa.ResourceManager().open_resource('USB0::0x1AB1::0x0588::DS1ED133612273::INSTR',timeout=3000,
chunk_size=1048576)*

This line didn't originally contain any timeout value but I added it awhile
ago in an effort to fix the problems I was seeing and I believed that
the* scope.timeout
= 60000 *would have overridden the 3000 timeout value anyway?
As it turns out, changing this value to 5000 fixes the problem so all is
well.

Sarada67, 2024-06-01:
@chrisjn52 Hi I am facing problems to open the file to fix timeout problem . I am using Python and pyvisa to read measurement from Tektronix MSO58 Scope using SCPI . Can you please help me to fix this ? 
I am sufferering a lot.