pyvisa/pyvisa #365: Setting the write and read terminations in MessageBasedResources

marccheneau, 2018-10-02:
First a big thank for your work!

I'm currently writing a wrapper around the `MessageBasedResource` class for a research experiment and I miss the possibility to pass the write and read terminations to some of the `read*()` and `query*()` methods. For instance, I can pass the read termination to `read()` but not to `read_ascii_values()`. Also, the `query*()` methods lack both parameters in their list of argument.

Would you fix that or is it on purpose? For now I have to copy and patch some of your code in my wrapper, which isn't very productive.

MatthieuDartiailh, 2018-10-02:
Yes I would accept a PR making the arguments taken by those method more homogeneous. But I am surprised you need to actually change the termination character for just some operations after creating the resource. Usually instrument stick to a single termination character (which may be changed at runtime but this is rarely useful).

marccheneau, 2018-10-02:
I was trying to include all possible cases because I don't know what kind of hardware the end users will have. But leaving open the possibility to change such settings for specific I/O operations might be too much effort for too little gain (if any). In any case I didn't find how to set the read_termination of the `MessageBasedResource`. The `write_termination` property has a setter, but not the `read_termination` property.

marccheneau, 2018-10-02:
Oh, I'm confused. There is a setter for `read_termination`. I don't know how I missed it... Sorry for the waist of time. Still a more homogeneous handling of these variables would improve the readability of the code. Which way would you go? Allow per communication setting or only global setting?

marccheneau, 2018-10-02:
A related issue: if `ValuesFormat` will be deprecated in in version 1.10, what will be the proper way to set the related parameters at the global level?

MatthieuDartiailh, 2018-10-02:
We definitively want global settings for those. Having them on the basic methods (write, read) can be useful for debugging or for temporary settings (even if those should be rare/non-existent). I am fine with the status quo to avoid adding too many keywords arguments. Also note that the same argument could be made for encoding.

About your second question see : https://github.com/pyvisa/pyvisa/issues/327 I don't have time to look at this and probably won't have time for some months. But I would gladly review a PR in that direction.

marccheneau, 2018-10-02:
Regarding the first discussion, I will try to put up a PR with a homogenized handling of the termination settings in the next couple of weeks. If I understand you correctly, I should keep both the global setting for general use, and the local setting for debugging or temporary settings.

MatthieuDartiailh, 2018-10-03:
Sorry for the lack of clarity of my previous answer but thinking more about it I came to think that we are fine with just the global setting and the dynamic value only in the low level functions (read, read_raw, write, write_raw) since all the others (query, and more advanced read and write) are actually convenience function and do not require the kind of tinkering that dynamic setting allow. 
However if you believe this should be more clearly documented, feel free to propose an update to the docs.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh