pyvisa/pyvisa #367: ValueError after a few hours of communication with a device

druzinis, 2018-10-05:
I am using pyvisa and the @py backend to control a vector signal generator. Everything runs fine for a while but sometimes the following error gets raised. Any pointers would be highly appreciated. 

```
  File "visa_client.py", line 20, in command
    self.resource.write(command)
  File "...\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
  File "...\Python37-32\lib\site-packages\pyvisa\resources\messagebased.py", line 201, in write_raw
    return self.visalib.write(self.session, message)
  File "...\Python37-32\lib\site-packages\pyvisa-py\highlevel.py", line 368, in write
    ret = self.sessions[session].write(data)
  File "...\Python37-32\lib\site-packages\pyvisa-py\tcpip.py", line 182, in write
    flags, block)
ValueError: too many values to unpack (expected 2)
Error closing VISA link: wrong xid in reply 14206 instead of 14207
```

MatthieuDartiailh, 2018-10-05:
Could you post the output of `python -m visa info` (or equivalently `pyvisa-info`) so that I know what version you are running ?

MatthieuDartiailh, 2018-10-05:
My best guess is that a timeout occur inside the socket and you hit a bug vxi11.py
I pushed a fix on the branch tcpip_issue that you can test to confirm that. If you enable debug logging you should see a message and the error should now be a timeout error. I would suggest to try to close and reopen the connection to your instrument when it occurs.

druzinis, 2019-01-29:
Thank you for a prompt response. Unfortunately I was not able to test your suggested fix. Closing this issue for now; will reopen should this problem resurface again.

closed 2019-01-29 by druzinis

Lucifer00808, 2020-03-21:
Does anybody found the solution to the above problem? I am controlling an instrument via pyvisa and facing this value error randomly anytime. I tried to handle the exception but it still crashes my program.

MatthieuDartiailh, 2020-03-21:
Please try to install the master branch of pyvisa-py and see if the issue persist. Part of the TCPIP backend has been reworked and that issue may have been solved.

marcusvalentine, 2020-11-16:
This issue is still present in 1.11.4.dev1+g6881af7

It is easilu provoked by creating a bad query

`In [16]: h.query('SYST:ERR?')
---------------------------------------------------------------------------
RPCError                                  Traceback (most recent call last)
<ipython-input-16-de10d29fd81d> in <module>
----> 1 h.query('SYST:ERR?')

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in query(self, message, delay)
    636
    637         """
--> 638         self.write(message)
    639
    640         delay = self.query_delay if delay is None else delay

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in write(self, message, termination, encoding)
    195             message += term
    196
--> 197         count = self.write_raw(message.encode(enco))
    198
    199         return count

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in write_raw(self, message)
    155
    156         """
--> 157         return self.visalib.write(self.session, message)[0]
    158
    159     def write(

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/highlevel.py in write(self, session, data)
    541         # from the session handle, dispatch to the write method of the session object.
    542         try:
--> 543             written, status_code = self.sessions[session].write(data)
    544         except KeyError:
    545             return 0, self.handle_return_value(session, StatusCode.error_invalid_object)

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/tcpip.py in write(self, data)
    200
    201                 error, size = self.interface.device_write(
--> 202                     self.link, self._io_timeout, self.lock_timeout, flags, block
    203                 )
    204

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/protocols/vxi11.py in device_write(self, link, io_timeout, lock_timeout, flags, data)
    223                 params,
    224                 self.packer.pack_device_write_parms,
--> 225                 self.unpacker.unpack_device_write_resp,
    226             )
    227         except socket.timeout as e:

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/protocols/rpc.py in make_call(self, proc, args, pack_func, unpack_func)
    470         self.timeout += 1.0
    471
--> 472         return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
    473
    474     def connect(self, timeout=5.0):

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/protocols/rpc.py in make_call(self, proc, args, pack_func, unpack_func)
    236         if pack_func:
    237             pack_func(args)
--> 238         self.do_call()
    239         if unpack_func:
    240             result = unpack_func()

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/protocols/rpc.py in do_call(self)
    503             # Can't really happen since this is TCP...
    504             msg = "wrong xid in reply {0} instead of {1}"
--> 505             raise RPCError(msg.format(xid, self.lastxid))
    506
    507

RPCError: wrong xid in reply 15 instead of 17
`

MatthieuDartiailh, 2020-11-16:
If you have a self-contained reproduction example it would be great if you could post it in full. One previously identified cause for this issue was a bad recovery from a timeout, but that led to a difference of only 1 in the xid. I am curious to know how you achieved an offset of 2.

marcusvalentine, 2020-11-16:
Here's a ipython session:

```
~/tools$ ipython3
Python 3.6.9 (default, Jul 17 2020, 12:50:27)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pyvisa as visa   
   ...: rm = visa.ResourceManager()
   ...: s = 'a-n6705a-00727'
   ...: resource = f'TCPIP::{s}::INSTR'

In [2]: visa.__version__
Out[2]: '1.11.4.dev1+g6881af7'

In [3]: h = rm.open_resource(resource)

In [4]: h.query('*IDN?')
Out[4]: 'Agilent Technologies,N6705A,MY47000727,A.02.09\n'

In [5]: h.query('THIS WILL FAIL?')
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-5-cfe11922a6ed> in <module>
----> 1 h.query('THIS WILL FAIL?')

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in query(self, message, delay)
    642             time.sleep(delay)
    643
--> 644         return self.read()
    645
    646     def query_ascii_values(

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in read(self, termination, encoding)
    484         if termination is None:
    485             termination = self._read_termination
--> 486             message = self._read_raw().decode(enco)
    487         else:
    488             with self.read_termination_context(termination):

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in _read_raw(self, size)
    440                         status,
    441                     )
--> 442                     chunk, status = self.visalib.read(self.session, size)
    443                     ret.extend(chunk)
    444             except errors.VisaIOError as e:

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/highlevel.py in read(self, session, count)
    517             )
    518
--> 519         return data, self.handle_return_value(session, status_code)
    520
    521     def write(self, session: VISASession, data: bytes) -> Tuple[int, StatusCode]:

~/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/highlevel.py in handle_return_value(self, session, status_code)
    249
    250         if rv < 0:
--> 251             raise errors.VisaIOError(rv)
    252
    253         if rv in self.issue_warning_on:

VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

In [6]: h.query('*IDN?')
---------------------------------------------------------------------------
RPCError                                  Traceback (most recent call last)
<ipython-input-6-46ac0ec12ca9> in <module>
----> 1 h.query('*IDN?')

[... 81 lines cut ...]
```

results in

```
~/tools$ ./thisWillFail.py
Agilent Technologies,N6705A,MY47000727,A.02.09

Traceback (most recent call last):
  File "./learn/thisWillFail.py", line 12, in <module>
    print(h.query('THIS WILL FAIL?'))
  File "/home/valentinem/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/home/valentinem/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/valentinem/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/valentinem/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/valentinem/.local/share/virtualenvs/tools-BgNbJulN/lib/python3.6/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
Error closing VISA link: socket.timeout: The instrument seems to have stopped responding.

```

MatthieuDartiailh, 2020-11-16:
Thanks. I somehow believed I had fixed that before, but apparently not. And on top of that the error message is imprecise... I will try to dive in it sometimes this week or the next.

MatthieuDartiailh, 2020-11-17:
I tested on a Keithley 2450 and there everything works as expected. It appears your instrument does not faithfully implement VXI-11 since even is there is no message it should answer that it cannot answer within the imparted time, it does not which is why you get VI_ERROR_IO instead of VI_ERROR_TMO. 

This is tricky because not getting an answer could also mean something much more dramatic like a hardware failure (disconnected cable), so not so easy to work around. Could you try the same with an IVI compliant VISA implementation (Keysight provides an installer for Linux) ? I would like to know how the error is reported in that case.

marcusvalentine, 2020-11-17:
The Agilent DC power analyser I tried this on earlier is somewhat elderly. I just tried the same test as above on a Keysight 34461A DMM which is only 18 months old. The yielded the same results

```
isaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
...
RPCError: wrong xid in reply 9 instead of 10

```
Are you suggesting I should install the [Keysight IO Libraries Suite 2020 (for Linux)](https://www.keysight.com/upload/cmc_upload/All/ReadMe_2020.htm) ?

MatthieuDartiailh, 2020-11-17:
That is quite surprising. I will see if I can test on some other hardware. 

Yes if you can install the IO library suite to test how it behaves I would be interested. I have some contact with Keysight so I will try to reach out to them to figure out why the instrument is behaving that way.

marcusvalentine, 2020-11-17:
In the meantime, I see the same issue with a modern Keysight EXA spectrum analyser.

MatthieuDartiailh, 2020-11-17:
Interesting. Thanks for the update !

MatthieuDartiailh, 2020-11-17:
Going through the VXI-11 specification again I see some freedom in how to interpret the specification (namely that since no transaction occurs, the instrument does not have to answer). So it would make sense to fix PyVISA-py in that respect. I will try to have a look later this week, but feel free to beat me to it if you feel like investigating. The use of RPC tend to make thing a bit more cumbersome than necessary, so it may not be obvious where to handle the issue.

MatthieuDartiailh, 2020-11-19:
@marcusvalentine can you test https://github.com/pyvisa/pyvisa-py/pull/287 ? it should handle the issue (modulo any silly mistake I made while editing).

marcusvalentine, 2020-11-19:
Have I got this right?

```
pip install git+https://github.com/pyvisa/pyvisa.git@master
pip install git+https://github.com/pyvisa/pyvisa-py.git@tcpip-timeout

pip freeze | grep -i visa
PyVISA @ git+https://github.com/pyvisa/pyvisa.git@6881af78cbdb5abe02452c8736e3737d48b011c4
PyVISA-py @ git+https://github.com/pyvisa/pyvisa-py.git@d198fdf066231be511c9690388404085d2e4047c

```

MatthieuDartiailh, 2020-11-19:
That is the right way but my fix is not correct one. I will ping you again when I get a better solution.