pyvisa/pyvisa #368: Trouble passing string variable to AG34970a and gp-switch card.

czeck6, 2018-10-07:
I'm trying to control a gp-switch card in slot three of an AG34970a using pyvisa.

When I enter this in the console.

```
>>>dac.write("ROUTe:CLOSe (@301,302,303,304,305,306,307,308)")
```

the switch functions normally and quietly closes the appropriate switches. The console outputs

(48L, <StatusCode.success: 0))

my device is named dac.

but when I try to use a string variable instead the response in the console is the same but the instrument beeps and does not close the switches. The error code on the device is error 103 which I believe means an invalid separator but since one works I don't understand why the other wouldn't.

```
>>>print(switch_str)
'"ROUTe:CLOSe (@301,302,303,304,305,306,307,308)"'
>>>dac.write(switch_str)
(48L, <StatusCode.success: 0))  

```
People keep guessing that it's the double quotes but my decision to include the quotes in the string by using the single quote outside the double quote is that the string passed to the device within the parenthesis seems to need be surrounded by double quotes.  If I don't include them python simply does a syntax error.

Is it not possible to use a variable in conjunction with the .write command?

My variable switch_str is actually a concatenation of a couple other parts so I might try doing the concatenation right in the dac.write() part just in case but if anyone has any ideas let me know.  I'll attach the full script if anyone wants to see.  My style is ugly I know so be prepared.  It is for balancing cell voltages in a battery.

[balancer.txt](https://github.com/pyvisa/pyvisa/files/2453994/balancer.txt)

closed 2018-10-07 by czeck6

MatthieuDartiailh, 2018-10-08:
I believe  your issue was the following line:
```python
switch_str='"ROUTe:CLOSe (@' + formatted_str_list + '308)"'
```
which created a string with `"` inside while you probably wanted:
```python
switch_str= "ROUTe:CLOSe (@" + formatted_str_list + "308)"
```
Also noted that to build `formatted_str_list` you could have used `", ".join(str_list)` to avoid the explicit for loop.

czeck6, 2018-10-09:
Thank you.  Yes, that's exactly what I ended up doing.  I just needed some sleep to see what was wrong.  It's funny how much that helps.  I didn't know the join thing.  I'm still very much a newb.  I appreciate your advice.