pyvisa/pyvisa #369: issues with long command strings? - can NI-VISA Run-Time Engine 18.0 be used as backend?

OlafEichstaedt, 2018-10-10:
We are connecting to an ASTERION AST501 power supply over TCPIP using the default '@ni' backend. Communication seems to work, we can connect, send commands and query data from the device using command strings with typically +/-20 bytes. However, we fail to successfully transmit a command with a long command string (approx. 6'000 bytes). 

The supplier provided a remote desktop application that also connects via TCPIP but uses the NI-VISA Run-Time Engine 18.0 as backend. This application communicates without problems even for the long command strings. In fact, the first version of the long command string was copied to Python from that desktop application. 

Now we are wondering:
- is it possible that the 1.9.1 backend (for TCPIP) does something to long command strings so that the device no longer accepts them? .... when we use `visa.log_to_screen()` we do not see any irregularity in the string. So at least PyVisa is claiming (and probably believing) to send the correct string.
- have we overlooked a configuration that may interfere with long command strings?
- is there a way to use the NI-VISA Run-Time Engine 18.0 as backend? There is a directory `National Instruments\Shared\NI-VISA\` with a `VisaCtrl.dll` and a `niVISAsys.dll` but no `visa32.dll` . When we direct Visa to the directory `rm = ResourceManager('[...]\National Instruments\Shared\NI-VISA')` it complains that the backend cannot be found.
- when we try the pyvisa-py ("*py") backend, Visa sends happily, i.e. without error messags, but the device is not receiving anything .... we are still investigating if we have a configuration issue for the pyvisa-py backend. 

Does anybody have any ideas 
- how we could narrow down the cause of the issue to determine if there is a bug?
- if and how it is possible to use the runtime engine version 18 ?

Our current PyVisa setup from `pyVisa-info` is:

```
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      x86 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\[...]\python36-32\python.exe
   Version:        3.6.0
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 23 2016 07:18:10 (#v3.6.0:41df79263a11)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: c:\windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 18874368
         Spec. Version: 5244928
   py:
      Version: 0.3.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

MatthieuDartiailh, 2018-10-10:
Hi and thanks for detailed report.
Here is what you can try to investigate deeper:
- use NI Spy to inspect the calls made by the vendor application. I am not a 100% sure it will work but it is worth a shot. You can also look at pyvisa call this way. One possibility is that the vendor software cut the message in chunks.
- try to send the long command string using NI-Max and play with the chunk-size parameters.
- I am not sure one can use the runtime but I expect the visa library to be more complete than the runtime (which is a slimed down version of the library) and so it should be possible to the full lib to work.

Let me know how it goes. And do not hesitate to report issues you have with pyvisa-py.

OlafEichstaedt, 2018-10-11:
Thank you very much for your answer, I will look if I can find and use NI Spy and NI-Max to investigate what is going on ...

In the meantime, we took another look at the debug information: the command before the failing command and the failing command are reported as follows:
```
2018-10-11 12:39:43,264 - pyvisa - DEBUG - viWrite(36826808, b'trac:def T3\r\n', 13, 'c_ulong(13)') -> 0
define T3
2018-10-11 12:39:48,883 - pyvisa - DEBUG - viWrite(36826808, b'trac:data T3,22875.42,22978.24,20424.3,18930.35,17870.37,17048.18,16376.41,15808.43,15316.43,14882.45,14494.24,14143.07,13822.47,13527.55,13254.49,13000.28,[...],
-20424.3,-22978.24,-22875.42\r\n', 8393, 'c_ulong(8128)') -> -1073807339
```
I notice that for the short 'trac:def' command, the reported length=13 and c_ulong(13) are the same. And does the '-> 0' stand for a successful completion? All the other successful commands seem to show this '-> 0' as well.

For the long (failing) command string, the length is 8393 but u_clong is 8128. I also wonder what '->1073807339' stands for?

OlafEichstaedt, 2018-10-15:
Thanks for the feedback ... in fact, we currently went back a felt 1'000 steps in technology and reverted to serial communication over RS232 which works fine.

I am not very experienced with TCPIP or VISA. So you mean the computer may try a large chunk while the device does not complain about it but at the same time does not know how to handle it? Or does the device ask for smaller chunks and pyVISA ignores it? How would I modify the packet size using  pyVISA? 

Or should I try changing the packet size with Ni-Max first (with which I am not yet familiar) ...

MatthieuDartiailh, 2018-10-16:
I would definitively try to reproduce the behavior you see with NI-Max (since it is a GUI app debugging can be quite fast). As I mentioned you can try playing with the chunck_size which control the size of each chunk. Basically VISA does not have any notion of negotiation between the instrument and the computer so the computer can send too large data and the instrument simply fails.
Also if you can capture the vendor behavior using NISpy it may give us a clearer direction.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh