pyvisa/pyvisa #370: Possible pyvisa.constants duplicates

LaurentRDC, 2018-10-11:
Right now, in the `constants` module, the following enumerations are defined:
```
class StatusCode(enum.IntEnum):
    (…omitted...)
    #: Specified offset is not accessible from this hardware.
    error_nonsupported_offset = VI_ERROR_NSUP_OFFSET

    #: The specified offset is not properly aligned for the access width of the operation.
    error_nonsupported_offset_alignment = VI_ERROR_NSUP_OFFSET
    (…omitted...)
```

Is there a reason that `VI_ERROR_NSUP_ALIGN_OFFSET` is not used above for `error_nonsupported_offset_alignment`? It is defined in `pyvisa.constants` but not used.

If this is indeed a typo, I suggest that the enumerations in `pyvisa.constants` make use of the [`enum.unique`](https://docs.python.org/3/library/enum.html#enum.unique) decorator to ensure this doesn't happen by mistake.

LaurentRDC, 2018-10-11:
There is nothing in the [specifications](http://www.ivifoundation.org/docs/vpp432.pdf) that indicate these two enums should map to the same value.

MatthieuDartiailh, 2018-10-11:
This does look like a typo. Could you make a PR (with enum.unique) ? If you do please add a mention on the changelog.

MatthieuDartiailh, 2018-10-11:
Closed by #371

closed 2018-10-11 by MatthieuDartiailh