pyvisa/pyvisa #372: Message Based write_ascii_values Incorrectly Formats Commands

ryan-summers, 2018-10-19:
Utilizing Message-based resources to write ascii values contains an error in creating the message. A space needs to be inserted between the message and the values blocks to form the message properly.

https://github.com/pyvisa/pyvisa/blob/7f2602270e46cb5c409c5da74d732bcce15bc9d8/pyvisa/resources/messagebased.py#L257

Example:
```
instance.write_ascii_values('RES', [4], converter='d', termination='') 
```
Returns 4 bytes transmitted because the message that was sent was `RES4` instead of the properly formatted `RES 4`, which is 5 bytes in length when the space is included.

MatthieuDartiailh, 2018-10-19:
This may very well be instrument dependent. Some instruments expect for example `MSG:VALUE`. For such cases adding a mandatory space would be a big issue. If you want to submit a documentation update to clarify that the values are directly appended to the message, I would probably be fine with it.

ryan-summers, 2018-10-19:
I think another proposal would be to change the signature to allow for a 'joining' string.
E.g.
```
write_ascii_values(msg, values, converter='f', termination=None, prefix='')
```
where prefix would specify the string placed between the message and value blocks. This would permit `MSG:VALUE` as well as `MSG VALUE`. What do you think?

Also, I was assuming the layer underneath VISA was SCPI (in which case the values are spaced), so thanks for bringing up other scenarios when spaces aren't required. :)

MatthieuDartiailh, 2018-10-19:
No sadly not all instruments use SCPI (or respect the standard at the letter). We could have a joining string but we would have to keep the current behavior for backward compatibility. If you want to go this route please add the same option to `write_binary_values`.

MatthieuDartiailh, 2020-09-06:
The takeaway here is that write_ascii/binary_values append the values right after the specified header without inserting any extra character. 

Closing for now, feel free to reopen if necessary.

closed 2020-09-06 by MatthieuDartiailh