pyvisa/pyvisa #375: Mac Installation: VI_ERROR_LIBRARY_NFOUND

natolambert, 2018-11-07:
When trying to run pyvisa code to connect to GPIB devices, I encounter the following error:

> pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

Full error log here:

> File "IVF_GPIB.py", line 48, in <module>
>     supp= rm.open_resource('GPIB0::'+str(supp_add)+'::INSTR')
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1703, in open_resource
>     info = self.resource_info(resource_name, extended=True)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1663, in resource_info
>     ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/functions.py", line 1383, in parse_resource_extended
>     alias_if_exists)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
>     raise errors.VisaIOError(ret_value)
> pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

The lines of interest are below.
`rm = visa.ResourceManager()`
`supp= rm.open_resource('GPIB0::'+str(supp_add)+'::INSTR')`

Following solutions online, many of them are for windows and suggest pointing to the correct .dll file, by adding the line of code below, but the dll file does not exist on mac (or is my installation incorrect?)
`rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')`

When I check my pyvisa installation with
`from pyvisa import util; util.get_debug_info()`
I get the following output

> Machine Details:
>    Platform ID:    Darwin-18.0.0-x86_64-i386-64bit
>    Processor:      i386
> 
> Python:
>    Implementation: CPython
>    Executable:     /Users/nato/miniconda3/envs/iono/bin/python
>    Version:        3.5.6
>    Compiler:       GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)
>    Bits:           64bit
>    Build:          Aug 26 2018 16:30:03 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.9.1
> 
> Backends:
>    ni:
>       Version: 1.9.1 (bundled with PyVISA)
>       #1: /Library/Frameworks/visa.framework/visa:
>          found by: auto
>          bitness: 32, 64
>          Vendor: National Instruments
>          Impl. Version: 18874368
>          Spec. Version: 5244928
>    py:
>       Version: 0.3.1
>       GPIB INSTR:
>          Please install linux-gpib to use this resource type.
>          No module named 'gpib'
>       ASRL INSTR: Available via PySerial (3.4)
>       TCPIP INSTR: Available 
>       USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
>       TCPIP SOCKET: Available 
>       USB RAW: Available via PyUSB (1.0.2). Backend: libusb1

Installed with:
MacOS 10.14
python 3.5.6
pyvisa 1.1.
pyvisa-py 0.3.1

MatthieuDartiailh, 2018-11-07:
Here is what is happening here: 
Since National Instruments VISA library is not installed, pyvisa falls back to the pure Python backend (pyvisa-py). But pyvisa-py supports GPIB only on linux (and experimental support on windows). As a consequence it tries to use NI VISA (or Keysight or another vendor) shared library but cannot find it. I suggest you try to install NI-VISA (http://www.ni.com/download/ni-visa-18.0/7599/en/). Once done pyvisa should automatically find it.

MatthieuDartiailh, 2018-11-07:
The download is free I believe once you register.

natolambert, 2018-11-07:
I previously have install ni-visa 18.0, and tried reinstalling it with no success. Is there a way to point pyvisa to ni-visa?

natolambert, 2018-11-07:
What is to be done with the shared library?

It's at
`Library/Frameworks/VISA.framework/`

MatthieuDartiailh, 2018-11-07:
Provide this path when opening the resource manager (it is the Mac equivalent of the Windows dll). But I am surprised that ctypes did not find it.

natolambert, 2018-11-08:
I tried passing this directory, in all reasonable forms, into the line
`rm = visa.ResourceManager('/Library/Frameworks/VISA.framework/')`

and got the following error:

> pyvisa.errors.LibraryError: Error while accessing /Library/Frameworks/VISA.framework/: Could not determine filetype.

Full form:

> Traceback (most recent call last):
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/highlevel.py", line 130, in _init
>     lib = Library(self.library_path)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/ctypes/__init__.py", line 351, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: dlopen(/Library/Frameworks/VISA.framework, 6): no suitable image found.  Did find:
> 	/Library/Frameworks/VISA.framework: not a file
> 	/Library/Frameworks/VISA.framework: not a file
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "IVF_GPIB.py", line 46, in <module>
>     rm = visa.ResourceManager('/Library/Frameworks/VISA.framework')
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1526, in __new__
>     visa_library = open_visa_library(visa_library)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1498, in open_visa_library
>     return cls(argument)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 108, in __new__
>     obj._init()
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/highlevel.py", line 132, in _init
>     raise errors.LibraryError.from_exception(exc, self.library_path)
> pyvisa.errors.LibraryError: Error while accessing /Library/Frameworks/VISA.framework: Could not determine filetype.
> (iono) Natos-MBP15:Ionocraft nato$ python IVF_GPIB.py 
> Total Steps: 29 StopRampInt: 9 StopInt: 19
> Traceback (most recent call last):
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/highlevel.py", line 130, in _init
>     lib = Library(self.library_path)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/ctypes/__init__.py", line 351, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: dlopen(/Library/Frameworks/VISA.framework/, 6): no suitable image found.  Did find:
> 	/Library/Frameworks/VISA.framework/: not a file
> 	/Library/Frameworks/VISA.framework: not a file
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "IVF_GPIB.py", line 46, in <module>
>     rm = visa.ResourceManager('/Library/Frameworks/VISA.framework/')
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1526, in __new__
>     visa_library = open_visa_library(visa_library)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1498, in open_visa_library
>     return cls(argument)
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/highlevel.py", line 108, in __new__
>     obj._init()
>   File "/Users/nato/miniconda3/envs/iono/lib/python3.5/site-packages/pyvisa/ctwrapper/highlevel.py", line 132, in _init
>     raise errors.LibraryError.from_exception(exc, self.library_path)
> pyvisa.errors.LibraryError: Error while accessing /Library/Frameworks/VISA.framework/: Could not determine filetype.

Passing '/Library/Frameworks/VISA.framework/VISA' yields the same error as before: 

> pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

Is there not documentation for using Mac with the resource manager?

MatthieuDartiailh, 2018-11-09:
You need to pass the path to the actual shared library not just the path to the folder.

natolambert, 2018-11-09:
What is the "actual shared library"?

MatthieuDartiailh, 2018-11-09:
Actually I just installed Ni-VISA on a mac. When running `pyvisa-info`, the visa library is detected automatically like in your case (something I missed sorry). And I can reproduce your error !

Looking deeper I found a separate NI package with drivers for NI-488.2 (the GPIB standard) at http://www.ni.com/download/ni-488.2-for-mac-os-18.0/7749/en/. After installing it, I get a more sensible error since I don't have a GPIB instrument. 

If this fixes your error please consider contributing to the documentation.

MatthieuDartiailh, 2018-11-09:
And sorry for the not so accurate information in the previous comment ...