pyvisa/pyvisa #376: PyVISA Version: 1.9.1 (miscellaneous error)

javiercasian, 2018-11-08:
Hi 
First of all, thanks for stopping by to read my post. I'm writing a test software to control some instruments described below:

1.- Signal Generator Hewlett-Packard 8648C
2.- RF Counter Hewlett-Packard 53210A
3.- DAQ Hewlett-Packard 34970A

The problem I'm having is that some of the times when I run my test script, instruments works as expected, but, some others don't. Please see below my system and VISA information:

Microsoft Windows [Version 6.1.7601]
  Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\M>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 70 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.15
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr 30 2018 16:30:26 (#v2.7.15:ca079a3ea3)
   Unicode:        UCS2

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
         #1: C:\Windows\System32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377151778
         Spec. Version: 5244928
  #2: C:\Windows\System32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377151778
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named serial
      TCPIP INSTR: Available
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

As well, please see below my code:

 import visa 
 import time
 import pyvisa

  try:
    inst_visa = ("C:\Program Files\\IVI Foundation\\VISA\\Win64\\agvisa\\agbin\\visa32.dll")
    rm = visa.ResourceManager(inst_visa)
    rm.list_resources()
    print(rm.list_resources())
    print("Requesting Instruments ID \n")

    SG8648C = rm.open_resource("GPIB0::12::INSTR")
    SG8648C.timeout =  5000
    print (SG8648C.query("*IDN?"))
    time.sleep(3)

When I ran my code using the pyVISA Debug Mode, this is the error I'm getting:

    2018-10-11 19:09:37,707 - pyvisa - DEBUG - viParseRsrcEx(18L, 'GPIB2::11::INSTR', 'c_ushort(1)', 'c_ushort(2)', <ctypes.c_char_Array_256 object at 0x000000000A03D8C8>, <ctypes.c_char_Array_256 object at 0x000000000A03D0C8>, <ctypes.c_char_Array_256 object at 0x000000000A03BF48>) -> 0
    2018-10-11 19:09:37,717 - pyvisa - DEBUG - GPIB2::11::INSTR - opening ...
    2018-10-11 19:09:37,737 - pyvisa - DEBUG - viOpen(18L, 'GPIB2::11::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x000000000A03D0C8>') -> 0
    2018-10-11 19:09:37,747 - pyvisa - DEBUG - GPIB2::11::INSTR - is open with session 25
    2018-10-11 19:09:37,757 - pyvisa - DEBUG - GPIB2::11::INSTR - closing
    2018-10-11 19:09:37,767 - pyvisa - DEBUG - viDisableEvent(10L, 1073709055, 65535) -> 0
    2018-10-11 19:09:37,778 - pyvisa - DEBUG - viDiscardEvents(10L, 1073709055, 65535) -> 1073676292
    2018-10-11 19:09:37,798 - pyvisa - DEBUG - viClose(10L,) -> 0
    2018-10-11 19:09:37,808 - pyvisa - DEBUG - GPIB2::11::INSTR - is closed
    2018-10-11 19:09:37,848 - pyvisa - DEBUG - viSetAttribute(25L, 1073676314, 5000) -> 0
    2018-10-11 19:09:38,128 - pyvisa - DEBUG - viWrite(25L, '*IDN?\r\n', 7, 'c_ulong(7L)') -> 0
    2018-10-11 19:09:38,138 - pyvisa - DEBUG - GPIB2::11::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
    2018-10-11 19:09:45,197 - pyvisa - DEBUG - viRead(25L, <ctypes.c_char_Array_20480 object at 0x000000000A03D0C8>, 20480, 'c_ulong(0L)') -> -1073807360
    2018-10-11 19:09:45,217 - pyvisa - DEBUG - GPIB2::11::INSTR - exception while reading: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
    Buffer content: bytearray(b'')
    Unhandled exception while debugging...
    Traceback (most recent call last):
      File "\\Mac\Home\Documents\Python Scripts\Python Win Scripts\F-B\F.009.py", line 82, in <module>
        print (RF53210A.query("*IDN?"))
      File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 569, in query
        return self.read()
      File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
        message = self._read_raw().decode(enco)
      File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
        chunk, status = self.visalib.read(self.session, size)
      File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
        ret = library.viRead(session, buffer, count, byref(return_count))
      File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
        raise errors.VisaIOError(ret_value)

MatthieuDartiailh, 2018-11-09:
I would suggest you update your Keysight library. From the report, it looks quite old. And I have no better suggestion at the moment.

javiercasian, 2018-11-09:
Hi 
I'm using the 2018 libraries https://www.keysight.com/en/pd-1985909/io-libraries-suite?cc=US&lc=eng

      IO Libraries Version:	18.1.23218.2
       Copyright © Keysight Technologies, 1983-2018

      This software is installed on a 64-bit operating system.

      ==============================
      64-bit VISA information (2 installed):
          Keysight VISA
          National Instruments VISA

       64-bit IO Libraries Path:
         C:\Program Files\Keysight\IO Libraries Suite\
      64-bit IVI Components Path:
         C:\Program Files\IVI Foundation\IVI\
      64-bit VISA Path:
         C:\Program Files\IVI Foundation\VISA\

      ==============================
      32-bit VISA Information:
         Keysight VISA: 	 The primary VISA library

      32-bit IO Libraries Path:
         C:\Program Files (x86)\Keysight\IO Libraries Suite\
      32-bit IVI Components Path:
        C:\Program Files (x86)\IVI Foundation\IVI\
      32-bit VISA Path:
         C:\Program Files (x86)\IVI Foundation\VISA\

      ==============================
      Registry Path:
         HKLM\SOFTWARE\Keysight\IO Libraries Suite\CurrentVersion

MatthieuDartiailh, 2018-11-09:
I got confused by the version number 1377151778 since I knew Keysight was at 18. One thing you can try is use NISpy (or a Keysight equivalent) to capture the VISA activity. Also do you see the same error if you don't run through the pywin debugger ?

javiercasian, 2018-11-09:
Hi Matt
I have updated my Keysight libraries, in both cases I have the same error. Also, Running as "stand alone" (py.exe) I have the same error. 

The funny thing thought, as I mentioned in my post, is the fact that instrument is doing what supposedly be doing it, I see the instrument interact as I sent the command, however, it crash the pyVISA. 

[IOMonitorLog.zip](https://github.com/pyvisa/pyvisa/files/2566613/IOMonitorLog.zip)

I have attached the Keysight IO Monitor Log and screen shot of the problem too.

Thanks for your help, 
[image]

javiercasian, 2018-11-09:
Hi Matt, 
Here is another error... apparently, VISA could fail at the time that writes or reads. Please, see below. 

Btw, I have add a delay into my code, please see below also: 

          DA34970A = rm.open_resource('GPIB2::10::INSTR', query_delay = 2)
          DA34970A.timeout =  20000
          SG8648C = rm.open_resource('GPIB0::12::INSTR', query_delay = 2)
          SG8648C.timeout =  20000
          RF53210A = rm.open_resource('GPIB1::11::INSTR', query_delay = 2)
          RF53210A.timeout =  20000
    
          logging.info("initializing the instruments... \n")
          SG8648C.write("*rst; status:preset; *cls")
          time.sleep(.25)
          RF53210A.write("*rst; status:preset; *cls")
          time.sleep(.25)
          DA34970A.write("*rst; status:preset; *cls")
          time.sleep(.25) 
          
[image]

MatthieuDartiailh, 2018-11-09:
Can you also post the Python traceback of the second error ?

MatthieuDartiailh, 2018-11-09:
Can you also post your full test script ? I would like to know what you do in the finally or except clause.

MatthieuDartiailh, 2018-11-09:
Do you have any kind of front panel open to the instrument that may interfere with the communication ?

javiercasian, 2018-11-09:
Hi Matt 
Please see below the traceback: 

      Traceback (most recent call last):
        File "\\Mac\Home\Documents\Python Scripts\1.012.py", line 105, in <module>
          logging.info(SG8648C.query("*IDN?"))
        File "C:\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 569, in query
          return self.read()
        File "C:\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
          message = self._read_raw().decode(enco)
        File "C:\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
          chunk, status = self.visalib.read(self.session, size)
        File "C:\Python34\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
         ret = library.viRead(session, buffer, count, byref(return_count))
        File "C:\Python34\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
          raise errors.VisaIOError(ret_value)
      pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error 
      (miscellaneous error).

I also have edited my code so you can see all of it.

MatthieuDartiailh, 2018-11-09:
Do you always see the error on the same instrument ? Can you reproduce the error using the Keysight Expert connection software ? (the error seems to occur deep in Visa so I am wondering if pyvisa gets a configuration wrong)

javiercasian, 2018-11-09:
Hi Matt,
That is a very good question, "Do you always see the error on the same instrument ? " the answer is NO, however, is happen often in the DAQ D34970A. 

Can you reproduce the error using the Keysight Expert connection software ? Yes, please see below:

1.- My system: 

[image]

2.- Keysight Connection Expert Error:
[image]

So, do you think is a Keysight  VISA issue and pyvisa basically is just moving the error from Keysight  VISA into PYVISA  ?

MatthieuDartiailh, 2018-11-09:
Looks a lot like this yes. Or the instrument does something unexpected by the library (since those are rather old ones they may not fully adhere to the standard). If you can give a try to the National instrument implementation.

javiercasian, 2018-11-09:
Not sure what is a "National instrument implementation."

dmchester, 2018-11-10:
Hi
I have some recent experience that may be useful.  I am using Keysight's Vee Pro DirectIO feature to send ASCII commands (*IDN?) to an LXI (TCPIP) device.  The instrument vendor provides a GUI tool that loads the NI VISA driver, after initial failures I loaded Keysight's latest VISA drivers - it still failed.  The Vee Pro Direct IO calls lock the instrument before it sends the command so it always fails with a device locked error.  I am certain that this is a Direct IO issue but while diagnosing it I found the following useful (BTW Keysight's Command Expert works, so it looks like they aren't using Direct IO!).
When the Keysight VISA is the primary driver (ie it was installed last), open Keysight's IO Monitor and use the Options to turn on all the lower level protocols (eg SICL).  The SICL commands may reveal the true nature of the failure.
I am migrating to pyVISA to solve the Direct IO issue.
Hope this helps.

javiercasian, 2018-12-04:
Ok, after way so many days and nights working on this issue... I guess I have found the root cause of this problem. 

Per your recommendation, I have requested Agilent technologies support, however, I never receive any support from them. 

After I explored all WEB PAGES possible over the internet, here is what happened. 

I use a MacBook Pro, Mojave OS, with Windows 7 (As Virtual Machine -Parallels-), well, I've swap to a DELL desktop PC running Windows 10 64 Bits and guess what ..... I've running my code for the last 5(ish) hours with NO ISSUES

Everything works as it should be. 

Seems like Virtual Machine - Mac Book Pro and Agilent Technologies VISA aren't friends anymore.

Thanks for your support and help. I think we can close this issue now. 

Regards 
Javier

MatthieuDartiailh, 2018-12-04:
I can believe that VISA indeed does not behave well from inside a VM. I may add a note to the docs in that sense.

closed 2018-12-04 by MatthieuDartiailh